thotro/arduino-dw1000 #1: Data receive issue

jonathand1, 2015-06-17:
Hi,
I have two DWM1000 and I'm trying to make them communicate.
But when I send message to the other (with one of your test, like pin pong one), the receiver receives the sent message with a square letter at the end. Next times, only squares are received by the receiver.
Is it an issue you're also experiencing ?
Kind regards.

thotro, 2015-06-18:
Hi there,

unfortunately yes, I observe the same issue - thanks for pointing out! 
I must have introduced a bug with one of the recent commits, but will come up with a fix shortly!

Cheers, Thomas

stackbase, 2015-06-18:
Thanks for tout reply.

Do you confirm all inputs are at 3.3V (ground excepted of course) ?
Asking you this question because I dev with à 5V Arduino, and I have to pull down to 3.3 on all pins with resistors, and sometimes I get bad information from Device id and more... Like all data to zero for instance

thotro, 2015-06-18:
Thanks for your comment! Yes, I primarily use a Pro Mini, one which already operates at 3.3V.

At the moment I have an un-pushed fix for the issue of sometimes reading/receiving bad data - it was a message checksum problem. Though, it still requires some further testing, but should get the driver back on track in a moment.

Thomas

closed 2015-06-20 by thotro