thotro/arduino-dw1000 #2: Write API docs

thotro, 2015-06-19:


jonathand1, 2015-06-20:
Seems important.
Have you measured the accuracy of the ranging operation between two dwm1000 :) ?

thotro, 2015-06-20:
I already started to do tests on it, but the code quality was not sufficient at that time to continue. I think I can provide the first useful results on ranging by mid to end of next week.

I hope and guess that the ~10cm accuracy (was it?) that has been proclaimed by Decawave can be reached. Some of my early debug outputs already told me a TOF of about 1ns (i.e. 30cm), which, at that time, almost correctly resembled the distance between my two USB ports that powered the Arduinos. ;)

... will keep you posted.

closed 2015-08-10 by thotro