thotro/arduino-dw1000 #3: Extensive testing of certain configurations

thotro, 2015-06-19:
Testing and benchmarking of different settings (data rate, PRF, channels, etc.).

thotro, 2015-07-05:
Tested for all modes except 64 MHz, but should be fine too. Still, will be tested in the next couple of days.

leosayous21, 2015-07-26:
Hi ! Before wiring my arduino Nano which is working with 5v logic, i would like to be sure that i don't risk anything with 5v ? Did you try or just used 3.3V with arduino mini ?
Thanks !

thotro, 2015-07-26:
Hi! Yes, you should definitely NOT use 5V input, it will most likely destroy the IC. I have Mini Pro's that already take 3.3V and for other boards I use voltage regulators. 

Just had a quick look at an image of the top layer of the Nano board: What about using the pin right next to digital pin "D13"? It seems to say "3V3", which would mean 3.3V (output?) to me.

Hope it helps!

leosayous21, 2015-07-26:
Hi ! thanks for your reply!
yes there is a 3.3V output for tiny supplying but as the atmega on the arduino nano is supplied with a 5v voltage, the logic I/O of the board is 5v-0v ... i have 2 boards which are in 3.3V so i'm going to try that ! Otherwise there is still the possibility to use a standalone atmega ^^
Cheers !
Léopold

thotro, 2015-07-26:
... oh, sure - I forgot about the I/O part - it's already late! ;-)

leosayous21, 2015-07-27:
Hi ! I made them work thanks to your code ! 
I had really hard time to make all of this work... I used standalone atmega in 8Mhz internal clock (i just have 16Mhz quartz and you can't use them for 3.3V).
I didn't have a really good accuracy (something like +-30/30cm). Do you think the internal clock of the atmega can have an impact on it ? (internal clock is not accurate, it's around 8Mhz +-10% !) It seems that the duration of one iteration of the microcontroller does not affect the calcul of the ranging (travel time).
Also, my chip heated. I saw there is a thermometer inside the chip. Do we need to use it in order to correct the ranging or does the module correct it by itself ?
Thanks a lot !
Léopold

thotro, 2015-07-28:
Hi! Great that it works!

I guess that the accuracy doesn't depend on you uC clock; the module has a crystal osc. on board and the uC only reads the IC-side timestamps. So delays or other inaccuracies won't matter, I believe.

Yes, there is a temperature sensor inside and another yes: one can use it to adjust timestamp results. Unfortunately I have not implemented the adjustment yet, because from the datasheet it seemed to me that this is not so important. But who knows, maybe it is and I should simply do it :-)

Your 200mA are really high, I have to measure it too. Will come back to you on this!

Cheers!

leosayous21, 2015-07-28:
Thanks ! It's interesting to think that it can work in internal clock. I thought after that my question was stupid because an order of magnitude for Mhz drift would imply 1000m drift ^^ (and not centimeters...).
The datasheet indicates a 170mA max for RX mode and with a 40mA for the standalone atmega the result seems plausible.

Did you supply the module with your arduino pro mini (max current 150mA) ? I used to supply it with a arduino uno but the max output current for 3.3v is 50mA ! :O (that's why i have 2.7v and the chip which randomly restart... I will try with an external 3.3v regulator !)

thotro, 2015-07-29:
Hi! You're right, the 200mA are plausible. Maybe I should implement a setting that after an TX/RX the IC can be put to sleep for a specific time span. This would allow to save energy while still one can follow a certain timed protocol (like "I sent something and expect a response as early as in XY ms, so I can safely take a nap for about XY ms"). What do you think?

I use an external power supply; the IC works fine in this setup.

Cheers!

closed 2015-08-10 by thotro