thotro/arduino-dw1000 #6: Tests on ranging accuracy

thotro, 2015-06-20:
First ranging tests are planned to be done till the end of June.

ankitbhatia, 2015-06-25:
Hey, 

It looks like you're working hard towards getting this library out there. Thanks! 

I'm trying to implement your library on two arduino nanos. I was able to validate your sender and receiver examples but unfortunately the two way ranging code doesn't work well yet. Have you tested your two way ranging code successfully? I seem to get either completely absurd time measurements or a constant time of about 1.4~1.5 ns.

thotro, 2015-06-26:
Hi there,

my pleasure, I hope it will be entirely useful in the near future! ;)

Unfortunately I had some other parts to work on the last couple of days, so I could not spend to much on the ranging reference example. For the moment this code has definitely issues: Either the LDE microcode is not yet fully loaded, IC calibration is the key or there is a hidden bug in timestamp management. Finally, on the surface I print the TOF as a float value, which is inherently imprecise.

Currently I do a first draft of some API docs, but will then immediately head over to fix/implement the ranging stuff.

If there's anything that occurs to you that you think could be a bug, please drop me some lines; otherwise I'll keep you posted!

Cheers, Thomas

ankitbhatia, 2015-07-08:
Hey, I do not know if you are able to run the ranging code. It does nothing for me. The basic sender and receiver code is also not working. I get very rare message transmissions.

thotro, 2015-07-08:
Yes, for me the ranging code did work on the last commit (but will check again tomorrow). 

Rare message transmission sounds like there is still a bug in the tuning of the currently employed mode. For a quick test you could choose another mode (see http://tinyurl.com/enableModeFunc, call within "new..." and "commitConfiguration", MODE_SHORTDATA_FAST_LOWPOWER would be the chip default) and see if things can be made working.

Hope it helps!(?)

jonathand1, 2015-07-09:
Hi,

For me, the ranging test does not work anymore. No message sent from the tag for beginning ranging process.

However, send/receive tests works perfectly.

😊

Waiting for reply,

Thanks for your work !

De : Thomas Trojer
Envoyé : ‎jeudi‎ ‎9‎ ‎juillet‎ ‎2015 ‎01‎:‎05
À : thotro/arduino-dw1000

Yes, for me the ranging code did work on the last commit (but will check again tomorrow). 

Rare message transmission sounds like there is still a bug in the tuning of the currently employed mode. For a quick test you could choose another mode (see http://tinyurl.com/enableModeFunc, call within "new..." and "commitConfiguration", MODE_SHORTDATA_FAST_LOWPOWER would be the chip default) and see if things can be made working.

Hope it helps!(?)

ankitbhatia, 2015-07-09:
Hey,

So I tried a bunch of other modes. The ranging demo doesn't work on any of them. I was only able to get the send recv demo to work on the LONGDATA_FAST_ACCURACY mode and the SHORTDATA_FAST_LOWPOWER mode. 

Do you have any ideas why that could be happening? 

My arduino sees a reset in almost all other modes after the first message. 

Regards,
Ankit

thotro, 2015-07-11:
I'm sorry that you're stuck with it for the moment and the delay, I was horribly busy for the last few days. :-/

I will do a thorough check of all tuning parameters and regular settings (something I have to do anyway for a stable release) to see the problem. It usually boils down to this one bit, set at the wrong place at the wrong time. ;-)

Best,
Thomas

leosayous21, 2015-07-11:
Hi !
I'm waiting for my DW1000 to arrive in order to try your code and work on it if i can :-)
Did you hear about that ? : http://www.pozyx.io  (It seems that they are using this module)
it seems that they will partly open the source.
from here: https://www.kickstarter.com/projects/pozyx/pozyx-accurate-indoor-positioning-for-arduino/comments
"Most of the software will be open source: 
- all the Arduino libraries to communicate with the pozyx board. 
- all the Arduino examples for the board. 
- The pozyx firmware will be partly open: we will make most of the firmware available, including tutorials and examples to read out all the different sensors or how to make range measurements. However, the positioning algorithms will only be available through a library."
Regards,
Léopold

thotro, 2015-07-11:
Hi all,

@ankitbhatia and @jonathand1 in the latest commit I did some minor bugfixes and tested all modes for use with the ranging example; so far I had no problems. Would be pleased to here if it works for you again.

@leosayous21 yeah, just recently heard about their work on pozyx, thanks for the pointer. Hopefully I can get one when it's out on the market - looks like a nicely produced piece of hardware. :-)

ankitbhatia, 2015-07-13:
Ranging seems to work now although it is not very accurate. I will try to do an accuracy test and share the results with you.

leosayous21, 2015-07-13:
I think we need to do some average accuracy and to see how many samples we need in order to have a good accuray at the end.

thotro, 2015-07-13:
@ankitbhatia That would be great, thanks!

With one of the currently implemented modes (forgot which one, I just noticed during testing) and current parameters for tuning the chip I can get at best about +-30cm.

Below a certain distance (~75cm) results jump and don't seem to be very stable. Also negative values can/should be dropped I think.

Please note that this is not yet the optimally configured device - obviously ;-)
 * Considering measured noise, RX power level and first path signal power level should yield accuracy of another few cm.
 * Smart TX power is still on; this may not be the best option for ranging.
 * The code still uses standard SFD, although Decawave says their custom one gives better results.
 * ... (and I'm sure there is something else left to tweak)

@leosayous21 yes, for actual production code anything from averaging samples to Kalman filters might be an option to improve or stabilize results further.

I will find time to implement the new features by mid-to-end of this week and then let's see again. It would be great to see that +-10cm are possible out-of-the-box.

ankitbhatia, 2015-07-13:
I believe you on the tuning parameters. This is a complex enough chip that I bet there must be something more. 

As for the measurements at short range, I think the distance becomes non-linear there and we need to average for a bit and then try to fit a non-linear function. I don't know what function yet. Maybe the negative values have information in them. I will report back when I actually do some ranging tests.

leosayous21, 2015-07-13:
@thotro did you read "Sources of error in DW100 based two-way ranging schemes" (APS011 Application Notes) ? 
They talk about errors related to clock drift in two nodes and related to the incident signal level at a node
It seems that there are a lot of differents bias possible. here you have the graphs:
[image]
[image]

leosayous21, 2015-07-13:
Also, the clock drift seems to have a huge impact on the ranging accuracy

thotro, 2015-07-14:
@leosayous21 many thanks for pointing out - yes, I know the document. Parts of this will directly find itself in the timestamp correction routine that will then automatically adjust each RX timestamp right away. Next on the agenda! :-)

leosayous21, 2015-07-21:
@thotro Hi ! Thank you for your release :-)
 What ranging accuracy do you obtain at the end ? What additional optimisation can be done now in your opinion ? 
Also did you try to pass through wall ? (they say it can work through 1 or 2 walls)
Cheers, 
Léopold

thotro, 2015-07-21:
Hey Léopold,

in the current release I obtain about +-15cm (just a guess from my observations). What I mean by this is that the ranging results are stable with these tolerances from measurement to measurement.

To get absolute and correct distance measurements the module needs to be calibrated/biased (e.g. choosing values for the antenna delay) - but this should be easy then and stable ranging results are definitely more important, I think. (... the API for calibration will be available soon)

Yes, I also read about the "1 or 2 walls". I guess it depends on the walls: No chance to get through a single of my 50cm brick walls at my home! ;-) But I still need to try it with increased TX power and better antennas are obviously an option as well. Let's see if I stumble across a few other types of walls to test this furthermore. :-)

I keep you posted if I have results.

Cheers,
Thomas

leosayous21, 2015-07-22:
hey thomas,
I really look forward to see the accuracy you get at the end :-)
Do you know how long take a ranging (1 or 2ms no ?) and so what ranging frequency we can have with the module ? 
cheers,
leopold

leosayous21, 2015-07-28:
@thotro Hi !
Here the distribution I have with your examples for ranging
Samples number: 446 
mean: -3,118587444 (m in relative)
std: 0,085572889

[image]

It's really good i think ! You did a really good job :-)
Léopold

thotro, 2015-07-28:
Great!! Thanks Léopold, that's very helpful! You provide the stats I somehow never find time for! :-)
Very much appreciated!

Let's see if we can get better! ;-)

leosayous21, 2015-07-28:
@thotro I've had such a hard time to enter your code so i decided to do something easier XD 

Here an other test ! 
Samples number: 618
mean: -2,897082683
std: 0,078181635
RX power around -90dBm

[image]

I didn't move the modules, so the real distance is the same as for the other graphs. It's interesting to see that the mean has a 20cm drift in comparison to the other experience. 
The second experience had a mean RX power of -90 dBm whereas the first one was around -95 dBm. Also, the temperature might be different.

An other feeling i have is about a tiny drift while it was working, that why we have a "plateau" around -2.9 
This can be due to the heating of the chip which drift the result a little bit ?

leosayous21, 2015-07-31:
Hi!
I did others tests ! I created my own pcb because of this problem of 3.3v for me... Also, it allows me to put my own regulator which has a limit of 800mA. I put a ground under the module but it was just a one sided PCB so there is no ground in the other side. 
[image]

(i can share the files if you want)

I noticed that the RX dBm is really better when putting the module at the vertical and not horizontal ! (it goes from -90dBm to -67dBm !). Also the module heat more maybe because the supply is more powerfull (i need to measure the current)

here my results:
Samples number: 558
mean: 1.501
std: **0.0228**
RX power: around **-67dBm**
Receive quality: around 350
FP power: -82 dBm
Real distance: around 1.3m (not measured)

[image]

The ranging is really accurate ! I'm really impressed
 Cheers,
Léopold

thotro, 2015-07-31:
Hi Léopold,

great, the results really look impressive! RX power levels for this specific distance should be like that - hence my PCB layout was far from optimal. If you want to share your files with us this would be great. Always good to have some more resources and knowledge gathered together.

Thanks,
Thomas

MarioRuz, 2015-08-01:
Hi!

First of all, thank you Thomas for your valuable work. It is actually helping me to learn writing libraries :). I already have 3 dw1000. I can also confirm your libraries work with Arduino UNO and Arduinio Duemilanove. Leopold, I would be interested in the gerber files to build that small pcb :).

I am planning to do a set of ranging tests to evaluate the accuracy. However, I am facing some issues, and maybe you already has the answer:

1- When running examples such as ranging (RangingAnchor and RangingTag), how do you proceed to start and how do you reset transmission? In my case I plug the arduino "anchor" and after the arduino "tag". However, sometimes transmission stops, with serial monitoring I only get "### DW1000-ard", and nothing else. Does this happen to you? To restart the transmission I need to deplug and plug the serial interface, and sometimes more than once.

2- Do you get "outliers" when getting the distance?. In the case of Léopold graphs I don't know if he filtered data first. I did some test. For example, when tag and anchor are 50 cm I get values like these (notice the outlier data -28446,8800000000, I also got positive outliers, like 13523)

...
-2,01000000000000
-2,04000000000000
-1,97000000000000
-2,05000000000000
-2,02000000000000
-1,97000000000000
-2,04000000000000
-1,99000000000000
-2,18000000000000
-1,99000000000000
-28446,8800000000
-2
-1,98000000000000
-2,04000000000000
-2,07000000000000
-1,92000000000000
-2,05000000000000
-2,05000000000000
-2,02000000000000
-1,98000000000000
....

Being this my case, If i suppose a normal distribution (just as a first analysis) my std will be much bigger than Leopold results. Let me know if you did some filtering.

3- It seems when anchor and tag are very close each other, let's say less than 10 cm, transmission stops, does this happen to you too? I am also interested in being able to get data even when tags are very close each other.

Cheers!

leosayous21, 2015-08-01:
Hi @MarioRuz !
I don't have the gerber files but i will give the eagle files (and eagle library) to thomas in my next pull request :-) It's just a one sided pcb i did using the toner transfer method.

1. I used to have this problem due to the current problem. How do you supply your module ? The arduino uno provide only 50mA for the 3.3v ... It's really crap. That's why i did my own pcb with a onboard 3.3v regulator (LM1117).

2. Your values are not so bad i think. For my tests i erase the very huge number (as 28466) by hand because it really alters the results but i didn't do any other filtering. 
I think it comes from error when re-sending the message in case one message was lost. It added some microseconds which corresponds to ten thousands of meters...

3. I didn't try that..

@thotro I'm currently doing the correction on RX power ! I put these functions in the rangingAnchor sketch for now but i think it will be interesting to create a new class something like DW1000Ranging which will contain all the RanginAnchor & RangingTag functions. In this case it will be much easier to implement it in an arduino sketch and to add others functionalities (as multiple anchor, different correction, etc. !). What do you think about it ?
We can do it in using the famous type: " DW1000Ranging.available() " and " DW1000Ranging.read()" "DW1000Ranging.write(String data)" etc..

Cheers,
Léopold

thotro, 2015-08-01:
@MarioRuz Hi! With the last commit I added correction of timestamps if system clock overflows occur. This should at least help with the really large value outliers. If a message was lost during the ranging protocol, the entire ranging sample should be discarded - at least this was the intention, so I will re-check the code on it. :-)

With distances below 50cm I have the feeling that the timestamping and hence the actually computed range is not very accurate anymore. It seems that even the official eval board's ranging algorithm does not treat such distances at all - if I observed correctly. Still, I guess a little more testing may further clarify low-distance ranging capabilities.

@leosayous21 Hey! Great that you did an implementation on timestamp correction. I very much like the idea to extend the library towards a ranging class (I think that in the end we all got onto the DW1000 to have this ;-)). Although I think that the timestamp correction should go directly into the existing DW1000Time class, since each and every timestamp can/should be corrected - what do you think? Also I think that we could try to approximate the functions given in the power level / distance bias graphs - would be more efficient and maybe more accurate.

For a ranging class I guess it will be really important to have MAC level messages and node addressing and filtering implemented. I would love to setup a node simple by saying "I'm a node with address XY and my anchors are A, B, C, D".

Cheers! :-)

bigredbee, 2015-08-08:
I just want to thank all of you for the work you're doing.  I was able to get mine up and running in just a few hours with no modifications to the code using a SparkFun Pro Micro.  I had the adapter board fabbed at OSHPARK, three for just $7 USD, can't beat that price!  If anyone is interested in getting these boards for yourself, I can make them public and you can order with a click of a button.

I really look forward to being able to introduce multiple anchors into the network!

Greg

[image]

thotro, 2015-08-09:
Hi Greg,
looks great! Will do our best to further improve and in the meanwhile, have fun! ;-)
Cheers,
Thomas

leosayous21, 2015-08-09:
Hi @bigredbee  !
Look that you have a good equipment for testing the module and the library :-)
I look forward to see your result !
Cheers,
Léopold

bigredbee, 2015-08-09:
I just downloaded  the latest code, recompiled the anchor / tag and
uploaded w/out any changes.

Here's what I get from the anchor even though they are just 2 meters apart.

Do I have to change something in the examples?

Greg
..
from: 7D:00:22:EA:82:60:3B:9C    Range: -123.41 m        RX power: -89.57
dBm
from: 7D:00:22:EA:82:60:3B:9C    Range: 252.49 m         RX power: -89.22
dBm
from: 7D:00:22:EA:82:60:3B:9C    Range: 276.83 m         RX power: -88.40
dBm
from: 7D:00:22:EA:82:60:3B:9C    Range: -158.38 m        RX power: -88.44
dBm
from: 7D:00:22:EA:82:60:3B:9C    Range: -156.05 m        RX power: -88.53
dBm
from: 7D:00:22:EA:82:60:3B:9C    Range: 255.31 m         RX power: -89.12
dBm
from: 7D:00:22:EA:82:60:3B:9C    Range: -129.64 m        RX power: -89.57
dBm
from: 7D:00:22:EA:82:60:3B:9C    Range: -125.54 m        RX power: -89.48
dBm

leosayous21, 2015-08-09:
This come from the DW1000Ranging class... we are working on it !
Normally the RangingAnchor and RangingTag example works great !

bigredbee, 2015-08-09:
OK, I"ll wait until I hear from you that it's stabilized.

Kris523, 2015-09-29:
Hello!
Some quick test:
Samples: 4807
[image]
Note: No antenna calibration has taken place: most measurements are about 20cm too far, so overall both precision and accuracy look really good (Once antenna calibration API gets up)! 
This is evidence of your great work!

I wonder how both the accuracy and precision will do when we manage to increase transmission distances!
All the Best!

thotro, 2015-09-29:
Great, thanks for your efforts and sharing with us! 

Antenna calibration is now on the top of my list ;-) I will also double-check (and hopefully be able to fix) transmission power settings soon.

Cheers, Thomas

lesjno, 2016-06-09:
Hi,
I post my message here, even it has not been updated since Oct 2015...
I'm currently testing DW100 ... and I'm a little desappointed. 
I use the last src code from this web site. It runs on 2 Arduino Mega2560R3.
The IRQ pin used is the pin 2.
I performed the first test with BasicSender/Receiver sketches but I can see a lot of messages are corrupted.
Nevertheless I decide to test the DW1000Rangig_TAG and DW1000Ranging_ANCHOR sketches : That works also but it's not really accurate.
In order to get more visual results, I changed a little the TAG sketch to add several data and calculate an average with 20 moving values =>
Here is the code of my "newRange" function =>

void newRange() {
// (varables have been declared before. NBVAL is set to 20) 
// new values :
  H = millis();
  DT = H - Hmemo ;
  Hmemo = H;
  Adr = DW1000Ranging.getDistantDevice()->getShortAddress();
  Dist = DW1000Ranging.getDistantDevice()->getRange();
  Pow = DW1000Ranging.getDistantDevice()->getRXPower();

  // indicial incrementation
  indiceCourant++;
  if (indiceCourant >= NBVAL) indiceCourant = 0;
  
  // removing the older values
  DTmoy -= tabDT[indiceCourant]/NBVAL ;
  DistMoy -= tabD[indiceCourant]/NBVAL ;
  PowMoy -= tabP[indiceCourant]/NBVAL ;

  //adjust min and max with the last values
  // Dist
  // ****
  if (Dist > Distmax) Distmax = Dist;
  else if (tabD[indiceCourant] >= Dist) Distmax = ValMax(tabD,NBVAL);

  if (Dist < Distmin) Distmin = Dist;
  else if (tabD[indiceCourant] <= Dist) Distmin = ValMin(tabD,NBVAL);

  // Pow
  // ****
  if (Pow > Powmax) Powmax = Pow;
  else if (tabP[indiceCourant] >= Pow) Powmax = ValMax(tabP,NBVAL);

  if (Pow < Powmin) Powmin = Pow;
  else if (tabP[indiceCourant] <= Pow) Powmin = ValMin(tabP,NBVAL);

  // DT
  // ****
  if (DT > DTmax) DTmax = DT;
  else if (tabDT[indiceCourant] >= DT) DTmax = ValMax(tabDT,NBVAL);

  if (DT < DTmin) DTmin = DT;
  else if (tabDT[indiceCourant] <= DT) DTmin = ValMin(tabDT,NBVAL);

  
  // put the new values in the tables
  tabDT[indiceCourant] = DT;
  tabD[indiceCourant] = Dist;
  tabP[indiceCourant] = Pow;
  
  // add new values to the average
  DTmoy += DT/NBVAL ;
  DistMoy += Dist/NBVAL ;
  PowMoy += Pow/NBVAL ;
  
 // print values : to use copy/past the values displayed from the terminal to Excel (for example)
// Warning : ignore the 21 first data rows (needed to fill the tables ...)
  if (passage==0){
    Serial.print("from: "); 
    Serial.print("\t Range: "); 
    Serial.print("\t DistMoy: ");
    Serial.print("\t DistMax: ");
    Serial.print("\t DistMIn: "); 
    
    Serial.print("\t RX power: ");  
    Serial.print("\t PowMoy: ");
    Serial.print("\t Powmax: ");
    Serial.print("\t Powmin: ");
    
    Serial.print("\t DT: ");
    Serial.print("\t DTmoy: ");
    Serial.print("\t DTmax: ");
    Serial.print("\t DTmin: ");
    Serial.println();
  }
  
  Serial.print(Adr, HEX);
  Serial.print("\t"); Serial.print(Dist);
  Serial.print("\t"); Serial.print(DistMoy);
  Serial.print("\t"); Serial.print(Distmax);
  Serial.print("\t"); Serial.print(Distmin);
  
  Serial.print("\t"); Serial.print(Pow);
  Serial.print("\t"); Serial.print(PowMoy);
  Serial.print("\t"); Serial.print(Powmax);
  Serial.print("\t"); Serial.print(Powmin);
  
  Serial.print("\t"); Serial.print(DT);
  Serial.print("\t"); Serial.print(DTmoy);
  Serial.print("\t"); Serial.print(DTmax);
  Serial.print("\t"); Serial.print(DTmin);
  Serial.println();

  passage++;
}

float ValMax(float *tab,int taille){
  float maxi = tab[0];
  for (int i=1;i<taille;i++)maxi=max(maxi,tab[i]);
  return maxi;
}
float ValMin(float *tab,int taille){
  float mini = tab[0];
  for (int i=1;i<taille;i++)mini=min(mini,tab[i]);
  return mini;
}
long int ValMax(long int *tab,int taille){
  long int maxi = tab[0];
  for (int i=1;i<taille;i++)maxi=max(maxi,tab[i]);
  return maxi;
}
long int ValMin(long int *tab,int taille){
  long int mini = tab[0];
  for (int i=1;i<taille;i++)mini=min(mini,tab[i]);
  return mini;
}

Here is the result I got. 
The real distance between the DW1000 modules is 210 cm.
I do the test in my office...

[image]

As we can see, it's not really accurate. Results are not stables.
I suppose the 200ms (or 300ms once) occures when a packet is corrupted...

Did you already got better results ? If yes, what config did you used ?
I'll try to do other test but if you have suggestion, let me know !

Thanks

Umair-Anwar, 2016-12-26:
@thotro @ankitbhatia @leosayous21 
Hello all of you,
Can you please tell me, how do you calculate Rx dBm ?

sarwadenj, 2017-03-13:
Hi,
I'm running ranging_anchor and ranging tag programs I'm putting some Serial.print() function's I'm not getting any range. I'm running 2 anchors and 1 tag .Connection seems correct.

what should i do?

else if (msgId == RANGE)

DW1000-arduino-ranging-anchor

DW1000 initialized ...
Committed configuration ...
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 0A, Short Address: 01
Device mode: Data rate: 110 kb/s, PRF: 16 MHz, Preamble: 2048 symbols (code #4), Channel: #5
sent
received
IN transmitPollAck
received poll
sent
sent poll_ack
received
received unexpected message
sent
received
received unexpected message
sent
received
received unexpected message
sent
sent
received
IN transmitPollAck

received poll
sent
sent poll_ack

//////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
In TAG it not going inside
else if (msgId == RANGE_REPORT) function

DW1000-arduino-ranging-tag

DW1000 initialized ...
Committed configuration ...
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 0A, Short Address: 02
Device mode: Data rate: 110 kb/s, PRF: 16 MHz, Preamble: 2048 symbols (code #4), Channel: #5
Sent POLL @
Received wrong message # 255
Sent POLL @
Received wrong message # 255
Sent POLL @
Sent POLL @
Sent POLL @
Sent POLL @
Received wrong message # 0
Sent POLL @
Received wrong message # 0

Rotzbua, 2017-03-13:
@sarwadenj Please stop spamming, you have already opened a issue #174

VVHack, 2017-08-01:
Hi, first of all I would like to thank all of you and especially Mr Thomas Trojer for your hard work, this is a wonderful library!  I have been trying to develop a two-way ranging scheme on the DWUSB from Ciholas in the bridge mode(in which you can program the device yourself).  The coding is in Python and I have been trying to adapt this great library.  So far what I have is this:
The range I get for let's say when the devices are 1m apart is sometimes something like 15 m or sometimes it is in the negative like -1 m which is wrong but when I move the devices apart, the change in distance seems to reflect correctly which means that if it was initially 15m and then I increase the distance by 1m, it shows to be 16m and so on.  Would you have any advice based on your experience with this chip on what I could do in order to solve this issue?

PS: I have tried various antenna delays and that does not seem to solve the issue

Maxai2, 2017-09-14:
@thotro  @ankitbhatia  @jonathand1  @leosayous21  @MarioRuz 
Hello! Can anybody tell me how connect, decawave DW1000 TAG to Afrduino Uno? Which pins and which scetch give me correct ranging...
Thank u in advance!

MGasztold, 2018-01-15:
Hello,

I have exactly the same issue as @VVHack . Do anybody have any idea on how to approach this? Thanks.

wql7013, 2018-04-24:
I found the error may be dependent on the incident power rather than the range. If you change the range but remain the incident power, the error may not be changed. I think it may be caused by the delay of CMOS circuit which constitute the LNA, mixer and detector. So stable incident power, stable error. If an adaptive transmit power can be adopted the amelioration of the error may be achieved.

wql7013, 2018-04-24:
I built a wired-synchronization TDOA board with three DW1000 driven by the same oscillator. I found the errors of the two TDOAs were independent, meaning that the error may be mainly caused by the receiver rather than the transmitter or surroundings.

MariannaP95, 2018-07-09:
Hi everyone, 

First of all thank you for your amazing work! I am new to Github and using the DWM1000 chip and I have a question regarding the accuracy. I am using the DWM1000Ranging examples to measure the distance between a tag and an anchor for a university project. However, I found that for distances between 1 m - 20 m, the distance measurements is about 75 cm larger than the actual distance. Then after 20 m it starts to increase to around 1 m. I only carried out measurements up to 35 m indoors. I didn't do any calibration yet.  In issue #155 you mentioned that changing the antenna delay might help. I was wondering if my results show something that is expected and whether you know what it is caused from?  

Thank you for your help!!

[image]

Umair-Anwar, 2018-07-09:
Hi,

If your error is always one-sided, means if it always calculates
greater/lesser than actual then do change the delay until you get better
results. It is hit and trial method. will only if the error is one-sided.

In other cases, if the error is random (sometime lesser, sometimes greater)
then you need to calibrate the Rx and Tx antenna delays according to dw1000
calibration process. for this, you need more than one modules.

Regards
*Profound Regards,*
*Umair*

MariannaP95, 2018-07-11:
Thanks for getting back to me! My error is one-sided so it must be the antenna delay. What is the reasoning behind that though?
I'm also unsure about is the fact that the error increases with distance. It could also be that the error increases while the received power decreases (which is a consequence of increasing the distance). Do you have any thoughts on that?

Umair-Anwar, 2018-07-11:
The major reason is envioroment, envioroment variables like temprature and humidity effects the performance of any antenna. So I think the delay valuea are given according to the testing envioment. You need to calibrate to your sorroudings.

Yes, you are accuracy is decreased with the increase in distance. I do not the what is the reason but it also happened with me.

Regards 
~Umair

wql7013, 2018-07-14:
@Umair-Anwar It seems to be caused by the receive power. Maybe different receive power leads to different delay in the circuit.

Elon-Wang, 2018-10-09:
Hi everyone, 
I appreciate all the work you have already done, it's really amazing.  I have troubled by the below problem for a every long time.

I have 4 dw1000 modules connecting to Arduino Uno and I set three of them as anchor one as tag ,using the DW1000Ranging_ANCHOR/TAG examples. However,there are always a constant shift in every distance ranging, and the shift between any pair of tag and anchor are different, so I try to change the antenna delay by modify the DW1000.commitConfiguration function and change the antenna delay of 
every anchor to calibrate the distance between every pair of tag and anchor. 

This really improve the result I get, by changing the antenna delay,I can erase the distance shift. However, when I connect the tag with three anchor at one time. Only one anchor get real distance value, but the other two anchors get worse ranging result. 

I doubt it was the crystal frequency but I'm not sure, cause when I print the ranging report out, It shows Treply1 is bigger than Tround1, it's really confusing, Can anyone help me with my problem?
Thanks a lot for your help!!!

jivido, 2020-08-20:
- __

effendev, 2020-08-25:
> Hi!
> I did others tests ! I created my own pcb because of this problem of 3.3v for me... Also, it allows me to put my own regulator which has a limit of 800mA. I put a ground under the module but it [...]

Hei @leosayous21 , can you share your eagle file if you still have it? It would really be helpful if you can. 

So, I did a Sender and Receive test using Arduino Uno and Nano as the mcu. I can get a stable reading if I use Uno either for Sending or Receiving. But when I use Nano, the Sender function is working correctly but when I uploaded the Receiver code, the Nano started to shutdown and I couldn't open the serial monitor. I don't know whether the problem lies within the hardware/firmware compatibility betwen Thotro's code with Nano or is it the hardware fault since Nano can only supply 50 mA maximum current in its' pins. 

Anyway, really hope you can share your design file. Thank you. 

Regards,
Arifin.