thotro/arduino-dw1000 #8: API and reference example for (automatic) calibration of chips

thotro, 2015-07-01:
Iterative test that takes a predefined known range and adjusts the chips configuration until this range is actually computed.

greymfm, 2015-09-04:
That would be really nice - Not only to optimize range but also to get the modules working. Some may not work (actually all of mine) using the default library parameters (https://github.com/thotro/arduino-dw1000/issues/33)

thotro, 2015-09-17:
I think I can manage to implement this in about a week or so - it would be really convenient to have this! Will keep you posted.

belforti, 2016-10-11:
Hi,

was this ever implemented? I have a very hard time calibrating 3 modules. Even when I calibrate one of them, if I reset the system, the calibration is not precise any longer.

Thanks,
Gabe

berkerbatur, 2016-12-05:
Hi,

Is it the value of 

> 16384

 in DW1000Class::commitConfiguration() method, that we need to calibrate for antenna delay?

What are the roadmap items for this issue? something like followings? 
- new commitConfiguration for dynamic antenna delay commit
- calibration protocol between anchors (any need to include tags to this protocol?)
- put anchors known range for calibration, start calibration then start ranging

Best regards,
Berker

abencomo, 2016-12-27:
@berkerbatur , I'm afraid this project is not longer active.

maxicastle, 2016-12-27:
R.I.P.

jmta, 2017-01-29:
Hi all, I'm currently working on this project and am looking to try and do just as thotro said above. If people are interested in this please comment and also mention any other issues, got a few other people also working on the project with me and would be awesome to be able to share what we get working with you guys to implement into your projects. 

For reference,
We have created new PCBS as we were struggling to use the provided files, these have been made to run off simple 1s batteries (such as those used in RC toys, quadcopters etc)
We are nearly have a trilateration system working using 3 nodes based in c#. 
I am working towards a calibration system ( I suspect this will be the most interesting to you all)

winterscar, 2017-03-08:
Hi, I'm very interested in getting the calibration up and running. I too am setting up trilateration in c#, so if you have any issues with that, let me know and I can take a look.

rbrainard, 2017-03-15:
Ya, I am super interested as well!

DrummyFloyd, 2017-03-29:
where we can fin your work ? =D 'cause i'm very interested!!!!!! 
especially with dat ***** calibration

Alenux55, 2017-08-01:
Has this been investigated further? I realize that the project is no longer active, but this would be a great feature.

I've started implementing this, but my lack of experience with C and interrupts has so far been holding me back. However, I'm aiming to implement the method described [Section 8.3.1](http://thetoolchain.com/mirror/dw1000/dw1000_user_manual_v2.05.pdf) of the user manual. That is:

> A method to calibrate combined receiver and transmitter antenna delay is as follows:
> 1. Place the transmitter and receiver at the recommended distance apart as per Table 53 or at a
> shorter distance but ensuring that the receiver power level is as per Table 52
> 2. Using a two way ranging scheme (e.g. “DecaRanging” available from DecaWave), perform 1000
> ranges.
> 3. Adjust antenna delay until the average measured range matches the distance chosen in step 1 as
> closely as possible.

Further, a note in that section specifies that the transmitter antenna delay can be set to zero and the receiver can have the aggregate antenna delay for systems with comparable antenna delays. Since the DWM1000 modules are essentially identical, I think it's safe to assume we can do this. So I'm setting the anchor antenna delay to zero and varying the tag antenna delay.

To implement this ability, commitConfiguration of DW1000.cpp needs to be updated to either take the antenna delay value as an argument, or a new function needs to be made to set the antenna delay; I've currently chosen the former.

Alenux55, 2017-08-04:
FYI, I've forked this project and added a file **DW1000Parser** to parse incoming serial commands. My plan is to have the antenna delay configurable through the serial port, then do the actual calibration routine in Python.

werty37, 2017-10-05:
@Alenux55 Were you able to crack this?

Alenux55, 2017-10-05:
@werty37 IIRC I made a rudimentary calibration algorithm, but need to take a look at my old work laptop to remember what I did (this was for an old job). Might not be able to post the source code, only the relevant parts

Alenux55, 2017-10-08:
So I can't upload the Python code I used to do this, but the basic idea was that the distance values that the DW1000 returns is linear, but typically scaled by some factor. For example, see the plot of calculated distance (that returned by the DW1000) vs. actual distance (real world distance) below:
[image]
To make this plot, I took 100 samples at each point from 5 cm away to 100 cm away in 5 cm steps; this is the green curve. The blue curve is the actual distance between the anchor and tag. The dotted black lines are +/- 5 cm from the actual distance. Using the actual distance and the calculated distance, I was able to apply a scaling factor to the data to correct it, as you can see below:
[image]
Once that scaling factor is found, it can be applied to all subsequent data to get a more accurate distance measurement.

Main things to keep in mind here is that I only got this running between one tag and one anchor as that was all I needed for the project I was working on. Theoretically, it could be extended to multiple tags, but I never got that far.

werty37, 2017-10-12:
@Alenux55 Thank you for the detailed explanation.

jonathanrjpereira, 2020-10-14:
@Alenux55 Would I be able to implement this for a system with multiple anchors? 

For example: Consider I have a system of 4 anchors and I wish to create a Euclidean Distance Matrix (EDM) for the same anchors.

  | 1 | 2 | 3 | 4
-- | -- | -- | -- | --
1 | 0 | d12 | d13 | d14
2 | d12 | 0 | d23 | d24
3 | d13 | d23 | 0 | d34
4 | d14 | d24 | d34 | 0

Hence for this system, my requirement is to measure 6 inter-anchor distances: d12, d13, d14, d23, d24, d34. 
Would I need to find the scaling factor for each of these anchor combinations i.e 6 different scaling factors?

Alenux55, 2020-10-14:
@jonathanrjpereira unfortunately I haven't touched this project in a long time and I never ended up testing it with more than one anchor, and I only ever used an anchor-tag pair. Having said that, it would make sense that you'd need to individually scale the inter-anchor distances as each pair of anchors would have slightly different offset.

Sorry I can't add much more to help.