thotro/arduino-dw1000 #11: Question: Can this be used to create tags for the DW TREK Evalutation Kit

nachio, 2015-07-06:
You're doing amazing work on this library and seems that you're progressing very well. 
I've been snooping around a bit and have a few questions (some might not be directly related to the library but still very useful to have answered). that I hope you could answer if time permits.

1. We are currently testing with their evaluation kit, but would love to have a bunch of smaller custom tags. Your library is ideally suited to create our own...so could your library be used to create tags that work with DecaWave's TREK1000 evaluation kit? Is there some special configuration needed so the anchors can detect / identify these tags?
2. Would love to track some high speed things. Ideally would love to have location updates at least 50 - 100 x per second. Is this at all possible with about 30 tags at once? What is the max. update rate currently supported for localization and on what does it depends? 

Thanks!

thotro, 2015-07-06:
Hi there,

Many thanks!

Regarding your questions:

 1. Well, without having dug into the code of the eval kit so far, I would say yes, it should be possible to have custom tags talking to anchor nodes of the eval kit. Certainly, a specific (not necessarily "special" :)) configuration will be required and most importantly, the message format that is proposed by the eval kit needs to be implemented (the reference example here uses a simple custom one). As the Arduino driver already supports all operation modes and the message format of Decawave won't be too complicated (I guess), this is doable. 

 2. Unfortunately I've done too few benchmarks and experiments to tell you if this is possible. My guess is that for such an update frequency a TDOA scheme might be better than TOA (since less messages need to be transmitted). Some preliminary numbers though: My Arduino Pro Minis (16 MHz clock) currently do about 15 updates per second. I still need to benchmark how these numbers improve with "better" code (e.g. using another scheme, trying if interleaving update messages work, or employing redundant anchors for cyclic range computations, ... and so on), but also how it behaves with many more tags in use.

What update frequency is claimed to be possible with the eval kit and what are your observations?

Hope it helps for now; will get back to you soon with some more benchmarks on ranging.

Cheers,
Thomas

Kris523, 2015-07-13:
Question:
- Is it possible that you could publish the Gerber files for the adapter board, I could create my own, but it wouldn't emulate your results as well.

I love what your doing here, 
Thanks alot!

thotro, 2015-07-14:
@Kris523 Many thanks! The Gerber files are now contained in the "adapterBoard" folder. Hope it works!

Cheers!

leosayous21, 2015-07-16:
@thotro many thanks for the gerber files ! I just send them to pcbway and i will be able to try your code quickly :-)

bigredbee, 2015-07-16:
I've loaded these onto oshpark.com  Cost is $7.00 for three boards.  If
there is interest, I can make them publicly available.

closed 2017-04-24 by Rotzbua