thotro/arduino-dw1000 #15: Is there any reason why this would not work with an Arduino Yun?

te316, 2015-07-22:
In order to communicate with an external "thing" like a robot, I am conceptually finding a need to have a master anchor tag with wifi capabilities.

So, is there any reason why one of the tags can't be an Arduino Yun (plus the dwm1000) while the others remain a mini pro?

I guess my question is can pretty much all of the arduinos be used with your code and the pin layout that you have?

Thanks

thotro, 2015-07-22:
I did not try anything beyond the Mini Pro, but since I don't use any architecture specific code, I would expect the Arduino platform to be able to manage it. Hence, I see no reason against a Yún.

Regarding pin layouts: Different Arduinos may have different features offered at different pins, therefore you will at least have to align the pin settings in the source codes with your actual wirings (it boils down to a chip select and optional reset line and the correct wiring of SPI related pins). The pin layout of the adapter board (in case you want to use it for simplified prototyping) is valid for the DW1000 module and independent of the type of Arduino.

Best,
Thomas

Blue-Orb, 2015-07-25:
Hi,

I have checked the code with Arduino Uno and Yun; it works pretty well.

Cheers!!

thotro, 2015-07-25:
Thanks for the info - good to hear! :-)
Cheers!

closed 2015-08-04 by thotro