thotro/arduino-dw1000 #16: Synchronizing clocks among multiple chips?

jogando, 2015-07-22:
Hi Thomas,
I'm really impressed with the work you've done :), thanks!
I wanted to ask you if you know if there's a way of syncing the DWM1000 clock externally.
I know that the DW1000 has the SYNC pin, but i don't think this is enabled in the module DWM1000.

I want to have the Anchors' clock synchronised.

Any ideas?

Thanks!
Julian

thotro, 2015-07-23:
Hi Julian,
many thanks! :-)

According to the data sheet of the module, the external sync functionality cannot be used with the module, just with DW1000 (don't know why). It seems that you need pins EXTCLK and SYNC and only SYNC is offered on the module (as GPIO7).

Hence, in case you start using plain DW1000s to be able to synchronize clocks, just let me know and I will do my best to provide a simple API for this.

Just in case you haven't seen this yet: [APS0007: Wired Sync RTLS ](http://www.decawave.com/support/download/file/nojs/451) looks like a good starting point.

Cheers,
Thomas

curlyz, 2016-11-29:
@thotro Is that mean DWM1000 can not do TDoA???
In APS007 they mentioned 2 way, with and without SYNC.
And is it mean that more works with the code.

closed 2017-04-26 by Rotzbua

AlexisTM, 2018-03-27:
Reverse TDOA implementation: The anchors are messaging and the tags are passive (only listen). This means you can have as many tags as you want.

This is done by the Loco positioning system by Bitcraze for CrazyFlies, using the DWM1000.

Anchor software:

anchor.c: https://github.com/bitcraze/lps-node-firmware/blob/master/src/uwb_tdoa_anchor2.c
Tag software:

tag.c: https://github.com/bitcraze/crazyflie-firmware/blob/a4dc79f53af771788c8fc67275b07b29ccc7a3a1/src/deck/drivers/src/lpsTdoaTag.c
tag.h: https://github.com/bitcraze/crazyflie-firmware/blob/a4dc79f53af771788c8fc67275b07b29ccc7a3a1/src/deck/drivers/interface/lpsTdoaTag.h
config: https://github.com/bitcraze/crazyflie-firmware/blob/a4dc79f53af771788c8fc67275b07b29ccc7a3a1/src/deck/drivers/src/locodeck.c