thotro/arduino-dw1000 #19: maximum ranging test (indoor)

leosayous21, 2015-07-31:
Hi !
I tried a test to verify the maximum indoor distance we can get. I had hardly 12m in live view... Also i can't go through a wall ! 
mode: DW1000.MODE_LONGDATA_FAST_ACCURACY
I noticed that if i use a replyTime near the limit (1.9ms here), the max distance is shorter !  I had 7m with 1.9ms and i add the 12m using a 3ms replyTime
Cheers,
Léopold

ankitbhatia, 2015-07-31:
The fast mode is supposed to be short range. I was able to get about 40m in the 110Kbps mode. Which is still troubling as the datasheet advertises 300m line of sight.

Kris523, 2015-09-29:
Hello!
I have finally gotten my system up and running and am currently (slowly) entering the code. 
I too am troubled by the small range (10m in my case). I have been looking into the settings, and the current ones should be giving a much larger range. This lead me to the transmission power of the module. Datasheet states the most common is -41.3 dBm/MHz while I noticed in the code (and in practice) that the module only transmits at about -61dB. The receiver looses signal around -95dB which appears typical. Anyone have any alternative views on this topic?

Also: Currently, for me only DWRanging_ANCHOR and DWRanging_TAG examples are working correctly. Is this the same with all of you?

Thanks everyone!

pascalbros, 2016-06-23:
I bring this discussion back to life because i'm having some troubles to get a nice distance, with the TAG-ANCHOR sketches (without changes), it can reach maximum 8-9 meters LOS and just 4-5 meters in NLOS (behind a wall).

And then I started to play with the TX power and I found out with `MODE_LONGDATA_RANGE_LOWPOWER` and overriding the TX_POWER register with `writeValueToBytes(txpower, 0x0E08281FL, LEN_TX_POWER);` it gives 12 meters in NLOS and more than 20 meters in LOS (i will try in a bigger space).
Any suggestion on the topic? Thanks. ;)

MikeS159, 2016-06-24:
Are you using the DWM1000 module? Looking at the data sheet for their antenna, they aren't that good. I've had about 18m of LOS range after a bit of experimenting, but NLOS is very inconsistent. I've had much better results using the TREK demo kit, the antennas have much better gain (I've got >50m LOS and need a larger area to test in :) )

pascalbros, 2016-06-24:
Hi @MikeS159 
Yes DWM1000.

So ok, you just confirmed the inconsistency of that product.
I was thinking about [Radino DW1000](http://shop.in-circuit.de/product_info.php?products_id=186&language=en), it should have a better antenna and if it's not enough, it's detachable.
Already checked and the source code included is nothing more than this library with nRST, IRQ and SS changed, so it will work.

If I buy this, do you think some code in the calibration should be changed? It looks the same of the TREK demo kit.

Let me know, Pascal.

MikeS159, 2016-06-24:
@pascalbros 
I've not seen that device before, but it looks very interesting and will save you from having to do the complex designed to move a DW1000 chip onto your own board. Its probably worth getting to try (they aren't that expensive compared with the TREK kit).

The only issue I have is that they claim it comes with software but the link is dead. That sort of thing makes me worry about how long anything they sell will be supported. Since it is pin compatible with Arduinos you should just be able to continue using the latest version of this project.

pascalbros, 2016-06-24:
Why "dead"? It works [http://wiki.in-circuit.de/index.php5?title=radino_Library#Download](http://wiki.in-circuit.de/index.php5?title=radino_Library#Download)
You'll find the code in /libraries/radino32

The cost with the USB stick, antenna and enclosure is about 75$, not too expensive compared to the TREK kit but not too cheap also.

I'll see, if I buy it, i'll let you know.

MikeS159, 2016-06-24:
I didn't look at the links closely enough.
I didn't realise there was an extra cost for antennas, they are just PCB's with connectors soldered on, you could probably got those cheaper elsewhere or even made pretty easily.

Saflight, 2017-04-18:
Hi,

I bring this issue back to life as I'm experiencing the same trouble on my first tests. Can anyone confirm what could be the maximum range with relatively good precision. I need at least >50m however I'm not able to go further than 10 (aprox.) for the moment.

Any advice suggestions based on experience?

Thank you mates

MikeS159, 2017-04-18:
I haven't worked with these for some time. but form what I remember 50m was about the limit. For the DWM1000 modules I was getting about 20m LOS (line of sight) and with the Trek kit (which includes the larger patch antennas) I got ~50m LOS.

To get a greater distance you can up the power, there are several application notes which give some advice.

APR001: UWB Worldwide Regulations - http://www.decawave.com/support/download/file/nojs/726
APS004: Increasing the Range of DW1000 Using LNA - http://www.decawave.com/support/download/file/nojs/860
APS006 Channel Effects on Communications Range - http://www.decawave.com/support/download/file/nojs/450
APS017: Max Range in DW1000 Systems - http://www.decawave.com/support/download/file/nojs/714

Generally I had trouble getting communication working through walls, and it did reduce the accuracy of the distance measurements quite significantly. If you application is indoors I would consider having groups of anchor devices in each room.

Saflight, 2017-04-18:
Thank you for your extremely useful and quick response, I will check all those documents as they need some time for reading and come back again with the doubts.

Did you anyhow successfully get to implement any of those solutions suggested by decawave on your DWM1000 assembly before switching to TREK?

MikeS159, 2017-04-19:
There is support in the library for a mode which increase the maximum power in the highest data rate mode (6.8MB/s?) due to the face that the average power remains the same. I don't remember it helping in any significant way.

How have you implemented your circuitry? I had issues with an initial board prototype and ranges because the power delivery wasn't that good. Ensure you are providing adequate power from your 3.3V supply, have a clear area around the antenna (i.e. no copper or components on the underside) and decoupling close to the chip.

Saflight, 2017-04-19:
I'm using Wayne's assembly --> https://sites.google.com/site/wayneholder/uwb-ranging-with-the-decawave-dwm1000---part-ii

I took the boards to be soldered at professional labs, so I think that bad welding should not be the root cause=S. the Supply I'm using is through USB (as in the Wayne's tutorial)

Should I get better results with that? How was your assembly?

Saflight, 2017-04-19:
I know that it is possible to play with some parameters such as number of channels, bandwith, TX/RX power, datarate and the others which may impact on maximum range.

Would you have any clue on how to modify those parameters?

Thank you in advance

Saflight, 2017-04-20:
Any suggestion?

MikeS159, 2017-04-22:
Your board shouldn't be the problem then. I haven't worked on the DW1000 in about 7 months so I don't really remember how to configure most stuff, but some is configured when you initialize the device.

e.g. `DW1000Ranging.startAsAnchor("82:17:5B:D5:A9:9A:E2:9C",DW1000.MODE_LONGDATA_RANGE_ACCURACY);`

Sadly the DWM1000 just doesn't have great range. If you really need better range you need to look at making a board with an antenna connector like the ones in the TREK kits. Decawave also have application notes on selecting antennas.

Saflight, 2017-04-22:
Yes, I think that it would be a definitive solution. In the other hand I was able to find the different modes that the DWM1000 can use getting significant improvements so for the moment I will keep trying to get the maximum results from it before switching to TREK.