thotro/arduino-dw1000 #23: ranging frame format with standard MAC layer

leosayous21, 2015-08-03:
Hi @thotro,
I want to try to implement the MAC layer according to the IEEE 802.15.4 standard.
I don't really understand the current frame format for a message.
When we send a message, we call the setData which writes bytes on TX_BUFFER register. 
So if i understand well, for now we fill the content of "Data" that we can see on the right of the Figure 31 below ?
[image]
That mean when we send a POLL in setting data[0]=POLL we set the first byte of this portion of frame ?

[image]
So now according to the figure 34, we need to create this frame using software in order to make the data variable following the structure. 

Cheers,
Léopold

EDIT: Finally I found the response for my question... It's in the 11.4 section :-) "The DW1000 transmits data from the TX_BUFFER in a frame with data length as specified in the TFLEN and TFLE fields of the Register file: 0x08 – Transmit Frame Control, inserting the 2-octet FCS as the last two octets of the data payload. The DW1000 will not do any other MAC level transmit processing. So, it is up to the host system software to prepare the correctly formatted frame conforming to the IEEE 802.15.4 standard MAC if this is required."
But I let the message, i think it can be interesting to understand how frame works in the radio protocole !

thotro, 2015-08-03:
Yes, that's the plan for MAC: doing it by hand! :-) As you wrote, currently we simply set the TX data buffer, decide on data length according to frame check settings and that's it.

Only a subset of MAC messages needs to be implemented and in general I think its pretty straight forward - luckily the format is not really complex. The driver's "setData()" function should be overloaded to also take an instance of a "MACMessage" (or the like); more specifically one of its concrete sub-types. 

In the end we are able to do auto-acknowledgement and frame filtering with it - two really great features.

Just let me know what I can do to assist: First draft, actual programming, code review, testing or maybe nothing at all? ;-)

leosayous21, 2015-08-04:
Here for the general ranging frame format we need to implement too !
[image]

thotro, 2015-08-04:
Yes we could do that. For the payload portion of MAC we can follow this format (the rest is plain MAC), although it is not part of this figure and we need to process it on the host-system anyway. 

Another viewpoint of this issue: I would be curious to see if the payload of these messages corresponds to the messages implemented in the TREK and eval board of Decawave; since there are already requests here on the repo to be capable to mix code/device "vendors" in the future.

leosayous21, 2015-08-04:
yes it’s to meet the IEEE 802.15.4 standard ! « […] The general message format, shown at the top of Figure 3, is the IEEE 802.15.4 standard encoding for a data frame.  […] ». It’s in the « aps013_dw1000_and_two_way_ranging.pdf » page 6 :-)