thotro/arduino-dw1000 #24: Sent Handler is not fired after sending

athlan, 2015-08-03:
I have run example code BasicSender in Arduino Mega 2560 R3. The output from Serial Monitor is as follows:

```
### DW1000-arduino-sender-test ###
DW1000 initialized ...
Committed configuration ...
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: EUI: 0:0:0:0:255, OUI: 255:255:255
Network ID & Device Address: PAN: 10, Short Address: 5
Device mode: Data rate: 110 kb/s, PRF: 16 MHz, Preamble: 2048 symbols (code #4), Channel: #5
Transmitting packet ... #0
```

Unfortunately, after sending packet, the callback is not fired, which causes, that sentAck is not set to true. Main loop is breaking every each iteration.

MarioRuz, 2015-08-03:
Same problem and I still do not know the reason (it also happens with the ranging example). Try to disconnect and connect one of the arduinos, by doing this it worked to me. I am wondering if this problem is due to the limited current the 3.3 V output can provide. Cheers.

leosayous21, 2015-08-03:
How do you supply your module ? I had the same problem before when i used the arduino 3.3v regulator …

athlan, 2015-08-04:
Guys, thank you for quick response. I am supplying DWM1000 directly from Arduino's 3.3V output.

athlan, 2015-08-04:
Guys, thank you for quick response. I am supplying DWM1000 directly from Arduino's 3.3V output. But Auduino is supplied from USB itself.

thotro, 2015-08-04:
Hi there!

(From the datasheet) Arduino's internal regulator only produces max. 50mA which is not enough to drive the TX current peaks of around 140mA (from the datasheet) of DW1000. As far as I remember @leosayous21 has a post somewhere around here on using an external regulator and further his adapter board may be a good reference - see [here](https://github.com/thotro/arduino-dw1000/tree/master/adapterBoard/dwm1000%20arduino%20standalone). 

Please also note that logical high on your I/O pins is about 5V; too much for the DW1000.

Hope it helps!
Cheers!

closed 2015-08-10 by thotro

abencomo, 2016-11-02:
@thotro, I'm using Wayne's [adapter](https://sites.google.com/site/wayneholder/uwb-ranging-with-the-decawave-dwm1000---part-ii) so I'm supplying 5V.  However, when I connect it to an [USB "Power Bank" Charger](https://www.amazon.com/gp/product/B01G1XH46M/ref=ox_sc_sfl_title_9?ie=UTF8&psc=1&smid=A2KUZVNQ9LP7N9), it only draws 40mA.  How should I power the module so it can be supplied it with 140mA as you indicated above?

[image]

wholder, 2016-11-03:
Hi, this is Wayne.  40ma is about what I'd expect the Arduino to draw by
itself,  So, I'm guess there is something wrong on the DWM1000 PCB.  Check
the voltage on the two pins marked 3.3 volts near the right lower edge of
the board (with DWM1000 at the top.)  This will show if the 3.3 regulator
is working.

Wayne

abencomo, 2016-11-03:
Hi Wayne, do you mean pin 6 or 7 highlighted below?

[image]

wholder, 2016-11-03:
No.  See attached.

Wayne
[image: Inline image 1]

abencomo, 2016-11-03:
Wayne, your attachment didn't come through...

wholder, 2016-11-03:
Try this link:

https://sites.google.com/site/wayneholder/uwb-ranging-with-the-decawave-dwm1000---part-ii/DWM1000.png

Wayne

abencomo, 2016-11-03:
Wayne, I do get 3.3V on those two pins (see picture below).

[image]

wholder, 2016-11-04:
Ok.  That's good.  At least the 3.3 regulator is working.  My other guess
is that perhaps the DWM1000 module is not completely soldered to the PCB.
I had some trouble with soldering the castellations myself and had to
reflow some of the connections to get the module working.  Did you build
more than one module?  If so, perhaps you can compare voltage levels on the
different pins.

Wayne

abencomo, 2016-11-04:
I built four(4) modules and all of them show 3.3V on those pins.  I'll try to reflow the connections again, but it is quite odd that all four are not working.  Did you have to make any code changes (set unique ids, channels, etc.)?

wholder, 2016-11-04:
It's odd that all 4 units would fail to work.  Can you post a photo of one
of the assembled boards without the Arduino mini installed so I can see how
you installed the components?  Perhaps I can spot a clue that way.  I used
a version of the software here that was posted some time ago, so something
may have changed since then.  I'm currently swamped with other tasks, but I
intend to try the latest software when I have the time.  But, until then, I
can only speculate as to the cause of your problem.  I do know that several
others have successfully built modules using my PCB, so I'm fairly
confident of the design.

abencomo, 2016-11-04:
1 and 2 were my first try.  Thanks Wayne for looking into this.

[image]
[image]
[image]
[image]
[image]

wholder, 2016-11-05:
To my eye, everything looks OK.  Since you say all 4 boards exhibit the
same symptoms, I'm staring to think it must be a software issue.  Perhaps
something has changed in the github code since I got my boards working.
I'd offer to send you my code as a test but, at the moment, I'm traveling
and don't have access to it.

To debug it yourself, I suggest you start by seeing if you can get the
BasicConnectivityTest sketch working.  I don't recall the exact values it's
supposed to print out and can't check while traveling, but perhaps someone
else here can give you the info.

Also, just to make sure, you are using 5 volt, Arduino Pro Minis on all the
boards, right?

Wayne

abencomo, 2016-11-05:
Yes, I'm using the 5V [Arduino Pro Mini](https://www.sparkfun.com/products/11113) as well as the [SparkFun Pro Micro](https://www.sparkfun.com/products/12640).  Both display the same result when run the BasicConnectivityTest sketch, as shown below.

Once you are back home, would you mind trying the latest code from Master on one of your working the modules.  If that doesn't work, then you were right and it's a software issue.  That will also means that the version of the code that you have it's gold!

[image]

SATHYALAKSHMI123456, 2019-06-22:
hello my board working but it detect only 5 meters .how can i increase above 50m .