thotro/arduino-dw1000 #31: Make the library work with teensy

leosayous21, 2015-08-14:
Here an issue in order to make the library work with teensy compiler !

fornzix, 2015-08-15:
I'd like to second this request.  For some reason it won't even compile for Teensy 3.1.  

Here's the error it's spitting out:

In file included from C:\Users\Documents\Arduino\libraries\arduino-dw1000-master\src/DW1000.h:248:0,
from BasicReceiver.ino:23:
C:\Users\Documents\Arduino\libraries\arduino-dw1000-master\src/DW1000Time.h:27:0: warning: "TIME_RES" redefined [enabled by default]
#define TIME_RES 0.000015650040064103f
^
In file included from BasicReceiver.ino:23:0:
C:\Users\Documents\Arduino\libraries\arduino-dw1000-master\src/DW1000.h:26:0: note: this is the location of the previous definition
#define TIME_RES 0.000015650040064103
^
In file included from C:\Users\Documents\Arduino\libraries\arduino-dw1000-master\src/DW1000.h:248:0,
from BasicReceiver.ino:23:
C:\Users\Documents\Arduino\libraries\arduino-dw1000-master\src/DW1000Time.h:28:0: warning: "TIME_RES_INV" redefined [enabled by default]
#define TIME_RES_INV 63897.6f
^
In file included from BasicReceiver.ino:23:0:
C:\Users\Documents\Arduino\libraries\arduino-dw1000-master\src/DW1000.h:27:0: note: this is the location of the previous definition
#define TIME_RES_INV 63897.6
^
In file included from C:\Users\Documents\Arduino\libraries\arduino-dw1000-master\src/DW1000.h:248:0,
from BasicReceiver.ino:23:
C:\Users\Documents\Arduino\libraries\arduino-dw1000-master\src/DW1000Time.h:77:31: error: 'constexpr' needed for in-class initialization of static data member 'const float DW1000Time::SECONDS' of non-integral type [-fpermissive]
static const float SECONDS = 1e6;
^
C:\Users\Documents\Arduino\libraries\arduino-dw1000-master\src/DW1000Time.h:78:36: error: 'constexpr' needed for in-class initialization of static data member 'const float DW1000Time::MILLISECONDS' of non-integral type [-fpermissive]
static const float MILLISECONDS = 1e3;
^
C:\Users\Documents\Arduino\libraries\arduino-dw1000-master\src/DW1000Time.h:79:36: error: 'constexpr' needed for in-class initialization of static data member 'const float DW1000Time::MICROSECONDS' of non-integral type [-fpermissive]
static const float MICROSECONDS = 1;
^
C:\Users\Documents\Arduino\libraries\arduino-dw1000-master\src/DW1000Time.h:80:35: error: 'constexpr' needed for in-class initialization of static data member 'const float DW1000Time::NANOSECONDS' of non-integral type [-fpermissive]
static const float NANOSECONDS = 1e-3;
^
Error compiling.

sondree, 2015-09-04:
I have been working on this. As suggested leosayous21 removing references to arduino specific "digitalWriteFast" gets the examples compiling. My fork can be found here: https://github.com/sondree/arduino-dw1000 

Do note that it is not currently working, I seem to get information back from the transceiver however I am unable to successfully transmit between two modules. With changes found in my fork you should be able to compile and use the "Connectivity example"

sondree, 2015-09-04:
@fornzix Your issue seems to be that you have to guarantee that the expression in DW1000Time.h are possible to evaluate compile time. Fixing said issue is as easy as adding "constexpr" to all the offending declarations. I remember having this issue when I first started, but my local library no longer has the offending lines. You should probably pull the latest changes.

paulerolland, 2015-12-01:
The way I fixed this issue is by adding the "f" after the floating number on the DWM1000.h file.
Everything uploads and work after removing the reference to the digitalWriteFast file except the interrupt. Anyone had the same problem? The program never goes to the interrupt. I can fork my version if someone is interested!

sondree, 2015-12-01:
@paulerolland This seems very very similar to the issues I had. Sadly I do not have a fix. As stated, I could run the module connection test examples, but any transmit/receive examples would not work (as they require interrupts)

paulerolland, 2015-12-01:
@sondree Thank you for your answer. I will try to find a way to make it work. I will update this thread if I find something!

paulerolland, 2015-12-01:
Just made it work! I chose the right mode for the interrupt pin using pinMode. Here is my fork if you want to take a look at it. I also added the option to choose the IRQ, RESET and CS pin for the Basic Receiver and Sender, the ACHOR and the TAG ino files.
I hope that can help you!

https://github.com/paulerolland/arduino-dw1000

sondree, 2015-12-01:
@paulerolland  I will test your fork tomorrow and see if I can replicate it. I really hope so, will report back.

sondree, 2015-12-02:
@paulerolland  It works, well done :) My test setup sometimes times out, does your do this?

paulerolland, 2015-12-03:
@sondree It does the same with me. I will work on this and the simple send and receive to check if I can fix it.

rbalsan1, 2016-02-14:
@paulerolland  could you please elaborate on your setup?  I'm running your fork on 2 Teensy 3.1's and trying the tag and anchor sketch and not getting any communication between the two.  On the com port for the tag I'm seeing ###tag### and on the comp port for the anchor I'm seeing ###anchor###, but nothing else.

Rotzbua, 2017-04-24:
outdated

closed 2017-04-24 by Rotzbua