thotro/arduino-dw1000 #32: Multiple anchors

leosayous21, 2015-08-20:
Hi !
I commited the first sketch for multiple anchors. I tested it with 2 anchors, but i will keep you in touch when trying with more devices ! 
Also, there is optimisation to do on replyTime: for instance the first anchor will have a reply time of 7000us but the second when will have 3*7000us ! Otherwise it doesn't work and i don't really understand why...
I will try to visualize how is the time for sending/receiving, etc. with an oscilloscope in order to understand that !
Cheers !
Léopold

rasamaya, 2015-08-23:
Hi Léopold,
Sounds great!, I am going to attempt 4 anchors and 2 devices.
Would love to hear more! I have 6 units on order.
I hope to share my findings too. 
I will have 9dof imu on each tag for orientation and hope its accurate enough and fast enough to send good positional data of the objects I will be holding in my hand. 
Good luck!

az-z, 2015-08-27:
@leosayous21 i will have multiple units all communicating to one/two anchors. Is it possible to use this library for this purpose or (this lib) only allows communication with limited amount of devices?
Full disclosure - i'm a total noob in all things arduino and dw .

leosayous21, 2015-08-29:
Hi @rasamaya  !
Sorry for the delay.. I hope you will do a good job on the library :-) When will you receive your units ?
 @az-z , For now, the only configuration which is programmed here is: 1 tag and multiple anchors.
That is: all anchors are waiting for a signal from 1 tag. And when they have talked to the tag and calculated the distance, they send back to the tag the distance they have.

There is still work to do on the library: multiple tags and to test more than 3 anchors. Also, it will be good to have the possibility to send the distance back to an other device (which can be called the collector)

Good luck for you :-)
Léopold

az-z, 2015-08-29:
@leosayous21 :
>...And when they have talked to the tag and calculated the distance, they send back to the tag the distance they have.
does it mean that they can't talk to each other? 

another question - can tags talk to a different anchor after some time ? basically do i have an ability to communicate to the anchor in the  room 2, after i'm done "talking" with the anchor in the room 1?

az-z, 2015-08-29:
@leosayous21 
will this lib work with arduino due? or i should be looking at the different board?

leosayous21, 2015-08-29:
Yes normally you can do what ever you want because there is the mac filtering. That mean you can select the device you want. But you need to enter deep in the code for that.. (there is not an easy API for now). 
Basically here, in the last commit, you can have 1 tag and some anchors in your house. When the anchors will be next to the tag, it will discover it and start ranging. when you are too far from an anchor, the tag delete the anchor from its "list of anchors" (2s timeout).

leosayous21, 2015-08-29:
I've just tried to compile the code with an arduino due but there is a lot of error (It seems to come from the digitalWriteFast i used to trigger the SS pin during writeBytes and readBytes. We need to add a #ifdefine here i think because digitalWriteFast is not necessary for arduino due )

az-z, 2015-08-29:
what arduino do you recommend to get ? i will need to do some triangulation and communication over the wifi and maybe gps too. but this is a long term goal ( hopefully 6-9 months i'll be able to achieve that). @bigredbee shared with me he is using pro micro.

jogando, 2015-09-12:
Hi Léopold!
Could you please explain how to make things work with multiple anchors?
I tried to do it and this is what i got:
Anchor 1: works fine
Anchor 2:
blink; 1 device added ! ->  short:69DD
delete inactive device: 69DD

If i disconnect Anchor 1, then Anchor 2 starts working.

Thanks a lot for your excellent effort :)

Regards,
Julian

kshitijdadhekar, 2016-03-15:
@jogando Did you have any luck with one Tag and Multiple anchors? I am facing the same problem

headcrabned, 2016-03-17:
I'm getting a similar behavior with more than 1-2 anchors at once. It works fine with a tag and one anchor, and mostly fine with 2 anchors (I start to see some big negative distance estimates), and I haven't gotten a tag to detect 3 anchors at once.

They all have different MAC and short addresses, and I'm not sure what the problem is. I wouldn't mind scheduling the tag to initiate two-way ranging to a short list of fixed tag addresses (I only have 4 anchors and their addresses are fixed), but I'm trying to figure that out too. I believe that's how the EVK1000 evaluation kit operated.

kshitijdadhekar, 2016-03-19:
@headcrabned  Did you supply the TAG with external power?

headcrabned, 2016-03-19:
The tags and anchors did have external power, from a 3.3V regulator. I checked the output on a scope, and it stays above 3.25V.

Rotzbua, 2017-04-26:
closed in favour of #181

closed 2017-04-26 by Rotzbua

KeyYD, 2017-05-13:
Hi, @kshitijdadhekar 
you can find a solve of your problem in #179