thotro/arduino-dw1000 #33: no interrupt at receiver (Arduino Nano)

greymfm, 2015-08-26:
I did connect 2x DWM1000 modules each to one Arduino Nano (5v) using 5v<->3.3v level shifting (details below). 

The 'BasicSender' works (in setup,  'char msg[1024]' had to be reduced to 256 bytes) on both my Nanos:  the device ID is read correctly, and send interrupts appear. 

However, if I try 'BasicReceiver' on one of my modules, on the receiver the DWM1000 VDD voltage goes down from 3.3v to 3.0v, and no interrupt appears. The device ID is read correctly, but when measured with oscilloscope at DWM1000, no receive interrupts appear. However, they appear on that module when configured as sender.

Tried both modules, different DWM modes and delta times without success. Any ideas? I have ordered Arduino Pro Mini 3.3v to retry this. Thx, Alexander

Circuit details:
DWM1000 IRQ  (via 12k pull-down) -- Arduino Nano D2 input 
DWM1000 RSTn (via level shifter) -- Arduino Nano D9 output 
DWM1000 SPICLK (via level shifter) -- Arduino Nano D13 output
DWM1000 SPIMISO (via level shifter) -- Arduino Nano MISO D12 input
DWM1000 SPIMOSI (via level shifter) -- Arduino Nano MOSI D11 output 
DWM1000 SPISS (via level shifter) -- Arduino Nano SS D10 output

level shifting circuit (jy-mcu-converter_5_3_bidir)
http://www.openhardware.pe/store/image/cache/data/components/diagram_jy-mcu-converter_5_3-500x500.jpg

greymfm, 2015-09-01:
Tested again using two STM32F103 and Arduino STM32 Libs (http://grauonline.de/wordpress/?page_id=1004). 
I can receive a single (error) packet after sending 2000 packets. Any ideas?
[image]

bigredbee, 2015-09-01:
I'm using the Arduino Pro Micro.  For reasons that I don't completely
understand, I had to move the interrupt to a different input pin.(leaving
the code unchanged)  Perhaps the Nano has the same issue ?

Greg

greymfm, 2015-09-03:
I added a counter to the interrupt handler function. Then I triggered the interrupt pin manually by pulling it to HIGH (3.3v) - Then the interrupt handler is executed (counter increases).

So it cannot be the pin - it must be that the receiver is actually not generating an interrupt. Oscilloscope measurement confirms this.

Any further ideas? I have tried 2 other modules now. Same problem. I feel like an idiot :/

greymfm, 2015-09-05:
Problem solved!  

1. I had to use this mode:  DW1000.MODE_LONGDATA_RANGE_ACCURACY
2. I had to reduce the number of bytes sent:  msg = "42"
3. Also, I had to reduce the serial console output: just count transferred/received packets via variables, and only output these counters to serial console every second

And then it's working :-)

closed 2015-09-05 by greymfm

leosayous21, 2015-09-05:
Hi ! 
Did you also tried to change the delay_reply_time to much higher value ?

greymfm, 2015-09-05:
In this issue I was using BasicReciver and BasicSender code examples. I was not using Ranging in my first test I did here  (btw, I did successfully do ranging today, yet some other issue there: it hangs after a while, will try to change delay_reply_time there - I have the feeling some 'deadlock' can happen in the code somehow - the actual BasicReceiver/BasicSender example does not hang at all ...).

greymfm, 2015-09-05:
PS: I have opened an issue for the DW1000Ranging_TAG+ANCHOR deadlock issue here (https://github.com/thotro/arduino-dw1000/issues/37). May you any idea :)

thotro, 2015-09-17:
@greymfm Thanks for the LDE-load-from-OTP code! If you allow I will integrate it for the next commit, or maybe you want to issue a pull request? Cheers!

curlyz, 2019-02-19:
I am using uPython with ESP32
Using MODE_LONGDATA_RANGE_ACCURACY solve everything.

Leylan24, 2019-08-14:
Hi @greymfm I am using the STM32f103 and trying to get it to work with the dwm module. I can read from 0x00 and receive DECA0130 which is correct. However if I read twice the second time the value is wrong. This is the same if i write to 0x21 register and read from it, the first time it works fine however the second time it does not. Do you have any ideas as to why this happens? If possible could I see your SPI read and write functions mine is as follows

`void SPI_read(byte reg, int len, byte data[]){  
  SPI.beginTransaction(SPISettings(16000000L, MSBFIRST, SPI_MODE0));
  digitalWrite(SS, LOW);
  byte x=SPI.transfer(reg);
  delayMicroseconds(5);
  for(int i=0; i<len;i++){
    data[i]= SPI.transfer(0x00);
    Serial.print("\nREAD:");
    Serial.print(data[i],HEX);
    delayMicroseconds(5);
  }
  digitalWrite(SS, HIGH);
  Serial.print("----------");
  SPI.endTransaction();  
}

void SPI_write(byte reg,int len, byte data[]){
  SPI.beginTransaction(SPISettings(16000000L, MSBFIRST, SPI_MODE0));
  byte header= 0x80 | reg;
  digitalWrite(SS, LOW);
  SPI.transfer(header);
  delayMicroseconds(5);
  for(int i=0; i<len;i++){
    SPI.transfer(data[i]);
    delayMicroseconds(5);
  }
  digitalWrite(SS, HIGH);
  SPI.endTransaction();
}`