thotro/arduino-dw1000 #34: DWM1000 UWB46

lvenki, 2015-08-31:
We are using DWM1000 UWB46, we connected the chip to Arduino R1 (Deumilanove). We are using the BasicConnectivity sketch, the below is the output. Any idea what could be the issue, we wired for spi (ss,mosi, miso, clk), intr, reset

Device ID: DECA - model: 0, version: 0, revision: 0
Unique ID: 00:00:00:00:00:00:00:00
Network ID & Device Address: PAN: 00, Short Address: 00
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5

Thanks in advance
Venkatesh

MarioRuz, 2015-08-31:
Hi,

Same problem happened to me. You should check your wires/connections. Turn off and turn on again the Arduino power. I still do not really know what is the problem, it may be related with the wiring and the limited current of the 3.3 V output. What I can tell you is that I had the same problem and I could make it work eventually.

Best,

-Mario

greymfm, 2015-09-05:
Do you say the Arduino library will work with your DWM1000 UWB46 antenna? I just noticed I also have the UWB46 antenna on the chip (and not the UWB42 as shown here https://github.com/thotro/arduino-dw1000/wiki/Testbed-and-Adapter-board).

Should it work? I have stable voltages (3.3v), can read device ID, send packets - but no receive without errors. Tried all diffferent modes without success. Maybe it's related to the UWB46 antenna?

stuartaw, 2015-09-07:
And my DWM1000 chips carry UWB44 antenna chips.. interesting.
I would be surprised if it made a large difference, but obviously the antenna is 'in development'.
I need to get back in to some performance checking..

jogando, 2015-09-21:
I used the UWB46 antenna for ranging without any problems

thotro, 2015-09-21:
@jogando Thanks for sharing the info! 

My two cents on the topic: The module interface is (obviously) independent from the module's auxiliary components, so I would also be very surprised if the type of antenna (as long as conforming) matters. Antenna calibration could be a topic though, but we will hopefully have a simple calibration sketch implemented soon (#8).

Cheers!

closed 2017-04-24 by Rotzbua