thotro/arduino-dw1000 #35: Is it ok to program arduino pro mini right on the breadboard?

evgeniynickolaev, 2015-09-03:
Hello! Thank you very much for you work!

I have a question, sorry it is probably a noob question.

I program my arduino pro mini with USBASP programmer.
Is it ok to program arduino which is connected to a dwm adapter? 
Or shall I disconnect it and program separately?
I wonder because adapter connected to the same pins of arduino (as it uses SPI too).

leosayous21, 2015-09-16:
Hi ! sorry for the delay...
did you find a solution ?
What i can tell you is that you can use different devices in SPI if you use a different SS pin for each devices

evgeniynickolaev, 2015-09-16:
I guess I can program it via UART. But have not done it yet.
As for different SS pin - I am afraid to change anything from the standard scheme until I get something working
Thanks for the response!

closed 2015-09-16 by evgeniynickolaev