thotro/arduino-dw1000 #36: SPI data not written correctly for extended adresses

greymfm, 2015-09-04:
readBytes and writeBytes are not setting the extended address bit correctly:

This is the **corrected** code:
```
     static const byte EXT_ADDR = 0x80; // extended address
      .......

		} else {
			header[1] = EXT_ADDR | (byte)offset;
			header[2] = (byte)(offset >> 7);
			headerLen+=2;
		}
```

I found it out by verifying the sent data (in writeBytes) by reading back the writte data.

```
wr@23:4=> 70,88,
wr@23:C=> 7,A9,2,25,
wr@23:12=> 35,0,
wr@27:2=> 1,0,
wr@27:4=> 87,0,
wr@27:6=> 20,0,
wr@27:8=> 2D,0,1A,31,
wr@27:26=> 28,0,
wr@2E:806=> D,
  ERROR: 0,
wr@2E:1806=> 7,16,
  ERROR: 0,AD,
wr@2E:2804=> 8E,42,
  ERROR: 0,0,
```

greymfm, 2015-09-06:
BTW:  accuracy seem to incease by this fix because otherwise the tuning parameters are not written correctly to the chip  :)

thotro, 2015-09-17:
Hi there,

sorry for delay and thanks for your help in debugging! 

Although I did not yet have time to verify this extended address issue, your "corrected" code seems to be equal to the one in already in place. Your EXT_ADDR is equal to the currently used WRITE constant. 

Did I miss something? ;-)
Please clarify and many thanks!

Cheers,
Thomas

greymfm, 2015-09-18:
For extended addresses, regardless of read/write, bit 7 of Octet 2 must be set (page 12 in user manual). In the current code, it's not set for read operations, and so the read fails (PS: try to write and read back an extended address).

thotro, 2015-09-18:
Uh, oh! Thx! Will fix it.

closed 2017-04-24 by Rotzbua