thotro/arduino-dw1000 #38: DW1000.cpp: setData behaving strange (only certain data is received)

greymfm, 2015-09-05:
After debugging, I discovered a strange transmission behavior. I tested all of the code examples, some did work, other did not work - finally, I found out that the problem is actually the transmitted data:

1. Default BasicReceiver/BasicSender code does not work, until I set data to this:
String msg = "42";   DW1000.setData(msg);
However, this data does NOT work:
String msg = "aa";   DW1000.setData(msg);
It is never received at the receiver side (no interrupt either).
2. RangingAnchor/RangingTag does not work. No first ack at receiver. Until I change the data to "42", then I also get an receiver ack.
3. DW1000Ranging_ANCHOR, DW1000Ranging_TAG works.

Any ideas?  I did all tests using DW1000.MODE_LONGDATA_RANGE_ACCURACY.

Same behavior on Arduino Nano and STM32.

leosayous21, 2015-09-05:
Maybe it’s due to the frame filtering !
Try to disable it in setDefaults() of the DW1000Class.
-> change  setFrameFilter(true); to false in line 1093

greymfm, 2015-09-05:
That was it :) Thanks so much. Good side is I learned the DW1000 code :)

closed 2015-09-05 by greymfm

leosayous21, 2015-09-05:
Yes we talked with Thomas to disable frame filtering in DW1000.cpp and add it just in the DW1000Ranging class in adding a function to activate it after but we didn't do it yet :(

thotro, 2015-09-05:
Oh yes, sorry guys, this is still on my list! ;-) 
Will be soon back on the repo more frequently ...

greymfm, 2015-09-05:
No problem at all :-) My hardware code is working (happy) - I learned (forgot) that I should not consider 'master' branch as stable ;-)  (in our github projects, master is always latest stable, and we use another 'develop' branch for working code...)