thotro/arduino-dw1000 #40: digitalWriteFast is not functional on STM32..

stuartaw, 2015-09-09:
After some testing I can confirm, at least for me, digitalFastWrite is not actually doing what is intended on stm32 builds.
I would be surprised if it does anywhere.
it requires __builtin_constand_p() to be true for both args, and it does not appear that will be true for the current implementation as the compiler does not seem to consider them constant, at least in the examples.

For stm32 I have substituted an alternative that does work, however am not sure of the best compatible solution.
Perhaps the AVR compiler is managing to do the right thing, whereas the arm path is not?

on stm32 the overhead is not too bad, 1us per call, but still not great. on AVR the difference will be much more obvious - perhaps look on a scope how early CS is asserting?

leosayous21, 2015-09-09:
Hi !
digitalWrite is about 7us for arduino uno (atmega328p 16Mhz 5v) so we can expect it to be 14us for 8Mhz (3.3v)...
We must use a #ifdefine in the code in order to use digitalWriteFast when we are using an atmega but use simple digitalWrite for others (teensy, etc..).
What do you think about it ? Do you think you can do it and pull request ?
regards,
Leopold

stuartaw, 2015-09-10:
If you can wait a couple of days I should be able to give you some options. on stm32 I am using a different faster IO call that is 10 times faster (digitalWrite there is 1uS, my other call is 0.15uS).
The problem with the digitalFastWrite is the fact that the pin is not const from the point of view of the compiler.. I will have a play.
I dont use atmega chips, but I can check the dissassembly, if you can wait.
The only cost I believe is performance, and that depends a lot on the SPI clock being used.

commit da133a7f (2016-10-24): remove digitalWriteFast

as suggested in #89 digitalwrite is faster in newer ide (>=1.6)
fixes  #40
closed 2016-10-24 by thotro