thotro/arduino-dw1000 #41: Support for multiple tags!!

Blue-Orb, 2015-09-12:
Hi,

Great work!! I have used the library for 3 Anchors - 1 Tag with Arduino Lilypad (USB). It works very well. It seems that the library doesn't support multiple tags yet. Is any upgrade under development currently? 

Will adding the code for "if(_type==TAG)"  in "addNetworkDevices" in "DW1000RangingClass" be sufficient to extend the functionality to multiple tags? If not, can you please advice me on changes required to support multiple tags.

Thanks

leosayous21, 2015-09-12:
Hi blue-orb ! 
This is great you give us a feedback for the 3 anchors, because i didn't test with it ! Thanks!
Yes, at the end, we want the library to support multiple tags.
It's not this simple..; In fact, it's the TAG which emit a message (the poll message) to all anchors in order to start ranging. This message is sent as a specific frequency. (#define DEFAULT_TIMER_DELAY 80 at the beginning of the DW1000Ranging.h) It means each message for starting a ranging cycle is send every 80ms. 
(in fact it's the time for 1 anchor. When we have more anchors, the timer delay is changed dynamically in order to wait more because there is more message to be sent...)
So in conclusion, tags need to know how much tags are working together. In order to dynamically change the frequency. And the messages must be sent not in the same time. (that mean: TAG 1 need to init ranging with anchors, do ranging cycle. then TAG 2 need to init ranging with anchors, do ranging cycle.)
I didn't really think about all of that precisely, and i didn't find documentation from decawave...
Did you understand what i mean ? I hope i was enough clear ... 
I will start again to work on the library soon !
cheers,
Léopold

maxicastle, 2016-02-22:
That sounds really complex, why not use the same architecture as in the GPS technology?
where the satellites are sending information in a broadcast configurations and the receiver gets the dynamic position of each satellite and with that information can get the position.

I suggest the next architecture to create a web server system to manage multiple TAGs location

[image]

what do you think about this architecture?, do you think that is possible to create this system with the current library? Any feedback can be helpful.

Thanks

ttaber, 2016-02-22:
Flip that architecture around and I think you've got the beginnings of a solid TDoA setup which, according to DecaWave, can support up to 11,000 simultaneous tags (YMMV).

In the most simple terms, rather than your Anchors transmitting, they'll need to be setup to receive.
One Anchor is going to need to acting as a timekeeper and TWR the other anchors to get time of flight and timestamps.
This will allow you to calculate a time offset in the ns range for your Location Engine to measure by.

From there, you should be able to Blink your Tags periodically and then relay the receipt of that from your Anchors to the Location Engine which can then use the offsets generated by your TWR + the time-stamped receipt of the Blink from the "Master" Anchors and feed that into a multilateration algorithm to give you a position.

 Or something like that anyways... I've been mulling over making an attempt at trying to put something like that together using this library but haven't found the time to get started.

YeongJunKim, 2016-08-16:
HI
I found message from library about "for now let's start with 1 TAG" in DW1000Ranging.cpp
Do not yet two more TAGs are supported?

manomitbal, 2016-08-16:
I see that you dynamically assign polling delays with the number of devices  (ANCHORS) present:

`_timerDelay = DEFAULT_TIMER_DELAY+(int)(_networkDevicesNumber*3*DEFAULT_REPLY_DELAY_TIME/1000);`

Where and how would be put in the check for a second TAG and dynamically assign delays for polling so that there are no conflicts?

Thanks

totterfree, 2016-11-27:
I also have the need for multiple tags, to work with at least one anchor. Although it may not be the most efficient, could you have the tags request a range operation before actually performing it? I'm thinking it would work something like this For tags A, and B, as well as anchor A... Tag A, and tag B both decide they'd like to do a range check at the same moment in time. They both send a request to anchor A. Tag A's request happens to get read first. Tag B's request is added to a queue of ranging requests to work through. Both tags immediately wait for a message back before starting a ranging operation. At this point, Tag A gets a message back, and does a ranging operation. Tag B however, has not heard anything back, so it assumes it has been queued and waits. Tag A's ranging operation completes and anchor A checks its queue to see that it should reply to tag B's ranging request. Tag B finally gets a message back and begins its ranging operation. Both Tags have now been ranged at the expense of one additional "ping" per range operation

Technickster, 2017-03-01:
I was wondering if this has been updated before I dig in and have a look?

Thank

Rotzbua, 2017-04-26:
closed in favour of  #181

closed 2017-04-26 by Rotzbua

Arsenezhu, 2018-04-27:
I put three Anchors around the 2m*3m competition venue。
(0,0)
(2,0)
(1,3)
with height of 0.35m
And the tag is on the robot with height 0.35m。The picture below is one of the 3 Anchors
[image]

The test video is with only 3 anchors and 1 tag in channel_7.
https://drive.google.com/open?id=1TxSGrx_w0xiMKlBBKkp-kTj5YrbTJhDZ
It works great.

Then I am trying to implement 6 Anchors 2 tags system. 3 Anchors and 1 tag are all in channel_7 and the other set of 3 Anchors and 1 tag are all in channel_4. Both tags works well with around 3 minutes. Then one set of the tag starts not to receive one of the 3 Anchors in same channel.
Below picture is the two Anchors. I put 2 Anchors in same point.
[image]

(0,0) (0,0)
(2,0) (2,0)
(1,3) (1,3)
with height of 0.35m and 0.3m
I think maybe the problem is I put two anchors too close? And the electromagnetic wave is too strong, so it interrupt?
Hellow @leosayous21. How you solve your problem? Any advice is welcome

really needs help