thotro/arduino-dw1000 #43: Arduino Type

zach401, 2015-09-24:
I was wondering if you were using the Arduino Pro Mini 3.3V/8MHz in your examples or if you used the Pro Mini 5V/16MHz with a voltage translator board. From your layout example it looked like you were using a 3.3V Arduino but I wasn't sure if 8 MHz would be fast enough.

thotro, 2015-09-24:
Hi! 

I was using a 3.3V version and by now have successfully tested both 8 MHz (Arduino Pro Mini default) and 16 MHz (maximum clock the Atmega can operate). Voltage regulators are an option too, of course, although not convenient at all. 

Regarding "fast enough": It really depends what you try to accomplish in the end! ;-)

Hope this helps, otherwise let me know!

Cheers, Thomas

zach401, 2015-09-24:
Thanks, I will order the 3.3V version. I appreciate the help.

closed 2015-09-24 by zach401