thotro/arduino-dw1000 #44: DW1000Ranging_Tag/Anchor

NewStart14, 2015-10-08:
Firstly, many thanks for the great work you have done in maing this technology more available. I am very grateful!

I am trying to implement the DW1000Ranging_Tag/Anchor example but cannot get it to work. I am driving the Anchor with an Arduino Uno and the tag with a TinyShield (AT328P).

Once the sketches are done uploading, I open the Serial Monitor for the Anchor and see:

device address: 82:17:.......
### ANCHOR ####

but nothing else. Is some further initiation required? I have checked wiring errors with no joy.

Your thoughts would be very welcome. Thank you.

NewStart14, 2015-10-13:
Scratch that. It WAS a wiring issue and the code works perfectly, thank you once again for your work.

jesp007, 2015-10-13:
Could you please be more specific about the wiring issue, i am having similar problems.  Any help will be greatly appreciated.

NewStart14, 2015-10-13:
Sure. I had incorrectly wired DWM1000 pin 22 (IRQ) to the arduino I/O pin 8. The schematic clearly shows that this should be connected to I/O pin 2 for the interrupt. I am not using the adapter board so had messed up my pin outs. I hope your issue is as easy to fix.

closed 2015-10-13 by NewStart14

zach401, 2015-10-14:
Has anyone else had this issue despite having the wiring correct? I and a friend double checked our wiring but we are still only getting

device address: 82:17:.......
### ANCHOR ####

and no other data or ranges. We are also unable to get any of the basicSender and basicReceiver code to send and receive messages. We are using two Pro Mini's 3.3V and the breakout board.

jesp007, 2015-10-14:
I have the same setup, same problem.

NewStart14, 2015-10-14:
It sounds like the modules are either not getting power, or are asleep. Check the following:

1) Does the module get warm after being powered for a few minutes? If not, it is not getting power.
2) Check your external interrupt (Arduino Pin 2 -> Adapter Board Pin 17 -> Module Pin 22). 

It does not sound like an communication (SPI) issue as you should see more on your serial monitor.

zach401, 2015-10-15:
I have double checked and I made sure that my modules are drawing power (about 125 mA a piece) and that the interrupt wires are connected correctly. Has anyone had trouble with the two modules not being able to communicate between one another? I know my SPI connection is working because the Basic Connectivity script is working. 

Also, is anyone else having trouble with the serial prints sometimes getting corrupted?

leosayous21, 2015-10-15:
Hi ! Can you check if the frameFiltering is desactivate ?
line 1093 of the DW1000.cpp. must be setFrameFilter(false); if you want to communicate without adding mac address in front of your message...
(But the DW1000Ranging.cpp do this for you) 
regards

MarioRuz, 2015-10-17:
Hi. I had the problem @zach401 is commenting, in my case i am using arduino 1. I did not check if it was due to the maximum current that can be drawn. Actually, I am not sure if arduino 1 mini pro provides more current than arduino 1 (that sounds odd to me).

In the data sheet of arduino 1 you can read that the maximum current is 50 mA. https://www.arduino.cc/en/Main/ArduinoBoardUno
That can be a problem when powering the decawave chip. However, I also read that tha maximum current from the 3.3 V output is 150 mA, which should be enough:
http://electronics.stackexchange.com/questions/67092/how-much-current-can-i-draw-from-the-arduinos-pins

Can someone give a confirmation on this?

You can read about what we discussed here: https://github.com/thotro/arduino-dw1000/issues/6

When the problem happened, I plugged off and then on the arduino power. But still I didnt figure out where the problem is.

PS: I have to test it with the last version of the libraries.
Cheers!!

zach401, 2015-10-17:
We just got our setup working. It must have been a wiring issue, because once we unplugged everything and reset it up, everything worked well. We also made the change that leosayous21 suggested, but it did not work until we rewired. 

We are at a university, so we are powering the DWM1000 from a 3.3 V power supply instead of from the arduino's regulator. In our test each DWM1000 was drawing about 150 mA, which when combined with he current draw of the arduino itself might be too much for the onboard regulator (or at least near its limits). If you have access to another power supply or an external regulator that might help.

MarioRuz, 2015-10-24:
Ok, @zach401 . It seems then there may be issues with the 3.3 V output powering the DW1000. The conclusion is that it might be possible that the 3.3 V maximum current from some arduinos is not enough.

Cheers!

curlyz, 2016-11-29:
@MarioRuz  you didnt count for the current Arduino use,use external power supply

Maxai2, 2017-09-07:
Hi! Please tell me what we do wrong?
We connect decawave dw1000 to arduino Uno by SPI connection, upload this sketch 
DW1000Ranging_TAG
https://github.com/thotro/arduino-dw1000/blob/master/examples/DW1000Ranging_TAG/DW1000Ranging_TAG.ino

But when we open arduino monitor, we see nothing, as if Decawave receive nothing... And on the Decawave led "ERROR INIT FAIL". 

What we miss, where our mistake? Help please. Thank u.