thotro/arduino-dw1000 #45: DW1000Server & GUI

jogando, 2015-10-16:
Hey guys,
I created a server and a GUI for combining the range reports from the anchors and making the calculations for positioning the Tag using trilateration.
Here you can find the source code:
https://github.com/jogando/dw1000server

here you can find an example:
https://www.youtube.com/watch?v=5WXWx_jxGw8

The only change you need to make to your Arduino projects is to replace the DW1000Ranging_ANCHOR.ino file:

void newRange(){  
  Serial.print("["); Serial.print(DW1000Ranging.getDistantDevice()->getShortAddress(), HEX);
  Serial.print(":"); Serial.print(DW1000Ranging.getDistantDevice()->getRange()); Serial.println("]"); 
}

This way the Serial will print a string like [tag_address:distance]

Thanks a lot for your great work.

cheers,
Julian

mrAyaz, 2016-01-30:
Hello dear Jogando. Please help! How to run your Application on Windows?.

maxicastle, 2016-02-29:
I have a problem with the DW1000Ranging_TAG and DW1000Ranging_ANCHOR sketches,
I can't compile  any sketch, I get next errors:

Arduino:1.6.6 (Windows 7), Placa:"Arduino Pro or Pro Mini, ATmega328 (5V, 16 MHz)"
[image]

any Idea how to solve it?
also, the web server was created using ECLIPSE IDE?