thotro/arduino-dw1000 #46: Two Way Ranging - change message rate

NewStart14, 2015-10-21:
Sorry for the newbie question, but I am struggling a little to get my head around the code. Can someone provide some pointers on how I change the rate at which range data is reported please? I am using the simple DW1000Ranging_Anchor/Tag examples.

My ultimate aim is to achieve range updates at 100Hz over a LOS range of 30m.

Your guidance would be appreciated.

thotro, 2015-11-05:
Hi there,
sorry for the huge delay - I'm really busy these days. :-/

Your update rate requirements depend a little bit on the speed of your host controller. With Atmega 328 at 16 MHz I have no troubles in getting updates at a minimum (!) of 30Hz for one tag/anchor setup. So 100Hz should be doable with a faster controller, I guess.

By using the LONG_RANGE_... modes that you can use for the driver mode configuration I get ranges of over 60m. Although there is still space for optimization, e.g. regarding power management and some other bits and flags.

So, what is your current setup like and what ranges/update frequencies do you experience?

Thanks and hope it helps a bit!
Thomas

NewStart14, 2015-11-05:
Thanks Thomas,
 
I appreciate your response. I am having some break-out boards manufactured at the moment to make my setup more robust. I shall do some testing when they come in (next week hopefully) and come back to you with some real data. I am running on Atmega 328 at 8 MHz - it sounds like I need a new processor!
 
I'll be back with data.
 
NS

 
Date: Thu, 5 Nov 2015 06:28:48 -0800

rasamaya, 2015-11-05:
oooh, breakout boards. Mind sharing??

-- 
David Russell
3D Solutions
Technical Director and Spatial Processing
204.391.3880

CONFIDENTIALITY NOTICE:
The contents of this email message and any attachments are intended solely
for the addressee(s)
and may contain confidential and/or privileged information and may be
legally protected from
disclosure. If you are not the intended recipient of this message or their
agent, or if this message
has been addressed to you in error, please immediately alert the sender by
reply email and then
delete this message and any attachments. If you are not the intended
recipient, you are hereby
notified that any use, dissemination, copying, or storage of this message
or its attachments is
strictly prohibited.

NewStart14, 2015-11-05:
"Manufactured" was a bit of an embellishment, they are being made by someone with better soldering skills than myself(!) - no gerbers I'm afraid. The adapterboard on here looks pretty good already.
 
Date: Thu, 5 Nov 2015 07:31:33 -0800

rasamaya, 2015-11-05:
Ah, thanks. I have not even opened my dw100's. I have a bag of them waiting
for my current contract to wrap up.
Looking forward to contributing.

-- 
David Russell
3D Solutions
Technical Director and Spatial Processing
204.391.3880

CONFIDENTIALITY NOTICE:
The contents of this email message and any attachments are intended solely
for the addressee(s)
and may contain confidential and/or privileged information and may be
legally protected from
disclosure. If you are not the intended recipient of this message or their
agent, or if this message
has been addressed to you in error, please immediately alert the sender by
reply email and then
delete this message and any attachments. If you are not the intended
recipient, you are hereby
notified that any use, dissemination, copying, or storage of this message
or its attachments is
strictly prohibited.

bigredbee, 2015-11-05:
I think I've posted this before, but here's a link to the adapter I had
made.

https://oshpark.com/shared_projects/8M99bseq

Price:  About  $7 for three boards.

Greg

rasamaya, 2015-11-05:
Nice, I just ordered 6 :)

-- 
David Russell
3D Solutions
Technical Director and Spatial Processing
204.391.3880

CONFIDENTIALITY NOTICE:
The contents of this email message and any attachments are intended solely
for the addressee(s)
and may contain confidential and/or privileged information and may be
legally protected from
disclosure. If you are not the intended recipient of this message or their
agent, or if this message
has been addressed to you in error, please immediately alert the sender by
reply email and then
delete this message and any attachments. If you are not the intended
recipient, you are hereby
notified that any use, dissemination, copying, or storage of this message
or its attachments is
strictly prohibited.

closed 2017-04-26 by Rotzbua