thotro/arduino-dw1000 #47: clock synchronization between the anchors in TDOA

KnifeFly, 2015-11-10:
Hi;
    I know you use the two-way ranging as the localization algorithm, but the TDOA save more electricity when the tags use the battery as the power supply. i am going to implement the TDOA algorithm, but i am confused by clock synchronization between the anchors. 
   Comparison of wireless clock synchronization algorithms for indoor location systems.pdf provided by the decaware ltd, one anchor is transmitting a packet every 150ms as measured by its clock, and the other anchor receives, the clock offset between the sender and receiver is about 31ns. the clock error is so large, how can we get the 10cm precision as the Decaware declare.

KnifeFly, 2015-11-13:
i found 31ns is the offset between the external crystal, not the dw1000 internal timesramp offset

ehigh2014, 2016-07-05:
Do you have a test of the wireless clock synchronization algorithms?
How about the precision of this method?

curlyz, 2016-11-14:
I think synchronization is the most painful problem to deal with , even using wire , we will still have some gap which result in some error.

curlyz, 2016-11-14:
my idea is to use a tag placed in the center of all anchor and send its timestamps to all anchor, The accuracy of this method will be up to you ,

Rotzbua, 2017-04-26:
part. ref #16

krskajo1, 2017-10-02:
Have you took in account the distance between the sender and the receiver? If not, then the clock offset will be proportional to the distance between these two. In this case the distance was around 10 m.

hdineth, 2017-11-07:
Has anyone find a way to increase the delay between two range measurements?