thotro/arduino-dw1000 #48: Serial Monitor Display Problem

tn0432, 2015-11-11:
Hello!

I try your code BasicConnectivityTest with Arduino Pro Mini 3.3v. The received characters on serial monitor is not shown probably. I have tried different baudrate and put delay function, it is not improved. 

Hope you can help! Thanks a lot!

[image]

evgeniynickolaev, 2015-11-11:
Did you change the baudrate in the "setup" method?

Original version has:
```
Serial.begin(9600);
```

I see you serial monitor has different baudrate, set it to 9600

tn0432, 2015-11-11:
Yes, I try also try 9600

evgeniynickolaev, 2015-11-11:
Try to upload this sketch

```
void setup() {
  Serial.begin(9600);
}

void loop() {
  delay(1000);
  Serial.println("test");
}
```

tn0432, 2015-11-11:
Great! I find out the problem! My bad! I forgot to set Tools / Board to Arduino Pro Mini. Your code works and the BasicConnectivityTest code also works. What should we set for Tools / Programmer? default is AVRISP mkII

However, the display of character has some small errors. Do you know how to improve that? Thanks

[image]

evgeniynickolaev, 2015-11-11:
I had not paid attention to it)
I program Arduino pro Mini via USB-FTDI adapter.
Just have looked at the Arduino IDE and there I have "USBAsp".
It just left there after I programmed attiny13 via USBAsp.

dtysky, 2015-11-13:
I solve this problem today, here are sulotions:
Insert delay(100) before every printf, as follows:

<pre>
    DW1000.getPrintableDeviceIdentifier(msg);
    Serial.print("Device ID: "); Serial.println(msg);
    delay(100);
    DW1000.getPrintableExtendedUniqueIdentifier(msg);
    Serial.print("Unique ID: "); Serial.println(msg);
    delay(100);
    DW1000.getPrintableNetworkIdAndShortAddress(msg);
    Serial.print("Network ID & Device Address: "); Serial.println(msg);
    delay(100);
    DW1000.getPrintableDeviceMode(msg); 
    Serial.print("Device mode: "); Serial.println(msg);
</pre>

You can also try to use a independent power for dw1000.

Rotzbua, 2017-04-24:
outdated

closed 2017-04-24 by Rotzbua