thotro/arduino-dw1000 #49: Independent power is needed for dw1000...

dtysky, 2015-11-13:
Hi.
I used dwm1000 with arduino nano, something wrong was happened but solved at last.
The power from arduino nano or mini (USB) can not support dwm1000 enough, if your code can not work or just work a little time, these may help you.

dtysky

evgeniynickolaev, 2015-11-13:
My modules had not work with some wall adapters. 0.55A, 1.25A -the only one that worked had 2A current.
Actual current was not so big - maximum 160mA (including arduino mini pro itself).
The issue was solved after I added 1000uF oxides between VCC and GND right after the power supply.
Adapter could be 1A, 1.25A or more but the same time it could have relatively small output capacitors so might get lack of power on some consumption peaks.
Though, I couldn't get it work with 5V 0.55A adapter even with oxide. 

Also please note that in case of Arduino Pro Mini it is certainly great to have FTDI apdapter's VCC not to be connected to the external power. I've just disconnected it. So I can program arduino without switching off anything and use Serial Monitor and it is safe for the computer's USB ports.

It would be very great if the project's wiki had a page about power supply. For instance, describing that external power is a must etc.

dtysky, 2015-11-13:
hi, evgeniynickolaev.

After added 500uf, this module works longer, then i think there is a problem with power. And as your said,  VCC and GND on dwm1000 are not connected to arduino in my solution, then I can program arduino with the USB ports.

Hope for a page about power supply too...

EarthBreaker, 2016-02-22:
Right now, I've had problems running most of the examples. Could it be due to the way I power my DWM1000? Right now, its not really connected to an external power supply.

dtysky, 2016-02-22:
@EarthBreaker 

If you want to use DW1000 for product but not just as a toy, I think the EVK1000/TREK1000 or others products are better......  

Actually... even as a toy, this project is better:  

[DW1000 with MBED](https://developer.mbed.org/users/manumaet/code/DecaWave/)

tn0432, 2016-02-29:
Hello @dtysky 
Can you describe more about your power supply? 

I have tested with BasicConnectivityTest, my arduino nano work okay. After that, I move to test  BasicSender and BasicReceiver, they do not work. The serial monitor only prints out "### DW1 ". Do you have the same performance like that? Thanks

dtysky, 2016-03-01:
@tn0432 

Emm...the most improtant factors is current...
If you can read chinese, play check it here:

[My records for uwb](http://dtysky.moe/tag/uwb.html)

kshitijdadhekar, 2016-03-11:
@dtysky here is the screenshot of my serial monitor window
[image]

i tried connecting a step down module of 3.3 volts to the DWM. My question is the Unique ID goes FF:FF:FF:FF:FF:FF:FF:FF in the basic connectivity program and if I disconnect the DWM from the PRO MINI then it goes 00:00:00:00:00:00:00:00

what must be the problem?

dtysky, 2016-03-12:
@kshitijdadhekar 
I don't think it's a problem, only if the DWM1000 has been connected and initialized successfully will change the IDs from default to your setting, if not, the default IDs will be shown.
You can check these in source code.

Maxai2, 2017-09-18:
Hi!
Please somebody help me with dw1000 connect to arduino... I need some help...