thotro/arduino-dw1000 #50: Ranging problem in 2D (multiple Anchors)

Pyro-Brad, 2015-11-16:
Hello everyone,

First of all, great work guys! It's a large library and it works pretty good!

But here is my problem. (I’m sorry)

I have one tag and three anchors. 
This is my set-up to measure in 2D:

[image]
(Distances are centimeters)

This is what happens:
[image]

Each time an anchor joints the network the distance changes.

I tried it with a larger distance:
[image]

Same result. The difference is a little bit smaller.
After one sample the distance is stable, good sign I think?
But... The distances doesn’t agree with the physical distance. 

The distance offset could be calibrated but that is not the problem in this case I think.
Because the distance changes each time when there joints an anchor.

- Could it be the transmit power?
(How do I change the transmit power in the library?)
- Could it be the transmit mode? 

When this problem could be solved I can calculate the position in 2D.
Next step is 3D by adding a fourth anchor. (Firmware is already complete for 2D and 3D)

Could anyone help me? 
Thank you in advance.

Pyro-Brad, 2015-11-16:
Setup is on a wooden table:
[image]

All the UWB modules are placed vertically.
Antenna is vertically polarized and increases the free space around the module antenna.
[image]

The Tag and Anchors are a little bit higher than the table.
Because of possible UWB reflections.

Pyro-Brad, 2015-11-16:
What I already tried:
(14-11-2015 14:33 uur)
- Anchor position from two to four.
- Channel from 5 to 2. (Because it was already channel 5)
- Switch off my wifi from my laptop.
*Still no other result.*

(place for next update)
-
-

dtysky, 2015-11-17:
Hi, i  did some tests too and get some problem...  
  
[image]  

[image]  
  
1. The power which supply for dw1000 is really important, good power can let it work better and bad power could make it never work. And different power could the longest distance of two dw1000.  

2. The gap of distances from different anchors to tag are very large, this is result of following test:
test:  

[image]  

result:  

[image]

Broeserl, 2016-06-02:
Hi everbody, 

I have the same problems as the two guys before me.

I am testing with 2 Anchors and 1 Tag.
[image]

When i start only Anchor 1 and the Tag are on. The distance deviation is okay for me, about +-20 cm.
[image]

Now I power Anchor 2 and the distance of Anchor 1 changes.... 
The distance of Anchor 2 is "bad"...
[image]

But when I turn off Anchor 1. (The value shown by A1, is an old one and not true)
The distance of Anchor 2 is now pretty good!
[image]

I'm using DW1000.MODE_SHORTDATA_FAST_ACCURACY and channel 2.
I have tried all available modes and channels. The combination I used works best for me.

Could anyone help me? 
Thank you in advance.

maxicastle, 2016-07-09:
You Can Use This Document to understand your Problem.
[Application Note APS014 Antena Delay Calibration.pdf](https://github.com/thotro/arduino-dw1000/files/355389/Application.Note.APS014.Antena.Delay.Calibration.pdf)

AliSiddiqui4, 2017-01-15:
@Pyro-Brad can you please post the link to the code of your anchor and tag here. I am using the library example and my tag doesn't add the 3rd tag. I have tried changing the MODE.

Rotzbua, 2017-04-26:
closed in favour of  #181

closed 2017-04-26 by Rotzbua

duonghuu123, 2023-05-26:
If the distance is close, the error will be large, if you expand the distance to about 5m, the result is quite accurate