thotro/arduino-dw1000 #51: Transmit power erro

Pyro-Brad, 2015-11-18:
Hello everyone,

There seems to be a problem wtih the transmit power.
I can't chagne the transmit power.

Simple setup:
Anchor -> Tag

I use the standard library.
Correct me if I'm wrong:
Default settings are: channel 5, smartpower OFF.

The uC comes here:
DW1000.cpp line 550.
" writeValueToBytes(txpower, 0x85858585L, LEN_TX_POWER); "

This is the code line where it should be flashed in the DWM1000 module:
"writeBytes(TX_POWER, NO_SUB, txpower, LEN_TX_POWER);"

At the Tag I see no difference by RX power. 

Could anyone help me please?

ebps, 2015-11-27:
Hi Pyro-Brad could you contact me at info at ebps.nl ? I see no option on git to PM you.

Pyro-Brad, 2015-11-27:
Ga ik nu doen ;) (Lekker in het Nederlands)

ebps, 2015-12-01:
Hi pYRO-bRAD,

 

Heb je gemaild op 1228943gerrits zuyd.nl Weet niet zeker of je mijn mail hebt ontvangen.

 

Met vriendelijke groet,

Bas van Dorenmalen

 <http://www.ebps.nl/> EBP Software B.V.

T: 040-2429988   |   W:  <http://www.ebps.nl/> www.ebps.nl
EBP Software B.V.
Bedrijvenweg 1  |  5627 BW  |  Eindhoven  |  The Netherlands

Pyro-Brad, 2015-12-01:
Ik heb een mail terug gestuurd

Pyro-Brad, 2015-12-04:
Goedemorgen Bas,

Heb je mijn mail ontvangen?

Groeten,

Bert Gerrits

ebps, 2015-12-08:
Hi Gerrit,
Ik heb je terug gemaild.

Pyro-Brad, 2015-12-15:
U heeft een reactie. Sorry voor de late mail.

closed 2017-04-26 by Rotzbua