thotro/arduino-dw1000 #52: BasicConnectivityTest producing rubbish over serial port

madmike159, 2015-11-23:
I have been playing around with the library and have been getting rubbish out of the serial port when trying the basic connectivity test. I have managed to create a fix (this issue was also affecting other examples and my fix also allows them to run successfully).

The issue was coming from the message buffer, the data was not obviously being overwritten, but the strings that were printed contained seemingly random characters. My fix was simply to change the single message buffer into 4 separate ones (see below).

Has anyone else had this issue, or know why I might be getting it?

	char msg1[50];
	DW1000.getPrintableDeviceIdentifier(msg1);
	Serial.print("Device ID: "); Serial.println(msg1);

	char msg2[50];
	DW1000.getPrintableExtendedUniqueIdentifier(msg2);
	Serial.print("Unique ID: "); Serial.println(msg2);

	char msg3[50];
	DW1000.getPrintableNetworkIdAndShortAddress(msg3);
	Serial.print("Network ID & Device Address: "); Serial.println(msg3);

	char msg4[50];;
	DW1000.getPrintableDeviceMode(msg4);
	Serial.print("Device mode: "); Serial.println(msg4);

ChrisDziurzik, 2015-12-10:
I had the same problem.

I reduce the size of "msg" to 256 and it works. I'm searching for the reason but can't find jet.

madmike159, 2015-12-10:
It could be that there isn't enough SRAM (2k bytes for the 328). Declaring 1024 chars is half of it gone right there. I'm not 100% sure what happens when you declare more memory than an Arduino has, but I bet it's nothing good.
I will fix that section of code and submit a pull request.

closed 2016-01-04 by madmike159

tommag, 2016-03-22:
I just found out the hard way that 64 is not enough for `getPrintableDeviceMode()`... 

When powering up, the default output goes to 79 characters. The result was a processor crash in my case (on Cortex M0+). The code would run once without problems, then the processor would hang. Switching to a 96 or 128 byte buffer fixes the issue.

MikeS159, 2016-03-30:
It is probably due to the Cortex M0+ being a 32 bit processor (unlike the normal 8/16 bit of other Arduinos). The function `getPrintableDeviceMode` defines 3 unsigned short variables and 2 unsigned integer variables. My guess is that on your system the ints are 32 bit rather than 16 (which was used in testing). Something to watch out for in other areas of code with statically sized arrays and pointer arithmetic. I believe all the development of this library was done on 16 bit AVR chips.