thotro/arduino-dw1000 #54: BasicSender_BasicReceiver

raffa1993, 2015-12-10:
Hi Everybody,
I'm new in this application.
I'm tryng to setup a communication between an Arduino Uno and an Arduino Nano (with Atmel 328) using two DWM1000. I followed the electrical scheme showed in the thread. So, the BasicConnectivityTest seems to work properly, but when I try

EarthBreaker, 2016-02-11:
@ttaber, I'm here. So you roughly know my board and set up. When I tried to run the program above and check the serial monitor, I saw unknown symbols. The thing is, on another forum, there was a gentleman who suggested I use a Logic Level Converter together with my Arduino Nano (5V Input) or use an Arduino Pro Mini (3.3V Input) altogether. What's your take in this?

ttaber, 2016-02-11:
I also was getting unusual results when I was using an Arduino Mega 2560 running at 5V.

I actually cover this briefy here along with some recommended logic level converters that are compatible with SPI and I've personally used with success:
https://github.com/thotro/arduino-dw1000/issues/58#issuecomment-180063326

Either of those boards will work so long as you use the converter with the Nano.
I personally use a Pro Mini for my anchors and a Zero for my tag.

EarthBreaker, 2016-02-11:
@ttaber, I think I understand it alot better now. At first I didn't really quite get why there is a need use LLC to shift down from 5V to 3.3V since I was powering the DWM1000 via the 3.3V pin. I think I was too absorbed by that fact. Just to reinforce the learning, the reason we use LLC is mainly:
1. Yes, DWM1000 can be powered via the 3.3V pin on Arduino Nano, but that's just only powering the DWM1000.
2. When the SLK, MOSI, MISO and CS pins from the Arduino Nano board are set high, a 5V output is read,  and this is incompatible with DWM1000 as the chip operates at 3.3V. Hence all pins connected to DWM1000 has to go through a LLC before connecting to their respective pins on the Arduino to shift from a 5V to a 3.3V which the DWM1000 can read?

ttaber, 2016-02-11:
That's exactly correct.

EarthBreaker, 2016-02-14:
@ttaber, Good day Sir. I just purchased two sets of Arduino Pro Mini (3.3V) with an FDTI Basic Breakout Board so that I can try to recreate the results you have. But so far, I do not quite achieve the same results. Below is a screenshot of my BasicSender on the Serial Monitor.

[image]

I'm pretty sure I have the pins connected properly. Not quite sure what could be the problem.

ttaber, 2016-02-14:
In the bottom right corner of your console window, change "No line ending" to "Both NL & CR"

EarthBreaker, 2016-02-14:
These are my pin connections at the moment.

[image]

I've changed "No Line Ending" to "Both NL & CR" as you've advice but I still somehow receive the same error. What is the proper output like in the serial monitor @ttaber? Do advise again. Much thanks Sir.

[image]

ttaber, 2016-02-14:
Not near my workbench at the moment so I can't generate the output from the sketch, but I'll see if I can get it uploaded later.
Looks like the DW1000 is cutting off before it can fully initialize.
The Pro Mini doesn't supply enough current to run the DW1000 on its 3.3V rail.
You'll need to supply power (only) to the DW1000 from an external 3.3V supply (the Pro Mini can still be powered by the FTDI adapter). You'll also want to make sure you tied the ground rails of the Pro Mini, DW1000, and external power supply together (common ground).

This is my go-to supply for all my breadboard work:
http://amzn.com/B010UJFVTU

EarthBreaker, 2016-02-14:
Ah I see, I saw on the datasheet that the max current needed can go up to 140mA to 160mA on Channel 5. The board (Pro Mini) only provides 40mA. Would it still be fine if I managed to find a power source with a current output of 800mA for DWM1000 instead of your reccommended board? Its so that I can test immediately.

ttaber, 2016-02-14:
That would be more than enough current.
Input voltage on the DW1000 ranges from 2.8V to 3.6V so anything in-between should be fair game.

EarthBreaker, 2016-02-14:
@ttaber, i'll let you know how I fare in a couple of hours. Gonna catch some shut eye first.

EarthBreaker, 2016-02-15:
@ttaber, so far still unsuccessful. I could not purchase the power supply you recommended on short notice, so I decided to power the DWM1000 from an Arduino Due pins (3.3V, 800mA). Still have the same results though.

greymfm, 2016-02-15:
I have no problems at all using Arduino Pro Mini (3.3v version). 
I have attached the Arduino and Windows code that I did use in the video.
[dw1000.zip](https://github.com/thotro/arduino-dw1000/files/130502/dw1000.zip)

EarthBreaker, 2016-02-15:
Hi there @greymfm, may I know if you're using the same connections as me? Please refer to my picture above!

greymfm, 2016-02-15:
@EarthBreaker : yes, looks the same, I'm attaching 5V to the Arduino RAW pin - maybe you can try the Arduino Sketch I attached above.

EarthBreaker, 2016-02-15:
@greymfm! Hello again! I did try your sketch! My results as follow, but first, I'll show you my connections again.

[image]

According to @ttaber, I could not power the DWM1000 using Arduino Pro Mini's 3.3V as it does not supply enough current (40mA) as the DWM1000 requires (up to 160mA). He recommended me to purchase a power supply module but I could not buy one on short notice. As such, I decided to use power from Arduino Due as it provides 3.3V, 800mA as stated on the Arduino website.

[image]

I tried running your sketch and I have some observations:
1. The DWM1000 started becoming hot immediately, not sure if it is because I'm tapping from an Arduino Due.
2. The Anchor version of your sketch could not run properly as well. It will initially run the DeviceInfo then after that it will just stop. There are no updates afterwards.
3. The Tag version will keep saying "not found" below the "###TAG###" on the serial monitor even though both sketches (tag and anchor) are running on one Arduino Pro Mini each.

Some questions:
1. Is it possible that my DWM1000 is fried? If so, in your knowledge, is there a way to test it?
2. Do you mind drawing out a rough sketch of all the connections of your Arduino Pro mini? I'm a bit confused when you said you connected 5V to the RAW pin. Is that 5V from a battery?
3. I'm not sure what went wrong. Do you have any idea?

Sorry for the trouble! Hope to hear from you soon!

greymfm, 2016-02-15:
@EarthBreaker :  I have no idea what is that problem at your side - the DW1000 becomes warm here, not hot (hot means for me that you cannot keep a finger on it anymore). 
I have connected similar as you did. I did use an USB-to-TTL converter, and connected it like this:
USB 5v pin -- Arduino Pro Mini RAW
USB TX pin -- Arduino Pro Mini  RX 
USB RX pin -- Arduino Pro Mini TX
USB GND -- Arduino Pro Mini GND
Then I connected Arduino 3.3v to DW1000 VDD3V3.
[image]

When using battery (for outdoor), I simply add a DC/DC converter (11v in, 5v out) to power the USB converter.
[image]

I think you have wiring issues (broken cables, too long cables or something).

EarthBreaker, 2016-02-15:
@ttaber, Hi Sir, I came across this supply stick, is it appropriate as well? Couldn't really find the Output current though: http://www.sgbotic.com/index.php?dispatch=products.view&product_id=1043

@greymfm, I'll look through everything again. I don't think its the pins though. Its kind of frustrating when you don't know whats wrong. Sigh. But thanks for the help.

EarthBreaker, 2016-02-17:
Hi there @greymfm, I just looked through the schematic for the pin lay out and realised that there is a 330 ohm resistor across the IRQ pin on the DWM1000. My own circuit did not have that. Why is there a need for it?

ttaber, 2016-02-17:
@EarthBreaker apologies I didn't catch that earlier on your diagram earlier.
Yes, you need to have a pulldown resistor coming off that pin and going to ground to prevent a floating input on the pro mini.
Doesn't have to be 330Ω exactly. You could probably get away with anything up to 10KΩ without running into any problems.
That could very well be the cause of the problem you're seeing with the board failing to initialize properly.

EarthBreaker, 2016-02-18:
Hello again! First up! My connections! I do apologize for spamming pictures though.

[image]

Arduino Pro Mini 3.3V: http://www.sgbotic.com/index.php?dispatch=products.view&product_id=1182
FTDI Basic Breakout - 3.3V: http://www.sgbotic.com/index.php?dispatch=products.view&product_id=765
9V Battery Snap: http://www.sgbotic.com/index.php?dispatch=products.view&product_id=1812
Breadboard Power Supply Stick 5V/3.3V: http://www.sgbotic.com/index.php?dispatch=products.view&product_id=1043

The actual pictures of my breadboard as follows:

[image]

I tried running the program again. Results as follows:

[image]

[image]

I've been powering the DWM1000 separately using the breadboard supply stick as I've been told that the Arduino Pro Mini does not have enough current (40mA) to supply to the DWM1000 (160mA) at its peak. The supply stick is connected to a regular 9V battery. The voltage is then regulated at 3.3V for the DWM1000.

The problem now is that I'm not sure if the DWM1000 is still working properly as I've been constantly testing it may times (previously without the 330 ohm resistor). As such, I just ordered a new adapter board from seedstudio so that I can solder the DWM1000 on it again. Is there anything I'm doing wrong or possibly wrong connections, seek your advice again @ttaber @greymfm. Much thanks for the help so far.

EarthBreaker, 2016-02-22:
Hello! Some good news! After adding the 330 ohm resistor and changing this line of code:
"char msg[1024];" to "char msg[128];" the output of the serial monitor looks better. It is as follows:

[image]

[image]

So moving on, there are a few issues. My supervisor told me that there was a buffer overflow because of char msg[1024]. Is that true? But after changing that piece of code, everything seems to run properly.

Second, I can't seem to get a proper received signal. The serial monitor for that side would always give me an error in receiving messages. I've also change the "char msg[1024];" in this code to "char msg[128];" as well.

Lastly, because I somehow damaged one of my Arduino Pro Mini boards, I've decided to stop using the supply stick for a while.

wholder, 2016-03-04:
I decided to use a 5v Mini Pro, as I have a few of them in my parts cabinet.  My solution was to design 2 PCBs, one is a simple breakout board for the DMW1000 that adds a 10K IRQ pull-up resistor and a 10uF decoupling cap.  The breakout board looks like this:

[image]

After testing this with a breadboard, I decided to build an Arduino Pro Mini "under board" adapter that added a 3.3 volt, 200 mA regulator for the DMW1000 and 5/3.3 volt level shifting using 3 SN74LVC2G17DBVR parts.  The basic schematic is, as follows:

[image]

The PCB looks like this:

[image]

This board is designed so that the Pro Mini plugs into a low profile, 24 pin socket with round "machine" pins (this requires using matching machine pins in the Pro Mini.  The, standard height (Arduino style) 7 pin female headers are used to plug the DMW1000 breakout board into the under board with just enough clearance to avoid pressing the Pro Mini's RESET button (which, unfortunately, is covered up by this design.)  And, there's space on the bottom of the under board for a JST PH connector to plug in a source of regulated 5 volt power (although I considering adding a 5v step down regulator to a future rev of this board to make it easier to run the anchors from batteries.)  The resulting stack of boards looks like this:

[image]

And makes for a very compact solution. I fabricated both PCBs through OSH park so, if anyone is interested, I can make the designs public and allow people to order copies.  You can 3 copies of both boards for about $10.  The other parts are available via Mouser, or Digikey.  Note: the design does require SMD parts (all passives are 0603), but they're not that hard to solder if you use a [technique like this](https://www.youtube.com/watch?v=QzoPxvIM2qE):

Wayne

MarioRuz, 2016-03-04:
Hi @wholder,

Very interesting! That would be nice if you can make public the designs, which software did you use? 
I am still wondering why some people have problems with 3.3 V output version of mini pro, and some others not, such as @greymfm . Actually, I am not sure if the current limit of the 3.3 V output of Arduino mini pro is 50 mA. To all who are having the power ouput issue, read this: 
http://electronics.stackexchange.com/questions/67092/how-much-current-can-i-draw-from-the-arduinos-pins

Regards

wholder, 2016-03-04:
The "official" Arduino Mini Pro uses a MIcrel MIC5205, which has a guaranteed 150 mA output.  My measurements suggests that the ATMega328 when running the code is drawing about 60-70 mA on it's own.  If the DWM1000 module is using a max of 70 mA, the total should be within limits.  But, there are a lot of "clone" Mini Pros out there, so it's possible that not all of them have a regulator rated at 150 mA.

Also, I noticed some people suggesting that the IRQ line needed a 330 Ohm pull up resistor where 10K seems to work fine for me.  At 3.3 volts, a 330 Ohm resistor is going to pull an additional 10 mA, which seems excessive, IMO.  Not enough to be a power drain, but it's possible the DWM1000 might have trouble pulling down to a reliable logic 0 with a 330 Ohm resistor.  This could result in erratic operation.  But, that's just a guess.

I did have a few shorts between pads when soldering the castellations on the DWM1000 module and this caused me a fair bit of frustration.  So, I suggest anyone experiencing problems should use a multimeter and check all pads for shorts, just  to be sure.

I'm in the process of fabricating another rev of the boards to fix a few issues.  I'll make the designs public when the new boards check out.

Wayne

greymfm, 2016-03-04:
@wholder: Great work! I think it will boost this project if there's a 'reference design' like yours.

wholder, 2016-04-09:
I've now completed a single PCB solution and published the PCB design and component list so you can build one (although you'll need at least two.)  The details are here:

  https://sites.google.com/site/wayneholder/uwb-ranging-with-the-decawave-dwm1000---part-ii

The completed boar looks like this and is designed to be powered by a cheap USB Battery Pack connected to the board's USB Mini Jack.

[image]

Wayne

greymfm, 2016-04-10:
Excellent work! My first thought was why not placing the antenna on the back side. Probably it doesn't make any difference for a good signal...

wholder, 2016-04-11:
I tried to design the board to fit the DecaWave specs, which say that the antenna needs a ground plane below and to the left and right of the antenna.  I actually cheated this a bit, as I didn't want to use as large a ground plane to left/right as they specified to keep the board size down and make it less expensive to fabricate.

[image]

Wayne

wazaahhh, 2016-04-12:
I have been trying to use DWM1000 with Arduino 101 (Intel Curie), with the same adapter board / connections as presented here : https://github.com/thotro/arduino-dw1000/tree/master/adapterBoard
It did not work at first, but at some point, I accidentally pulled the IRQ wire on the receiver board, and a piece of the message appeared. 

I surmise I have an IRQ problem, but I could not figure out what's going wrong.

Thanks for your help.

MikeS159, 2016-04-13:
@wazaahhh Which example program are you trying to run?

wazaahhh, 2016-04-13:
@MikeS159 I am using the basic sender / receiver .ino programs.

MikeS159, 2016-04-14:
As a first step so you know there are no errors with your board, can you run Basic Connectivity Test?

wazaahhh, 2016-04-14:
@MikeS159 yes, connectivity between arduino 101 and DWM1000 works great on both sender and receiver sides. I can even see that the sent message arrives at the receiver, but it's corrupted, either on the sender's or the receiver's side, but rather the latter in my opinion. I really suspect an IRQ problem on the receiver side.

khalil19, 2016-04-15:
@wazaahhh When you run sender normally you see messages : 
"ARDUINO delay sent [ms] ... "
"Processed packet ... #"
...
if not => your arduino is not sending anything...
I have the same problem with arduino Due.( I tested the IRQ alone and it works so it is not a interruption problem I guess..)
I tested BasicSender with an Arduino Nano and Uno and it works.. maybe because arduino 101 and Due are not AVR.

MikeS159, 2016-04-15:
The 101 & Due are 32 bit. 16 bit AVR code will still work, but you need to be careful with larger pointers causing overflows - https://github.com/thotro/arduino-dw1000/issues/73

Check any code such as the part below and try making the arrays 128 bytes.
`    char msg1[50];
	DW1000.getPrintableDeviceIdentifier(msg1);
	Serial.print("Device ID: "); Serial.println(msg1);

	char msg2[50];
	DW1000.getPrintableExtendedUniqueIdentifier(msg2);
	Serial.print("Unique ID: "); Serial.println(msg2);

	char msg3[50];
	DW1000.getPrintableNetworkIdAndShortAddress(msg3);
	Serial.print("Network ID & Device Address: "); Serial.println(msg3);

	char msg4[50];;
	DW1000.getPrintableDeviceMode(msg4);`

khalil19, 2016-04-15:
I did it with msg[128];
msg are showen in the serial monitor and it stops in ""Transmitting packet ... #""
and no interruption is detected by the arduino Due

lesjno, 2016-06-07:
Hello,
I see there is no news on this subject since 15 April 2016...
I'm currently trying to deal with 2 modules DW1000 and 2 Arduino Mega2560 : It seems to work on sender side , but not at the receiver side.
## Sender => 

_### DW1000-arduino-sender-test ###
DW1000 initialized ...
Committed configuration ...
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 0A, Short Address: 05
Device mode: Data rate: 110 kb/s, PRF: 16 MHz, Preamble: 2048 symbols (code #4), Channel: #5
Transmitting packet ... #0
ARDUINO delay sent [ms] ... 205395
Processed packet ... #0
Sent timestamp ... 342073.59
DW1000 delta send time [ms] ... 342.07
Transmitting packet ... #1
ARDUINO delay sent [ms] ... 12
Processed packet ... #1
Sent timestamp ... 1423762.50
DW1000 delta send time [ms] ... 1081.69
Transmitting packet ... #2
ARDUINO delay sent [ms] ... 12
Processed packet ... #2
Sent timestamp ... 2519538.25
DW1000 delta send time [ms] ... 1095.78_

## Receiver =>
_### DW1000-arduino-receiver-test ###
PinIRQ=2
Interrupt=0
DW1000 initialized ...
Committed configuration ...
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 0A, Short Address: 06
Device mode: Data rate: 110 kb/s, PRF: 16 MHz, Preamble: 2048 symbols (code #4), Channel: #5
Error receiving a message
Error data is ... Hell� � y� e�M=,y�����_ 

I performed to got the last 2 rows ("Error ...") once but when I reset again I only get details up to Channel: #5 ... tha's all.
For info, I tried it with and without a 1K resistor on the interrupt pin...

Any idea ?

Rotzbua, 2016-06-07:
Take my PR or set `setFrameFilter(false)` in dw1000.cpp

lesjno, 2016-06-07:
@Rotzbua Thanks for your answer, but could you explain?
What's the "PR" ? (Sorry ... I'm a newbie)
Regarding the SetFrameFilter(false) : do you suggest me to change the code within the setDefaults() funtion at the receiver side ?

lesjno, 2016-06-07:
To test, I changed the code in the setDefaults() funtion at the receiver side (setFrameFilter(false) in place of setFrameFilter(true) ).
Here is the result  =>
[image]

Clearly it's better because several are correct but a lot of them are corrupted ...

MikeS159, 2016-06-08:
@lesjno PR = pull request
They are changed made by someone waiting to be merged into the current version by someone with permission (usually the repository owner).

Rotzbua, 2016-06-09:
@lesjno Now it should work by default without changing setFrameFilter(), my pr #86 is merged

lesjno, 2016-06-09:
@Rotzbua Thanks for your work.
But when you say "#86 is merged", where can I get the final src files ? The last src files update has been done 4 days ago according to the internet page.
Nevertheless, I did the modification and I'm I a current situation where I get messages but there are a lot of corrupted messages.
I try the DW1000Ranging_TAG and DW1000Ranging_ANCHOR sketches : that works but this is not accurate at all ! I will open a new subject about that.

MikeS159, 2016-06-09:
> But when you say "#86 is merged", where can I get the final src files ? The last src files update has been done 4 days ago according to the internet page.

The changes were written and committed a few days ago, the pull request only got merged today, that's why it says nothing has been changed in 4 days.

Are you running one tag and one anchor for initial testing? (There can be issues where multiple anchors can cause corrupt data/incorrect range values)

lesjno, 2016-06-10:
@MikeS159
Yes I use only 1 TAG and 1 Anchor.

See my post (
https://github.com/thotro/arduino-dw1000/issues/6#issuecomment-224928428)
for result details on tests ...

MikeS159, 2016-06-10:
Are you using the most recent code? I have been having issues with it. (Problems [here](https://github.com/thotro/arduino-dw1000/pull/86) )
I am currently using code from [PR #80](https://github.com/thotro/arduino-dw1000/pull/80)

Maybe give that a go? I'll take a look at your other post and see if I can spot anything.

lesjno, 2016-06-10:
Yes, I use the PR #80 code.
I added a little correction in the DW1000.cpp (use
digitalPinToInterrupt(irq) in place of "irq")

*void DW1000Class::begin(int irq, int rst) {    // generous initial
init/wake-up-idle delay    delay(5);    // start SPI    SPI.begin();
SPI.usingInterrupt(digitalPinToInterrupt(irq));    // pin and basic member
setup...*

From my current tests, I can see the following :
- Several messages are lost. Lost at random but maybe periodical also...
(why ?)
- When TAG and ANCHOR do not move, the range average is not accurate :
There is of course a normal "noise" around the average, but the distance
(average) is not the same according to the configuration.
- I've seen another configuration test (from someone else with different
hardware) ; the AGC Tune3 (LEN_AGC_TUNE3 parameter in the DW1000.gcc row
281) value was not not the same : 0x0055 for him, 0x0035 for me (and you
;-) ). I tried the 0x0055 value with our code :result was better (more
stable). I tried several other values and I found the 0x0075 is the best.
But I don't have the DW100 full doc so I don't know what this parameter is
used for and how it can be managed...
- My test is done with 2 DW1000 modules. There is exactly 210cm between
them. When I walk in the room between the 2 modules, the distance average
change a lot and can be up to 40cm far from the good value !

MikeS159, 2016-06-10:
@lesjno 
I haven't needed to change any of the tuning parameters.
How stable are the ranges when they have a clear line of sight?

A few things to note:

- Objects obstructing the Deacwave chips can cause measurement errors (especially if they are close to the source)

- The measurements aren't that accurate at small distances (I never really put them closer than 2m)

- You may want to calibrate them. I put mine 3m apart, add code to ensure they read 3m at 3m (I subtract around 0.7m) then test again at other ranges to ensure they are linear

lesjno, 2016-06-10:
Ranges are not very stable...
I did a new test :
- I come back to the initial CAG Tune3 value (0x0035)
- I put the Anchor at 357cm from the TAG, clear line of sight
- I added an offset for calibration : +0.52m
result :
[image: Images intégrées 1]
Range is between 3.3m and 3.8m

I did a new test at 5m00 with the same calibration :
Result => main dist values are between 5.2 and 5.45 m => not really good !

I aslo observe a strange phenomenous :
I compute the exection time each time the "newrange() funtion is called. So
that I get the duration between 2 calls.
I observe the duration is about 101 to 103ms but sometimes I get 2X or 3X
this value :
[image: Images intégrées 2]
There are sevral random pics, but a lot of pics are clearly periodic :
every 20 loops.
Do you get the same ?

mic980000, 2016-08-07:
Dear All, I've managed to use the Ranging Tag&Anchor example sketch to print out some data. My question is, would that be possable to set different network ID to group up Tag&Anchors to work independently?
 Let say 1Tag 3Anchors as a group (Network 1) then i would have another set of group (Network 2) working in the same environment at the same time independntly without interference eachother.  I've tried to change DW1000.setNetworkId(2); in void setup() but with no luck, anyone could shine a light on me.. THX :)

wazaahhh, 2016-09-07:
@MikeS159 @khalil19 I finally fixed my problems, and tinkered a bit the Arduino 101 code to make it work with the sketches in this repo. Now ranging works well with Arduino 101 and I am moving forward with this design. Thanks for your help!

HappyDr0id, 2016-10-20:
Hi everyone, sorry for the nooby question but : can someone explain to me why we need a resistor in the pin layout ?
I read @ttaber talking about preventing a floating input on the pro mini, but I don't really get the meaning of it. Is it specific to the Arduino Pro Mini with its 5V output ? Do I also need a resistor if I use a Teensy 3.2 which is 3.3V ?

Thanks in advance for any clarification :)

AliSiddiqui4, 2017-02-09:
@greymfm when I run your TAG code I get garbage values, any ideas how I can fix this?

MikelGeelen, 2017-03-01:
@greymfm I had the same problem It was due to my power supply.

Maxai2, 2017-09-14:
@raffa1993  @EarthBreaker  @ttaber  @greymfm  @wholder  
Hello! Can anybody tell me how connect, decawave DW1000 TAG to Afrduino Uno? Which pins and which scetch give me correct ranging...
Thank u in advance!

Maxai2, 2017-09-14:
@Rotzbua 
I need a help, why u close!?

Maxai2, 2017-09-14:
@Rotzbua
What u talking about? May be there is found man, which help us! "Move to wiki", if wiki was help me, I dont wrote there! U dont know ok, so show me the way how can I fix my problem, and dont U spam!