thotro/arduino-dw1000 #57: Connect IRQ and RESET in dwm-1000 pin to Arduino Mega 2560

firdausidwiki, 2016-01-15:
I try to connect this dwm-1000 with Arduino Mega 2560. Aside from the SPI pin , which pin in Arduino Mega 2560 should connect to IRQ pin and Reset pin in dwm-1000 ? please help me, i'm researching this sensor for my final project, thanks before.

ttaber, 2016-02-06:
Here is an example wiring diagram for the Mega 2560 which mirrors that of the example using the Pro Mini:
[image]

firdausidwiki, 2016-06-20:
why the IRQ pin connects to pin 21 / SCL pin ?
can it be connected to the other pin ? because i use the SCL pin as well

reycreator17, 2017-01-30:
Hi @firdausidwiki 

Did it worked using arduino mega?

reycreator17, 2017-01-30:
@ttaber

Don't we need to use level shifter here since mega is running on 5V logic? 

Kinda new here so dont be harsh Thanks

firdausidwiki, 2017-02-03:
@reycreator17 actually i just search for the easiest ways to use dwm1000, because it is used for my final project in college, so i use it with pro Mini, i just dont have many times to continue research with arduino mega :')
I used arduino mega just for the data parsing from the pro Mini, because i need bigger memories to flash the program, and it works out fine

yes, you have to use level shifter from 5V to 3V3 if you want to connect with dwm1000, a little help for your question to ttaber. CMIIW

reycreator17, 2017-02-03:
can I see your schematics in fritzing. The DWM1000 --> pro mini --> mega and also the codes for each

Rotzbua, 2017-04-24:
A level shifter is required for mega. 5V logic micro controller are highly not recommended.

closed 2017-04-24 by Rotzbua