thotro/arduino-dw1000 #58: Problems with arduino Due!

khalil19, 2016-01-19:
Hi,
I'm testing the DWM1000 with an arduino Due but it doesn't work.
I think because of the port mapping...
Can any one help me please?.
Thanks !

ttaber, 2016-02-04:
Are you using a logic level shifter to step the Due from 5V down to 3.3V?
You might also want to take a look at the IRQ Pin which is (currently) statically defined as "0" in the DW1000RangingClass::initCommunication method:
`DW1000.begin(0, RST);`
and change that to whatever Pin you've got wired to interrupt line on the DW1000 or change the input parameters of that method to allow you to pass one in.

khalil19, 2016-02-04:
arduino due works with 3.3v
I change the RST, IRQ et Slave select..
the device is configured well but no inetrrupe occures ...

that was the Pb!

ttaber, 2016-02-04:
correct, the due will work with 3.3V input logic, but the DW1000 will not tolerate the 5V logic which is output from the due.

I've use both of the following logic level converters successfully with the DWM1000 and an ATMEGA2560 running at 5V:

https://www.adafruit.com/products/757

https://www.sparkfun.com/products/12009?_ga=1.60686436.874643147.1450744650

khalil19, 2016-02-05:
Hi, thanks for you response.
But arduino due works with 3.3V input and **output** (I verified it ...).
For the configuration it is ok : I run the basicsender and the DWM1000 is configured well but no interruption is done !

ttaber, 2016-02-05:
By default the library is defaulting to INT0 for the Arduino Pro Mini, which is physically PIN 2,
Fortunately with the Due, all digital pins are mapped to their same corresponding interrupt.

Unfortunately, you can't just wire up to PIN 0 on the Due because that's the same line that's used for the USB UART back to your PC.

You'll need to open up DW1000Ranging.cpp and, in the DW1000RangingClass::initCommunication method, you'll need to change the interrupt from 0 to the PIN you have wired up :
`DW1000.begin(0, myRST);`

Here's an example wiring diagram:
[image]

and the code would look like:
`DW1000.begin(2, myRST);`

I'm hopefully going to be setting a new pull request later with some refactoring that includes you to pass in the INT you want to use in the initCommunication arguments.

ttaber, 2016-02-05:
Pull #64 was merged up a little while ago.
This includes an overload for initCommunication that allows you to pass in the interrupt.
Using the same diagram above, your call to the method would be as follows:
`DW1000Ranging.initCommunication(9, 10, 2); //Reset, CS, Interrupt`

khalil19, 2016-02-09:
Hi,
I'm running basicsender. So I already modified pins.
And it didn't work.

EarthBreaker, 2016-02-11:
Hello there! I'm doing this with Arduino Nano, please refer to my pin connections as shown below:

[image]

I've also had problems running the BasicConnectivity test as it results in the following:

[image]

@ttaber, I've tried taking your advice and change DW1000.begin(0, RST); to DW1000.begin(2, RST); as you've said that I needed to change it to my current IRQ pin which is using D2 on the Arduino Nano. What I'd like to know is if my output is considered normal as Deca (Model) (Version) (Revision) are all 0. My unique ID is 00:00:00 as well. Do advise. Much thanks!

ttaber, 2016-02-11:
It looks like, in your diagram, your RST is tied to Pin 4 and you SPICSn is tied to Pin 9.
Based on that your code should look something like:
`DW1000.begin(0, 4); //INT -> Pin 2 = INT0, RST`
` DW1000.select(9); //SS `

Personally, I've been using the adapterboard to mount the DWM1000 but, just by eyeballing it, it looks like you've got all your power and ground connected up ok.

I know that some people have reported success with 5V logic, but I haven't been able to reproduce on a Mega 2560 without using a logic level converter to step down to 3.3V.

All that being said, you should be getting something like this if it's working correctly:
[image]

EarthBreaker, 2016-02-11:
@ttaber, much thanks for your advice! I've managed to replicate your results. Just two questions though. I was reading through the code just now, I didn't really understand what SS meant. In the code, the comment was:

void DW1000Class::select(int ss) {
	reselect(ss);
// try locking clock at PLL speed (should be done already, 
// but just to be sure)

So in that sense, SS basically refers to the Pin connected to your SPICSn?

Second question! Just to be sure I understand it right:
DW1000.begin(0, RST); 
//RST is basically the Pin connected to your RSTn on DWM1000?
//0 is basically the Pin connected to your IRQ on DWM1000?

ttaber, 2016-02-11:
SS stands for Slave Select. You'll also see it noted as CS (Chip Select) hence SPI**_CS_**n.
They both effectively mean the same this.

Here's how the SPI communication works in a nutshell:

- Your MCU (the Nano's ATMEGA328) is going to pull your CS/SS pin low to initiate the conversation with the DW1000
- It then shoves a bunch of bytes down the MOSI (Master Out, Slave In) line and, for each byte it sends, it gets one byte in return from the DW1000 on the MISO (Master In, Slave Out) line.
- It repeats this process several more times and stores each one of the received bytes in an array as it loops through the total number of bytes in a single communication cycles. 
- The MCU then takes the CS/SS line back high again and that communication cycle is over.
- The code then kicks in an evaluates the bytes array and decides what kind of information was received back and how to react to it.

As for your second question, yes the RSTn should map to the physical pin you're connected to on your board.

The IRQ parameter though doesn't refer to the physical pin connected to the IRQ on the DW1000.
It refers to the actual external interrupt assigned within your MCU.
On the Nano, Uno, Mini, etc. the physical Pin 2 maps to Interrupt 0 and Pin 3 to Interrupt 1.
[image]

This is really chip dependent and should be detailed on the pinout map of that particular board.

If you're not able to locate a pinout, and it's an "Arduino" supported board, you can call:
`digitalPinToInterrupt(int physicalPin);`
which will resolve to the actual interrupt mapped to that physical pin (if it exists).

Conversely, on some boards like the Zero, which runs a 32 bit M0 Cortex MCU, every pin can be mapped to an external interrupt and the interrupt address matches the corresponding pin number

You can take a look at the AttachInterrupt page in the Arduino reference for more examples:
https://www.arduino.cc/en/Reference/AttachInterrupt

EarthBreaker, 2016-02-11:
Ohhhhhhh! Yes! That makes a lot of sense now. Thank you so much. You replies are very detailed, I've learned quite a bit today. Mind if I ask you some questions on the BasicSender and BasicReceiver program?

ttaber, 2016-02-11:
No problem at all. Why don't you go ahead and post your questions on issue #54 which is already opened around BasicSender/Receiver .
I think @raffa1993 was posting something about this in another issue earlier today so maybe I could answer questions for you both.

pakizip, 2016-06-02:
Hi, i'm going crazy.

I have 2 Arduino Due and the BasicConnectivityTest works fine:
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 0A, Short Address: 01

But anyway no luck with any example, I tried to send, receive and tried all the examples but nothing, no a single interrupt, error, or message.

Can you please help me? The connections is the same of https://github.com/thotro/arduino-dw1000/issues/58#issuecomment-180477907

Thanks

MikeS159, 2016-06-02:
Have you tried DW1000Ranging_ANCHOR & DW1000Ranging_TAG?
Those are the two examples (other than basic connectivity test) that I know work. I've had issues with all the other ones.

pakizip, 2016-06-02:
Well yes, already tried but nothing.
Is there something to debug? For example can I read some register to understand if it's working or not?
Or maybe something to debug TX and RX and focus only on wrong one.

Ty.

Rotzbua, 2016-06-02:
@pakizip your spi connection do not work well, cable to long or too close and parallel (electromagnetic induction)

>  Network ID & Device Address: PAN: 0A, Short Address: 01

should be:

> Network ID & Device Address: PAN: 0A, Short Address: 05

because: 

>  DW1000.setDeviceAddress(5); // short addr
>  DW1000.setNetworkId(10); // PAN -> dec 10  = hex 0a

pakizip, 2016-06-02:
I changed the address, just to be sure I tried to run the BasicSender and it shows:

                       ### DW1000-arduino-sender-test ###
DW1000 initialized ...
Committed configuration ...
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 0A, Short Address: 05
Device mode: Data rate: 110 kb/s, PRF: 16 MHz, Preamble: 2048 symbols (code #4), Channel: #5
Transmitting packet ... #0

Rotzbua, 2016-06-02:
Ok, than wiring is correct.

Do you use `DW1000.begin(2, myRST);`? `0` is not correct anymore, I fixed this bug.

pakizip, 2016-06-02:
If you're talking about the DW1000Ranging class, yes. It looks like:
    DW1000.begin(2, myRST);
    DW1000.select(mySS);
In initCommunication method.

I can't understand, I can access the DWM1000, read its values but I cannot write? I was thinking maybe is a writing issue, what can I write and then read it just to be sure the writing succeeded?

I have also a Leonardo and a shifter, can I use this one to test at least the sender?
Can I follow this schema?
DWM1000 RSTn (via level shifter) -- Arduino D9 output 
DWM1000 SPICLK (via level shifter) -- Arduino D13 output
DWM1000 SPIMISO (via level shifter) -- Arduino MISO D12 input
DWM1000 SPIMOSI (via level shifter) -- Arduino MOSI D11 output 
DWM1000 SPISS (via level shifter) -- Arduino SS D10 output

Thank you very much for your time.

pakizip, 2016-06-03:
Opened another issue: https://github.com/thotro/arduino-dw1000/issues/87

khawajamechatronics, 2016-08-10:
@pascalbros 
Hi, Initially I had same issue as you have that is Basic test is working fine but anchor and tag code wasnt working correct. 

Please after uploading code, press reset both so they could sync and you will see output on serial.

Please make sure you have correct baud rate as basic connectivity code has baud 9600 and anchor/tag code had baud rate 115200

az-z, 2016-09-06:
@khawajamechatronics 
what did you mean "Please after uploading code, press reset both so they could sync and you will see output on serial." ? which "reset" to press ?
I'm stuck with UNO unable to communicate (issue #113)

Phenioxrj, 2016-09-07:
@ttaber What are all the changes you have to make to the MEGA to get it to work. I'm using logic level converted and it works perfectly with the 5vpro-mini. The mega and the anchor are talking but not ranging.