thotro/arduino-dw1000 #59: Rx & Tx LEDs

wholder, 2016-01-24:
It would be nice to have a way to enable the Rx/Tx for debugging purposes.

Rotzbua, 2016-09-10:
I found some code at https://github.com/bitcraze/libdw1000 .
@wholder maybe you can make a pull request

curlyz, 2016-11-15:
@wholder I read some discussion on Hackaday that the Atmel chip is not capable enough to work with the DWM and do the trilateration at the same time , they are currently working with STM32 and I was thinking is it possible that you make another PCB that work with this module 
http://www.ebay.com/itm/STM32F103C8T6-ARM-STM32-Minimum-System-Development-Board-Module-For-Arduino-NJIU-/222082706313

kuek, 2016-11-16:
curlyz, i guess it depends on where you want to have the localization data. if you are looking for a device to track and want to have the localization knowledge on a server its good to have the localization engine running on the server. so all anchors should dump the ranging data to the tags the a server.

if you want to localize the tag itself, it would be sufficient to do the ranging there. anyhow, i implemented my triangulation in just some lines of MATLAB code. havent ported it to the arduino, because I need central knowlegte of the tags... but it can probably be done.

I have done a PCB for a arduino 328p. but if you are interested in a PCB for an STM32, no problem...

curlyz, 2016-11-17:
Is it the higher frequency the board has, the more accuracy ranging can be made?. I don't know how timestamp work yet, can anyone help me with documents.

Rotzbua, 2016-11-17:
http://www.decawave.com/support

cimbalo, 2016-12-30:
I added minimal support for GPIO led blinking, see pull request #144

closed 2017-01-04 by thotro