thotro/arduino-dw1000 #60: Please help! How to run Range GUI on Windows

mrAyaz, 2016-01-30:
Hello All. Im downloaded webGui app but i can not run java files on my PC

ttaber, 2016-02-06:
Are you referring to the DW1000Server that @jogando put together and mentioned in #45 ?

mrAyaz, 2016-02-06:
Yes.

mrAyaz, 2016-02-06:
HOW CAN YOU HELP ME?

ttaber, 2016-02-06:
I'd need a little while to get my boards built to support this.
As of right now, only my tag has WiFi built in and I'd need to get my anchors setup with some spare ESP8266 modules so they could connect to the Java sever as well.

If your hardware is already setup, it looks like the instructions @jogando provided are pretty straightforward.

Clone the repo locally, modify the config file with the correct device IDs and paths, compile with your favorite Java IDE, fire up the server's compiled JAR file via command line using the arguments specific, and launch the html page that makes the GET requests to the Java server.
You'll also need to make sure you swap out your newRange code on both your Tag and Anchors with the code provided in #45 so that your device sends the telemetry to the webserver in the correct format.

mrAyaz, 2016-02-07:
This is support multiTag mode?

ttaber, 2016-02-08:
As of right now, the library current supports multiple Anchors and a single Tag.

If you try to start multiple Tags, the devices will fight over communication with the Anchors and result in a lot of "Not Found" errors being thrown as they're receiving inbound requests from Anchors they don't have registered in networkDevices because they expire due to communication with the other Tag(s).

I've not done a lot of research on what it would take to adapt this library to allow multiple tags using a ToA/ToF method.

Perhaps @leosayous21 or @thotro might be able to point you in the right direction if that's a path you're wanting to pursue.

Personally, I would definitely be interested in put in some dev time to support multiple tags, but I feel as though it would be too complicated to coordinate the "time allocation" between all the tags to allow them to complete a full rangingcycle and the library would need to be re-worked to turn the Tags into "beacons" and support a TDoA method of ranging.

mrAyaz, 2016-02-08:
OK Thanks

closed 2017-04-26 by Rotzbua