thotro/arduino-dw1000 #61: Using library with EVB1000

agolovenko, 2016-02-02:
Hi, I really appreciate your project. It's not an issue but more of a question. Is it possible to use this library for Arduino + EVB1000? 

I’ve  managed to connect SPI pins (SPICSn, SPIMOSI, SPIMISO, SPICLK) but couldn't find the way to connect RSTn and GPIO8. In fact I don't see RSTn and GPIO8 outputs on EVB1000 at all. Not sure how crucial are they in order the things to work properly (I'm kinda new to all this stuff). But anyway the basic example seemed to work while transmitter hangs.

Thanks, Ashot.

ramdip, 2016-03-04:
Hi, I've got the same problem. Did you find any kind of solution?
Help :( 
Thanks, Rams

MikeS159, 2016-05-09:
When you say the EVB100, do you mean [this](http://www.decawave.com/products/evk1000-evaluation-kit)?

What are you trying to do with an Arduino and one of these. They already come with a ARM processor, so you don't need an Arduino to go with them.

ramdip, 2016-05-09:
Hi, yes i used this Board. I try to work with arduino because i try to work
with Evb1000 without his arm processor. The arm processor is substitute
with Arduino... But it doesn't work. I think that the problem is the pin
reset.

MikeS159, 2016-05-10:
Is there a reason you want to get rid of the ARM processor?

ramdip, 2016-05-10:
Yes... It is a project for the  university!

juwangvsu, 2016-09-01:
on another note, if I program the anchor example to a dwm1000 + arduino mini pro, will it work with a evb1000 tag? 

I was testing such a setup with no result. But mine dwm1000 might have some bad soldering issues...

abencomo, 2016-10-11:
@juwangvsu, were you able to make the [DWM1000 + arduino mini pro](https://sites.google.com/site/wayneholder/uwb-ranging-with-the-decawave-dwm1000---part-ii/IMG_6229.JPG?attredirects=0) to work with the EVB1000 boards?

For my project, I'm installing the DWM1000 module on a small quadcopter and trying to use the EVB1000 boards as anchors for indoor location, but it's not able to receive or send data from/to anchors.

juwangvsu, 2016-10-11:
have not try that yet. I have some technical problems with dwm1000 + mini pro. it was working a few days ago: one programmed as RangingTag and the other programed with RangingAnchor, but not any more.
hopefully not hardware problem.

abencomo, 2016-10-11:
@all, has anyone be able to use a dwm1000+mini-pro to receive/send anchor distances from the EVB1000 boards using the thotro library?

Rotzbua, 2017-04-26:
@agolovenko Can I close this?

closed 2018-03-25 by agolovenko