thotro/arduino-dw1000 #62: Demo-Videos

greymfm, 2016-02-03:
Just wanted to share what is possible using this great DW1000 Arduino ranging library :-)

https://www.youtube.com/watch?v=trp2rDtK4SQ
https://www.youtube.com/watch?v=SwiUzBPI0zg

Feel free to add links to the Wiki.

thotro, 2016-02-04:
That's awesome! Many thanks for sharing! Glad to hear that it was fun! ;-)
Will put your vids into the wiki.

leosayous21, 2016-02-05:
Very nice ! this is really promising ! 
How did you calibrate the anchor/tag to have a "real distance" displayed ? just by subtracting an offset ?

greymfm, 2016-02-05:
I did not calibrate yet, just using the raw delivered values. Actually, for this application it is not necessary to have cartesian correct measurements, it's only important to have always the same logical position A at physical position B, so like a constant transformation C).

However, I plan to add calibration because measurements could drift over time. Later (if the ranging protocol will support this), I would like the tag (installed in my robot) to ask all 3 anchors for their measurements among each other. This would allow me to add auto-calibration (so computing the correct offset for each anchor) all the time. 

Do you think it would be difficult to add this to the ranging protocol? (somehow knowing the distances between all anchors).

leosayous21, 2016-02-05:
I don't think it will be difficult because i designed the Ranging library to allow dynamic modification from TAG to ANCHOR or ANCHOR to TAG but it needs some improvement. :-)

leosayous21, 2016-02-05:
why did you put the devices horizontally and not vertically ?

greymfm, 2016-02-05:
Good question - I did some orientation tests with them before long time ago, as far as I remember the measurements were less dependant by direction when in horizontal orientation (though I could be wrong here...)

leosayous21, 2016-02-05:
Yes I had really strange value when it was in vertical orientation but the received power was better ...

greymfm, 2016-02-05:
I checked the DWM1000 datasheet (p. 15/16), you are right - around the Azimuth Plane (that you are using), the signal seem to be stronger than around the Elevation2 Plane (that I'm using). I should try vertical orientation again :)

leosayous21, 2016-02-05:
I think so, but it was really not accurate: the values where really different just in turning the device among the "Z" axis but in keeping the same distance between them... It frustrated me so much :-(

raffa1993, 2016-02-11:
Hi greymfm,
you've made a really great job!
I need to calculate the distance between to Arduino:
When I download the "Basic Connectivity Test" all seems to work,
but if I use DW1000Ranging_TAG.ino and DW1000Ranging_ANCHOR.ino nothing happens.

Have you modified in any ways one or both softwares?
If you could help me I would appreciated it a lot.
Awaiting Reply
Raffaele

EarthBreaker, 2016-02-14:
Hi there @raffa1993! I'm trying to solve that problem as well. Now currently using an Arduino Pro Mini since that's the medium being used by the library owner. I don't suppose you're in the Decawave group as well?

raffa1993, 2016-02-15:
Hi @EarthBreaker,
Yes I'm in the Decawave Group on Google Groups.
I'm trying to set up a TWR connection for a very interesting project,
but I'm stopped right now, because of this kind of problems.
Now I will try with Arduino Nano and 2 Logic Level Converter
I hope I will get some result, I'll let you know ASAP

EarthBreaker, 2016-02-15:
@raffa1993, yea, I'm in that thread of yours, as 69.eddy. Hopefully we can get this up and running.

bigredbee, 2016-02-15:
One thing that's always been unclear to me is:  where does all of the data get aggregated?.  In the RangingAnchor / RangingTag examples, it's the Anchor that prints out  the range.   But in this video, it looks like the tag (attached to the laptop) is somehow getting all of the ranges from each of the anchors?

Is it possible for all of the ranges to be collected by one of the anchors? How can the tag get copies of the ranges from each anchor?

greymfm, 2016-02-16:
Both anchor and tag print out the range (I'm using a simple binary serial protocol for the tag serial data). The tag is attached to the laptop, and the laptop receives the ranges (to all anchors) from the tag. So the tag periodically outputs something like this:
range to anchor 1:  14m
range to anchor 2:  3m
range to anchor 3:  18m

bigredbee, 2016-02-17:
My tags (using the latest source from Github) don't print anything -- did
you have to add any code to make that happen?

Also, assuming that the Anchors are "fixed", is it possible for one of the
anchors to collect all the data instead of the tag?  Seems like a better
configuration that putting heavy compute requirements on a tag.

@AlexanderG, if you could contact me off-list, that would be wonderful!

greg at bigredbee.com

greymfm, 2016-02-17:
Well, the TAG should print something, have a look at the code (function "newRange" is called for each new measurement):
https://github.com/thotro/arduino-dw1000/blob/master/examples/DW1000Ranging_TAG/DW1000Ranging_TAG.ino

Regarding the compute requirements, you can do all calculations on an Arduino - I just did use a PC because I prefer using PC for prototypes (faster coding time, visualization etc).

bigredbee, 2016-02-17:
Thanks -- I was using the wrong example

mrcoolkaran007, 2016-03-11:
When I run basic connectivity program, i get a output like this - 
DW1000 initialized ...
Committed configuration ...
Device ID: DECA - model: 255, version: 15, revision: 15
Unique ID: FF:FF:FF:FF:FF:FF:FF:FF
Network ID & Device Address: PAN: FFFF, Short Address: FFFF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5

So I am just getting FFFF. I am using arduino pro mini. Supplying power from pro mini.

mrcoolkaran007, 2016-03-11:
When I run basic connectivity program, i get a output like this - 
DW1000 initialized ...
Committed configuration ...
Device ID: DECA - model: 255, version: 15, revision: 15
Unique ID: FF:FF:FF:FF:FF:FF:FF:FF
Network ID & Device Address: PAN: FFFF, Short Address: FFFF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5

So I am just getting FFFF. I am using arduino pro mini. Supplying power from pro mini.
Here is the code i am running.
https://github.com/thotro/arduino-dw1000/blob/master/examples/BasicConnectivityTest/BasicConnectivityTest.ino

khawajamechatronics, 2016-08-10:
@karanmakharia Were you able to solve the issue, as I have same output with a module.

I got two modules working perfectly fine. But third had same output as above. I feel that there was issue with power that crossed 5V and now its malfunctioned.

abencomo, 2016-10-10:
@khawajamechatronics please report back if you figure out the problem.  I have experiencing the same issue as well.

khawajamechatronics, 2016-10-10:
@abencomo 

No, I have third one not functioning I have asked for replacement, For proper work please make sure the following

Power Supply and I/O Level doesnt exceeds 3.3V

VDD and VDDON of DWM1000 connected to 3.3V

soonweng111, 2016-11-19:
@greymfm Hello, I am wondering how can I make one of the anchors to actually read the same readings from the tag itself. Is there any way I could do that? I could read 3 anchors from the tag but not the distance from tag by the anchor itself.

AliSiddiqui4, 2017-01-14:
@greymfm how are you getting 3 ranges on the tag. I just get ranges from 2 anchors when 3 are connected to a single tag. 
Furthermore are you using MATLAB for the triangulation gui in the video, if so then how are you sending all 3 ranges from the arduino to matlab?

Rotzbua, 2017-04-24:
@greymfm Can you add your results to the wiki? Thanks :)