thotro/arduino-dw1000 #63: Maximum/minimum stable range measurements

thotro, 2016-02-04:
Hey everybody,

I was told that stable measurements between 45cm and 70m are possible with the current driver version. More should be possible I think, so obviously the driver (or wirings) need improvement ;-)

What maximum/minimum measurements did you achieve with your boards?

Thanks and cheers,
Thomas

ttaber, 2016-02-04:
When running 4 Anchors and 1 Tag on Arduino Pro Minis (8Mhz / 3.3V), I've seen accuracy within 35 - 40cm when using the DW1000Ranging_ANCHOR / DW1000Ranging_TAG examples and commenting out all code in the newRange of the Anchors (not sure if that has anything to do with it...?).
I also created another Tag using an Arduino Zero running a 32-Bit ARM Cortex M0+ @ 48Mhz / 3.3v and saw sustainable accuracy levels around 25 - 30cm.

kshitijdadhekar, 2016-03-11:
@ttaber  How much can you expect with 1 anchor and 1 tag in an outdoor environment?

MikeS159, 2016-03-15:
With 1 anchor and 1 tag (using the DWM1000) I have been getting line of site ranges to about 25m, with accuracy around +/- 30cm.

kshitijdadhekar, 2016-03-15:
@MikeS159 Thanks for replying, I am getting just the same. Also can you tell me one more thing. Now I want multiple Anchors and a Single tag. What modifications are required in the DWM1000_Anchor and respective Tag program. That would be really helpful!

MikeS159, 2016-03-16:
@kshitijdadhekar There are other example files which are used for multiple anchors (I believe), although I haven't go this working yet as I have only just built enough hardware. There are other questions about this, I suggest looking though some of the other issues to see how other people got it working.

kuek, 2016-11-13:
hi guys, 
made some tests with a single anchor, just to see the internal offset for my location system. see pics attached. I just tested low range, but will do further testing outside.

then i moved over to 3 anchors, calibrated them and measured and recorded the raw data. in the plots below you can see some position measurements I made with my localisation code. the real position in the plot was 1m/1m. 

hardware:
[image]

offset test:
[image]

localisation
[image]

[image]

kuek

AliSiddiqui4, 2017-02-06:
@kuek 
Which code are you using for localization the same one from the given library example? Also on which platform have you implemented the above localization/transliteration?

kuek, 2017-02-06:
Hi Ali,
wrote my own code... you can find it [here ] (http://www.localino.net) 
as of today its implemented in python. But I have already thought to put it to the tag uC.

kuek

AliSiddiqui4, 2017-02-06:
@kuek thank you and what is the maximum range you are getting from a single anchor?

kuek, 2017-02-06:
@AliSiddiqui4 depends on the environment ;) you can see some results [here](https://hackaday.io/project/18296-localino-open-source-indoor-localization-system)

AliSiddiqui4, 2017-02-07:
@kuek is this software fully open source. I mean can I just download and run it using the hardware similar to this library?

kuek, 2017-02-07:
@AliSiddiqui4 it works with our software and hardware as explained on our [website](http://shop.localino.net).

AliSiddiqui4, 2017-02-07:
@kuek I just saw the instructable of the making the board and running the software. It seems I will have to design the PCB as it is not open source. Can you please confirm if the connection of the PCB are same as that shown below. Also are you using the DWM1000_Tag/Anchor Code on the arduinos from this library example? 

[image]

kuek, 2017-02-07:
@AliSiddiqui4 I can give you some of our boards, so you can save yourself a lot of time...

AliSiddiqui4, 2017-02-07:
@kuek are you referring to your localino website store where the boards are available for purchase?

skadiexe, 2018-02-14:
my results with DWM1000
channel 1, 110 kb/s, 2048 preamble lenght
range test LOS outdoor:
226m maximum range i got, signal lost
195m signal achived
80m stable connect without loss of the packets
20m stable connect from behind human body