thotro/arduino-dw1000 #67: Fast and Cheap Level shifter 5V -> 3.3V

maxicastle, 2016-02-21:
Hello friends, thanks a lot for the initiative to create this library (really useful), I only want to share a different way to communicate your arduino 5V Pro-mini/UNO/Mega/etc with the 3,3V DWM1000 module.

It's really simple if you need to communicate your module with your 5V system, but you don't have a level shifter (Chip or module) and you can't get it or need a fast solution you can use the next voltage divider on each ARDUINO_OUTPUT_PIN and connect the output of the voltage divider to the DWM1000_INPUT_PINS, and connect the DWM1000_OUTPUT_PINS(3.3V) directly to your arduino board.

[image]

is not forced to use these exact values the only condition that you need to satisfy is:

> 5V*(R2/(R1+R2))=3.3V  or R1=0.5151*R2

Example , for the next diagram this is te I/O configuration

[image]

DWM1000     ||   Arduino  
3 (RSTn)	           Pin 9 Reset (OUT)
13 (SPICSn)	   Pin 10 SPI chip select (OUT)
14 (SPIMOSI)	   Pin 11 MOSI (OUT)
15 (SPIMISO)	   Pin 12 MISO (IN)
16 (SPICLK)	   Pin 12 SPI clock (OUT)
17 (GPIO8)	   Pin 2   interrupt asserted line  (IN)

[image]
[image]
[image]
[image]

Works really well !!!

MikeS159, 2016-03-02:
Potential dividers are a simple way to make a single-direction level shifter. Can I suggest that hardware advice be kept to the wiki rather than the issues page?
I created a page [here](https://github.com/thotro/arduino-dw1000/wiki/Hardware) where everyone can list hardware that they have used successfully.

pakizip, 2016-06-05:
I think your configuration is wrong, because in all the pictures there's a wire (pin 12 on arduino) that is attached directly and the voltage is not shifted.

Maybe I'm wrong eh. I tried your configuration and i'm not able to connect and read the DW.

R2D2RU, 2016-08-10:
16 (SPICLK) Pin D13 SPI clock (OUT)

az-z, 2016-08-10:
How exactly that works?

Rotzbua, 2017-04-24:
@maxicastle Hi, please move your idea to the wiki. Thanks for your help:) !