thotro/arduino-dw1000 #69: Basic connectivity test fails

mrcoolkaran007, 2016-03-11:
I used arduino pro mini 3.3V and CP2102 with DWM1000. The power input given to both pro mini and DWM1000 is 3.3V. Still when I run basic connectivity test, i get this output. Please tell me what can be wrong with my arrangement.
```
DW1000 initialized ...
Committed configuration ...
Device ID: DECA - model: 255, version: 15, revision: 15
Unique ID: FF:FF:FF:FF:FF:FF:FF:FF
Network ID & Device Address: PAN: FFFF, Short Address: FFFF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5

```

mrcoolkaran007, 2016-03-15:
This was due to improper connection. After making proper connection everything worked out. Now I just want to know how to connect multiple anchors to one tag.

closed 2016-03-15 by mrcoolkaran007

khawajamechatronics, 2016-08-10:
Hi,

Can you let me know what issue was with the connection, as I have same output with a module.

prvn194, 2017-06-28:
hello karanmakharia  ,
i have having the same issue ,
please by saying connection what u have changed in your connection