thotro/arduino-dw1000 #72: I couldnt get the distances between 2 different arduino

LanceKing1803, 2016-03-18:
First test 
I have 1 Arduino Uno acts as a Tag and 1 Arduino Uno acts as an Anchor. I got distance readings from the two.

[image]

Second Test
I have 1 Arduino Uno acts as a Tag and 1 Arduino DUE acts as an Anchor. I cannot get distance readings from the two
[image]

Can someone give me an suggestion on how to fix this ?

MikeS159, 2016-03-30:
Do you have a second DUE to test on?
There have been some issues around 32-bit processors overflowing some statically sized arrays. You could try adding extra serial messages to determine exactly where the program is breaking.

LanceKing1803, 2016-03-30:
Unfortunately i don't have a second DUE to test on. However i got it working by using another Pro mini as an anchor and communicated to the DUE through I2C.

pakizip, 2016-05-20:
Can you provide the source code? Thank you.

sarwadenj, 2017-01-09:
@LanceKing1803 
i'm using Arduino Nano both side and i made all connection but

i'm running DW1000Ranging_ANCHOR and DW1000Ranging_TAG and getting

device address: 82:17:5B:D5:A9:9A:E2:9C
### ANCHOR ###

not getting range or any data...

hu265, 2017-03-03:
I am using Arduino UNO for both of my Anchor and Tag and having the same issues. Did anyone manage to figure out how to fix it?

Rotzbua, 2017-04-26:
@LanceKing1803  @hu265 Any update?