thotro/arduino-dw1000 #74: Strange programming of library

Rotzbua, 2016-04-15:
Why is everything in `DW1000.h` static? Is there a reason for programming it as a singleton?

MarioRuz, 2016-07-10:
I also thought about that @Rotzbua , especially point 1. Maybe @thotro can tell us that.

Thanks!

thotro, 2016-07-10:
Hi there,
both very good questions, indeed: neither the use of `static` nor the explicit use of `short` relates to a particular design decision ;-)

So, you're right, the `static` declarations should better go and see the garbage bin.

The `short`s are not really bad of course, but here their actual purpose is questionable. I'm indifferent regarding them.

Thanks and cheers,
Thomas

MikeS159, 2016-07-12:
Although I agree there should be justification for making something `static`, static functions do save a small amount of memory. The trade off isn't normally worth it, but in the case of a micro-controller it might be. The Ranging Anchor/Tab takes about 70% of program memory on a Uno, and adding my code on top I was getting to 97% before I started making some optimizations and design changes.

Rotzbua, 2017-04-24:
Solved.

closed 2017-04-24 by Rotzbua