thotro/arduino-dw1000 #77: Adding TDOA (time difference of arrival) feature

greymfm, 2016-05-07:
The Arduino library currently supports the ToA (time of arrival) feature of the DW1000 modules which uses two-way-ranging.

In additional to that it would be very useful to also use the TDOA feature of the DW1000 modules, e.g. if we could connect 3x DW1000 modules and get the time difference of the received signal between them. This would allow us to do direction finding (knowing the angle of the sender). Thereby, in addition to distance we would also get an angle, thus reducing the number of senders for position estimation (localization). In other words, we would not need 3 senders (3 distances), just 1 (1 distance, 1 angle). 

Combining both methods (TDOA optionally) would be very intesting. 

What do you think?

Rotzbua, 2016-05-07:
Looks similar to: https://github.com/lab11/polypoint

wql7013, 2017-04-21:
They both don't work as well as we expect, because of the short distances between each two antennas, we tested. The errors seem to be caused by the inner circuit of the IC or the signal detection algorithm rather than the environment surroundings or the clocks. We connected two DW1000 ICs to the same external reference clock, and  the standard variation of the time difference of arrival between them was about twice larger than that of the time of flight calculated by TWR, mathematically it implied that the errors of estimated timestamps of arrival of the same signal between two different sensors are independent.

Rotzbua, 2017-04-26:
Please move the information about TDoA to the wiki. Thanks.
duplicates #47 #77 #108 #128 #130 #152