thotro/arduino-dw1000 #78: Problems to get started with arduino mini pro

DF9BI, 2016-05-09:
Hello,
I made some tests with DW-1000 and arduini mini pro and have some strange behaviour, and hope, someone can help.

When I try BasicConnectivityTest, I get the following:

DW1000 initialized ...
Committed configuration ...
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 0A, Short Address: 05

When I call DW1000.getPrintableDeviceMode(msg), the arduino performes a reset.

MikeS159, 2016-05-10:
Could you provide a diagram of how the Arduino is connected to the Decawave. (Is it based on the example in this project?)

DF9BI, 2016-05-10:
Hello Mike,

the wiring is the same as in this project as in AdapterBoardTestBed.png. 
But I made my own adapter PCB. 
Before, I soldered the wires directly to the DWM1000. Same result.

I use arduino pro 3.3V with 8 MHz.
Power comes from the 5V USB connector over a TSL1117 3.3V regulator. Power line is ok, no drops in voltage.

Rotzbua, 2016-05-10:
> Device ID: DECA - model: 1, version: 3, revision: 0

Model and version correct, so it seems fine.

> When I call DW1000.getPrintableDeviceMode(msg), the arduino performes a reset.

msg must be a minimum size of char[128]. If msg is too huge you run out of ram.

DF9BI, 2016-05-10:
Hello Rotzbua,
that seems to be a good hint. Now I can call DW1000.getPrintableDeviceMode(msg) and I get: 
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5

I have to check the other examples.

DF9BI, 2016-05-11:
Hello,
now it is working. I programmed one as tag and the other as anchor with the examples. I had some problems with the timing. I had delay(100) in the loop. With this, i did not get any ranges. I removed all delays and it works.
There is on problem remaining, the short adress of the anchor changes every time when I switch power on and off. So, I have to find how I can indentify the anchors by a fixed ID.

thank you for the hints and

best regards

Stefan DF9BI

greymfm, 2016-05-11:
I modified that part in the library. Do not use the random function, use static addresses, so you know your anchors: 

void DW1000RangingClass::startAsTag(char address[], const byte mode[]){ 
   ... 
  _currentShortAddress[0]=address[0]; 
  _currentShortAddress[1]=address[1];﻿

DF9BI, 2016-05-11:
Hello greymfm,

thank you for the hint. It works, but I don´t see, how the bytes are converted. 

In the example, the DWM ist started with:
DW1000Ranging.startAsAnchor("82:17:5B:D5:A9:9A:E2:9C", DW1000.MODE_LONGDATA_RANGE_ACCURACY);

And I get as the short adress: 3238

I don´t find this bytes in "82:17:5B:D5:A9:9A:E2:9C",
only if I take the first two chars in the string, 8 and 2 what is 38H and 32H would make sense.

DF9BI, 2016-05-11:
Ok, thats it. 
I changed the adress string to "00:17:5B:D5:A9:9A:E2:9C" and now, the short adress is 3030.

I get the hex-value of the first two characters in the string, not the first two bytes.

DF9BI, 2016-05-11:
So, I changed the following lines:
_currentShortAddress[0]=_currentAddress[0]; //random(0,256);
_currentShortAddress[1]=_currentAddress[1]; //random(0,256);

Now, I get the first two bytes of the addressstring as short adress

AliSiddiqui4, 2017-01-13:
I am getting this error after uninstalling my arduino 1.6.13 and then installing 1.6.8. 

avrdude: verification error, first mismatch at byte 0x0002
0x8b != 0x8e
avrdude: verification error; content mismatch

closed 2017-04-26 by Rotzbua