thotro/arduino-dw1000 #79: Ranging : TAG/ANCHOR address should not be random

greymfm, 2016-05-11:
It is difficult to identify anchors if their addresses is random. Especially, if there is not wired connection between them. 

There should be an option for a static address.

MikeS159, 2016-05-11:
This is on my list of things to do next week :)

mic980000, 2016-08-11:
I've got this sorted! ha..haa...EASY!

mic980000, 2016-08-11:
[image]

manomitbal, 2016-08-16:
Is there a method of administering static addresses/names to the anchor/tag as we flash the devices?

curlyz, 2016-11-10:
have you made it yet Mike , we count on you a lot =))

MikeS159, 2017-04-19:
Sorry for never replying, I stoped working on this project some time ago. In DW1000Ranging.cpp there are two functions `DW1000RangingClass::startAsTag` `void DW1000RangingClass::startAsAnchor`

In these the following code creates the address
`	randomSeed(analogRead(0));`
`	_currentShortAddress[0] = random(0, 256);`
`	_currentShortAddress[1] = random(0, 256);`

I simply changed the function to take the last two bytes of the long address `currentAddress` which is set when you start a tag or anchor.

Rotzbua, 2017-07-10:
Changed by @gautierhattenberger . Thanks.

closed 2017-07-10 by Rotzbua