thotro/arduino-dw1000 #81: Maximum Number of Devices and Ranging Frequency

sinakahnemouyi, 2016-05-24:
Hi,

First of all I wanted to thank you for the great work. I am using the DW1000 Ranging library for a localization project. Both DW1000RANGING_ANCHOR and  DW1000RANGING_TAG work great for 1 tag and 4 anchors in the system. However, when adding one more anchor newRange() stops printing the ranging data. 

I tried changing MAX_DEVICES in the DW1000Ranging.h from 4 to 5, but the problem persists. Do I need to change another number in the code for it to work with more than 4 anchors? 

If this is a timing/frequency problem and the device is not fast enough to perform more than 4 ranging instances at a time, is there a way to slow down the ranging frequency?

Thank you.

sinakahnemouyi, 2016-05-24:
The following shows the tag output reading ranging from 4 different anchors. As soon as the 5th is introduced there's an unusually large reading and then the readings stop.

`from: 1	 Range: 140   1.40	 RX power: -72.17 dBm
from: 4	 Range: 152   1.52	 RX power: -84.63 dBm
from: 2	 Range: 168   1.68	 RX power: -86.87 dBm
from: 3	 Range: 143   1.43	 RX power: -80.92 dBm
from: 1	 Range: 139   1.39	 RX power: -71.76 dBm
from: 4	 Range: 156   1.56	 RX power: -84.50 dBm
from: 2	 Range: 170   1.70	 RX power: -86.34 dBm
from: 3	 Range: 143   1.43	 RX power: -81.31 dBm
from: 1	 Range: 140   1.40	 RX power: -71.70 dBm
from: 4	 Range: 151   1.51	 RX power: -84.58 dBm
from: 2	 Range: 169   1.69	 RX power: -86.74 dBm
from: 3	 Range: 136   1.36	 RX power: -80.43 dBm
from: 1	 Range: 144   1.44	 RX power: -72.74 dBm
from: 4	 Range: 153   1.53	 RX power: -84.69 dBm
from: 2	 Range: 175   1.75	 RX power: -86.09 dBm
from: 3	 Range: 141   1.41	 RX power: -81.18 dBm
from: 1	 Range: 140   1.40	 RX power: -71.94 dBm
from: 4	 Range: 152   1.52	 RX power: -84.92 dBm
from: 2	 Range: 170   1.70	 RX power: -86.32 dBm
ranging init; 1 device added ! ->  short:5
from: 3	 Range: 132   1.32	 RX power: -83.34 dBm
from: 1	 Range: 140   1.40	 RX power: -71.50 dBm
from: 4	 Range: 156   1.56	 RX power: -81.66 dBm
from: 2	 Range: 172   1.72	 RX power: -83.70 dBm
from: 5	 Range: 22768   227.68	 RX power: -67.41 dBm
`

MikeS159, 2016-05-24:
Do you need to read from more than 4 at once, or do you just want lots of anchors to ensure good coverage?
A (clunky) work around I have at the moment is to read the first 4 anchors the tag finds, then clear the devices (currently by just resetting). I don't need more than 4 range readings to get location, but this way I can have extra anchors around if there are dark spots.

sinakahnemouyi, 2016-05-24:
Thanks for the quick reply Mike. Unfortunately I need all 5 anchor readings in real-time. I'm writing an algorithm that increases the accuracy of the calculated tag position with each extra anchor in the system. Do you happen to know what the issue is that prevents the code from being able to read more than 4 anchor ranging data?

MikeS159, 2016-05-24:
Not at the moment, it is on my todo list though. I'll post here if I find anything.

Ak319, 2016-07-11:
Has any one completed this one?

ajaybhatt003, 2016-07-29:
hey sinakahnemouyi
I am also working on localization project.....kindly share with me your code if possible

mic980000, 2016-08-15:
@sinakahnemouyi
May I ask how you set you devices address to 1,2,3,4 ? I got Random address every time :(
[image]

Phenioxrj, 2016-08-30:
Hey all I was wondering what hardware you were using in addition to the DWM1000. Im having trouble getting more then two to range at the same time.

raakeshkamal, 2016-10-27:
Hey all,
I did a small experiment to check whether the no. of anchors can be increased.I changed the MAXDEVICES to 5 in the source code.Then at first I powered the tag and 4 anchors, everything worked as usual.As soon as I powered the 5th anchor ,I started getting range from it for a while after which the whole program just stopped working(I stopped getting any more data from the anchors).Does that mean the program ran out of memory..

A code snippet from DW1000Ranging.h ..

//Max devices we put in the networkDevices array ! Each DW1000Device is 74 Bytes in SRAM memory for now.
#define MAX_DEVICES 4

Does it mean if I try to add more devices it will run out of memory?
After creating and allocating memory for 4 devices can I de-allocate the memory for a existing device and add a new one.Is it possible?If so,Which part of the code should I change?

raakeshkamal, 2016-11-03:
Hey all,
@thotro ..I did a small experiment to check whether the no. of anchors can be increased.I changed the MAXDEVICES to 5 in the source code.Then at first I powered the tag and 4 anchors, everything worked as usual.As soon as I powered the 5th anchor ,I started getting range from it for a while after which the whole program just stopped working(I stopped getting any more data from the anchors).Does that mean the program ran out of memory..

A code snippet from DW1000Ranging.h ..

//Max devices we put in the networkDevices array ! Each DW1000Device is 74 Bytes in SRAM memory for now.
#define MAX_DEVICES 4

Does it mean if I try to add more devices it will run out of memory?
After creating and allocating memory for 4 devices can I de-allocate the memory for a existing device and add a new one.Is it possible?If so,Which part of the code should I change?

raakeshkamal, 2016-11-03:
Hey all,
@thotro @MikeS159 ..I did a small experiment to check whether the no. of anchors can be increased.I changed the MAXDEVICES to 5 in the source code.Then at first I powered the tag and 4 anchors, everything worked as usual.As soon as I powered the 5th anchor ,I started getting range from it for a while after which the whole program just stopped working(I stopped getting any more data from the anchors).Does that mean the program ran out of memory..

A code snippet from DW1000Ranging.h ..

//Max devices we put in the networkDevices array ! Each DW1000Device is 74 Bytes in SRAM memory for now.
#define MAX_DEVICES 4

Does it mean if I try to add more devices it will run out of memory?
After creating and allocating memory for 4 devices can I de-allocate the memory for a existing device and add a new one.Is it possible?If so,Which part of the code should I change?

raakeshkamal, 2016-11-03:
Hey all,
@thotro @MikeS159 @jwzawadzki @valeros @ttaber @pascalbros @aacton @tommag @Rotzbua @leosayous21..I did a small experiment to check whether the no. of anchors can be increased.I changed the MAXDEVICES to 5 in the source code.Then at first I powered the tag and 4 anchors, everything worked as usual.As soon as I powered the 5th anchor ,I started getting range from it for a while after which the whole program just stopped working(I stopped getting any more data from the anchors).Does that mean the program ran out of memory..

A code snippet from DW1000Ranging.h ..

//Max devices we put in the networkDevices array ! Each DW1000Device is 74 Bytes in SRAM memory for now.
#define MAX_DEVICES 4

Does it mean if I try to add more devices it will run out of memory?
After creating and allocating memory for 4 devices can I de-allocate the memory for a existing device and add a new one.Is it possible?If so,Which part of the code should I change?

az-z, 2016-11-03:
Stop spamming.
Do more experiments.

Rotzbua, 2016-11-03:
@raakeshkamal Learn c++, debugging,´and testing ..  and by the way I do not work for free on others problems.

AlexisTM, 2016-11-25:
@raakeshkamal Do you even put your code/hardware on Github? Why do you want free help if you don't accept to share your work?

Li-JEN, 2020-08-13:
Did anyone solve this problem? I met the same problem when I added the fifth anchor. Also, I read the library code and changed the MAX_DEVICE from 4 to 5, it didn't get a better result. Are  there some problems with power or submission frequency, even over the limit of SRAM

memirerdol, 2023-10-26:
Did anyone solve this problem? How can we increase the anchors number? Unfortunately, we reached maximum 4 anchors.

samequeq, 2024-06-17:
I have same problem on my project, i need 8 anchors in one time. anyone solve this problem?