thotro/arduino-dw1000 #83: Cannot get ACC_MEM register data.

only4hj, 2016-05-28:
Hi,
I'm testing this cool library with Arduino Pro mini and DWM1000.

I can get DWM1000 info including the receive time.
But I cannot get ACC_MEM register data.

To print ACC_MEM, I called the following function from loop() of BasicReceiver.ino when the 'received' variable is true.

void DW1000Class::printAccMem() {
    Serial.println("[ AccMem Start ]");

    byte cmd = 0x25;
    unsigned int n = 4064;
    byte header[3];
    int  headerLen = 1;
    int  i;
    header[0] = READ | cmd;

    SPI.beginTransaction(*_currentSPI);
    digitalWriteFast(_ss, LOW);
    for(i = 0; i < headerLen; i++) {
        SPI.transfer(header[i]);
    }
    for(i = 0; i < n; i++) {
        byte data = SPI.transfer(JUNK);
        char message[50];
        sprintf(message, "Data %d : %d", i, data);
        Serial.println(message);
    }
    delayMicroseconds(5);
    digitalWriteFast(_ss, HIGH);
    SPI.endTransaction();

    Serial.println("[ AccMem End ]");
}

The result is all zeros as follows.

[ AccMem Start ]
Data 0 : 0
Data 1 : 0 
Data 2 : 0
Data 3 : 0
....
Data 4062 : 0
Data 4063 : 0
[ AccMem End ]

When I change the 'cmd' value from 0x25 to others (such as 0x15),
it prints proper non-zero values.

Do you get any clue why I cannot print the ACC_MEM(0x25) value?

Thank you

himemiyachikane, 2018-04-08:
Polling Accumulator CIR memory data is really a "strange" operation for general ranging application.By default,this memory is powered off.

To enable Accumulator CIR output,FACE ,AMCE ,RXCLKS in PMSC_CTRL0 must be set.

> if the host system wants to read the Channel Impulse Response Estimate (CIRE) for diagnostic purposes then, the receive clock needs to be present to access the accumulator memory.
> -- page 181 of **DW1000 User Manual**

> In normal operation this bit should be set to 0 to allow the
> PMSC to control the accumulator clock as necessary for normal receiver operation. If the host
> system wants to read the accumulator data, both this FACE bit and the AMCE bit (below) need
> to be set to 1 to allow the accumulator reading to operate correctly.
>  -- page 182 of **DW1000 User Manual**

After basic initalizing of DW1000 and before your operation,add init code refer to the following:
```c
// libdw1000 on stm32
uint8_t pmscctrl0[LEN_PMSC_CTRL0];
dwSpiRead(dev, PMSC, PMSC_CTRL0_SUB, pmscctrl0, LEN_PMSC_CTRL0);
pmscctrl0[0] &= 0xF3;
pmscctrl0[0] |= 0x48;
pmscctrl0[1] |= 0x80;
dwSpiWrite(dev, PMSC, PMSC_CTRL0_SUB, pmscctrl0, LEN_PMSC_CTRL0);
```

aminsteve, 2023-09-14:
> Polling Accumulator CIR memory data is really a "strange" operation for general ranging application.By default,this memory is powered off.
>
> To enable Accumulator CIR output,FACE ,AMCE ,RXCLKS in [...]

Were you able to get the ACC_MEM data? Even after initialization and setting the PMSC, I read the ACC_MEM as zero.