thotro/arduino-dw1000 #84: Compiling for Arduino Due

pakizip, 2016-06-01:
Hi, how can I compile the library for the Arduino Due?
The compiler gives me a lot of errors, find below just a part of them.

`/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:132:86: note: in expansion of macro 'digitalPinToPortReg'
 if (__builtin_constant_p(P) && __builtin_constant_p(V))   __atomicWrite__((uint8_t*) digitalPinToPortReg(P),P,V) \
                                                                                      ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/DW1000.cpp:1507:2: note: in expansion of macro 'digitalWriteFast'
  digitalWriteFast(_ss, HIGH);
  ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:85:73: error: 'PORTC' was not declared in this scope
 (((P) >= 0 && (P) <= 7) ? &PORTD : (((P) >= 8 && (P) <= 13) ? &PORTB : &PORTC))
                                                                         ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:120:12: note: in definition of macro '__atomicWrite__'
 if ( (int)(A) < 0x40) { bitWrite(*(A), __digitalPinToBit(P), (V) );}  \
            ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:132:86: note: in expansion of macro 'digitalPinToPortReg'
 if (__builtin_constant_p(P) && __builtin_constant_p(V))   __atomicWrite__((uint8_t*) digitalPinToPortReg(P),P,V) \
                                                                                      ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/DW1000.cpp:1507:2: note: in expansion of macro 'digitalWriteFast'
  digitalWriteFast(_ss, HIGH);
  ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:122:29: error: 'SREG' was not declared in this scope
 uint8_t register saveSreg = SREG;                          \
                             ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:132:59: note: in expansion of macro '__atomicWrite__'
 if (__builtin_constant_p(P) && __builtin_constant_p(V))   __atomicWrite__((uint8_t*) digitalPinToPortReg(P),P,V) \
                                                           ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/DW1000.cpp:1507:2: note: in expansion of macro 'digitalWriteFast'
  digitalWriteFast(_ss, HIGH);
  ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:123:5: error: 'cli' was not declared in this scope
 cli();                                                     \
     ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:132:59: note: in expansion of macro '__atomicWrite__'
 if (__builtin_constant_p(P) && __builtin_constant_p(V))   __atomicWrite__((uint8_t*) digitalPinToPortReg(P),P,V) \
                                                           ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/DW1000.cpp:1507:2: note: in expansion of macro 'digitalWriteFast'
  digitalWriteFast(_ss, HIGH);
  ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/DW1000.cpp: In static member function 'static void DW1000Class::writeBytes(byte, word, byte*, unsigned int)':
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:85:28: error: 'PORTD' was not declared in this scope
 (((P) >= 0 && (P) <= 7) ? &PORTD : (((P) >= 8 && (P) <= 13) ? &PORTB : &PORTC))
                            ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:120:12: note: in definition of macro '__atomicWrite__'
 if ( (int)(A) < 0x40) { bitWrite(*(A), __digitalPinToBit(P), (V) );}  \
            ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:132:86: note: in expansion of macro 'digitalPinToPortReg'
 if (__builtin_constant_p(P) && __builtin_constant_p(V))   __atomicWrite__((uint8_t*) digitalPinToPortReg(P),P,V) \
                                                                                      ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/DW1000.cpp:1563:2: note: in expansion of macro 'digitalWriteFast'
  digitalWriteFast(_ss, LOW);
  ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:85:64: error: 'PORTB' was not declared in this scope
 (((P) >= 0 && (P) <= 7) ? &PORTD : (((P) >= 8 && (P) <= 13) ? &PORTB : &PORTC))
                                                                ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:120:12: note: in definition of macro '__atomicWrite__'
 if ( (int)(A) < 0x40) { bitWrite(*(A), __digitalPinToBit(P), (V) );}  \
            ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:132:86: note: in expansion of macro 'digitalPinToPortReg'
 if (__builtin_constant_p(P) && __builtin_constant_p(V))   __atomicWrite__((uint8_t*) digitalPinToPortReg(P),P,V) \
                                                                                      ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/DW1000.cpp:1563:2: note: in expansion of macro 'digitalWriteFast'
  digitalWriteFast(_ss, LOW);
  ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:85:73: error: 'PORTC' was not declared in this scope
 (((P) >= 0 && (P) <= 7) ? &PORTD : (((P) >= 8 && (P) <= 13) ? &PORTB : &PORTC))
                                                                         ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:120:12: note: in definition of macro '__atomicWrite__'
 if ( (int)(A) < 0x40) { bitWrite(*(A), __digitalPinToBit(P), (V) );}  \
            ^
[... 57 lines cut ...]
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/DW1000.cpp:1571:2: note: in expansion of macro 'digitalWriteFast'
  digitalWriteFast(_ss, HIGH);
  ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:122:29: error: 'SREG' was not declared in this scope
 uint8_t register saveSreg = SREG;                          \
                             ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:132:59: note: in expansion of macro '__atomicWrite__'
 if (__builtin_constant_p(P) && __builtin_constant_p(V))   __atomicWrite__((uint8_t*) digitalPinToPortReg(P),P,V) \
                                                           ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/DW1000.cpp:1571:2: note: in expansion of macro 'digitalWriteFast'
  digitalWriteFast(_ss, HIGH);
  ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:123:5: error: 'cli' was not declared in this scope
 cli();                                                     \
     ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/digitalWriteFast.h:132:59: note: in expansion of macro '__atomicWrite__'
 if (__builtin_constant_p(P) && __builtin_constant_p(V))   __atomicWrite__((uint8_t*) digitalPinToPortReg(P),P,V) \
                                                           ^
/Users/pascal/Documents/Arduino/libraries/arduino-dw1000-master/src/DW1000.cpp:1571:2: note: in expansion of macro 'digitalWriteFast'
  digitalWriteFast(_ss, HIGH);
  ^
exit status 1
Error compiling for board Arduino Due (Programming Port).`

Thank you.

Rotzbua, 2016-06-01:
As you can see the problem is `digitalWriteFast.h`. Replace `digitalWriteFast()` with normal `digitalWrite()`.

pakizip, 2016-06-01:
Thanks for your answer.
It will not affect the performance?

Rotzbua, 2016-06-01:
No, due has much more computing power than the uno. But I not know if the library works with due, sry.

pakizip, 2016-06-01:
Thanks, man.
I managed to read the ID of the DWM1000.

For reference, this configuration with Due is working:
https://github.com/thotro/arduino-dw1000/issues/58#issuecomment-180477907

PR #86 (2016-06-09): fix digitalWriteFast only for AVR and some more changes
fixes #84 #40 
digitalWriteFast only supports avr
no need of manual change any more, pre compiler does it

Important note: impact in timing and accuracy on other systems than avr has to be tested and evaluated and may vary highly
commit 7ae6f845 (2016-06-09): fix digitalWriteFast only for AVR

fixes #84
closed 2016-06-01 by pakizip