thotro/arduino-dw1000 #85: Arduino Leonardo

pakizip, 2016-06-01:
Hi guys,
I managed to read the ID from an Arduino Due, but i don't know why i cannot read it from an Arduino Leonardo.

I followed the same connection of the testbed:
https://github.com/thotro/arduino-dw1000/blob/master/adapterBoard/AdapterBoardTestBed.png

But the answer is the following:
DW1000 initialized ...
Committed configuration ...
Device ID: DECA - model: 255, version: 15, revision: 15
Unique ID: FF:FF:FF:FF:FF:FF:FF:FF
Network ID & Device Address: PAN: FFFF, Short Address: FFFF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5

I tried also the ICSP port (i don't know if it's correct) following the same schema of the Arduino Due:
https://github.com/thotro/arduino-dw1000/issues/58#issuecomment-180477907

But the result doesn't change, it shows the 00:00:00:00:00:00:00:00 id and everything to 0.

Thank you in advance.

Rotzbua, 2016-06-01:
Maybe the reason: Leonardo has operating voltage 5V, dwm1000 has 3.3V.

pakizip, 2016-06-01:
Mmh but the Leonardo has the 3.3v pinout.

Rotzbua, 2016-06-01:
Yes normally only for supply, but not I/O-pins.

pakizip, 2016-06-01:
I see, good point man.
I'm not an expert, so how can I use the DWM1000 with Leonardo?

Rotzbua, 2016-06-01:
You need a level shifter, see https://github.com/thotro/arduino-dw1000/wiki/Hardware .

pakizip, 2016-06-01:
Thanks.
I know i'm going OT but anyway:
I have also this one https://www.arduino.cc/en/Main/ArduinoBoardNano
Should I use a level shifter? Ty.

Rotzbua, 2016-06-01:
`Operating Voltage (logic level) 5 V` yes, you should use a level shifter.

pakizip, 2016-06-05:
I open this issue again because my configuration is not working with Leonardo.
I'm sorry for the mess, but anyway I was trying to figure out how to connect the level shifter.
I should connect 5 wires, but I have the 4 channel level shifter, so this means I need 2 level shifters, right?
I cannot find the 4 channels level shifter in Fritzing, so I used a similar one, anyway this is the one i'm using:
http://www.amazon.com/Converter-Bi-Directional-Compatible-Atomic-Market/dp/B00TMILEWY

And this is the configuration, the IRQ is connected with a pulldown resistance of 12k ohm.
[image]

I leave here the project file, feel free to download it and edit it if there's something wrong in my configuration:
[AdapterBoardTestBed-Leonardo.fzz.zip](https://github.com/thotro/arduino-dw1000/files/299421/AdapterBoardTestBed-Leonardo.fzz.zip)

Thanks, bye.

wholder, 2016-06-05:
Move the Interrupt signal to pin 3 on the Arduino Leonardo.

Wayne

pakizip, 2016-06-05:
Ok, i'll try.
What about the rest? Thanks.

az-z, 2016-10-11:
@pascalbros have you ever solved the problem?

pascalbros, 2016-10-11:
Unfortunately not, just surrendered and bought some Arduino Pro Mini 3.3v

az-z, 2016-10-11:
Pasquale ,
I noticed that you do _not_ shift the irq line (pin2). Is this by design?
In my setup (mini pro 5v), if I don't shift the irq - there are long delays
(few seconds) between sent messages. If I do shift - they are pretty much
are "flying" out on the screen.
Either way I can't receive a single one on another sensor..

Rotzbua, 2017-04-24:
@pascalbros Conclusion: library do not work with Arduino Leonardo?