thotro/arduino-dw1000 #87: Comparison with a working system

pakizip, 2016-06-03:
Hi,
I'm using an Arduino Due and I'm trying to figure out why it doesn't send or receive anything but I can read and write on registers without problems.

Already checked the wires, everything is in place and if I change the polarity of the IRQ with `setInterruptPolarity,` I can clearly see the HIGH on the Pin 2, so I don't think is an hardware or connection issue.

I opened another issue because maybe this can be a "reference" of expected outputs if someone is trying to debug the code in a different platform and like me don't have a reference.

Let's start with BasicSender example, replace the transmitter code with this:

    void transmitter() {
      // transmit some data
      Serial.print("Transmitting packet ... #"); Serial.println(sentNum);
      DW1000.newTransmit();
      DW1000.setDefaults();
      
      ////START/////
      String msg = "12";
      DW1000.setData(msg);
      unsigned int dataLength = DW1000.getDataLength();
      Serial.println(dataLength);
      byte msg1[dataLength];
      DW1000.getData(msg1, dataLength);
      Serial.println(msg1[0]);
      Serial.println(msg1[1]);
      Serial.println(msg1[2]);
      ////END////
  
      // delay sending the message for the given amount
      DW1000Time deltaTime = DW1000Time(10, DW_MILLISECONDS);
      DW1000.setDelay(deltaTime);
      DW1000.startTransmit();
      delaySent = millis();
    }

My output is:

    ### DW1000-arduino-sender-test ###
    DW1000 initialized ...
    Committed configuration ...
    Device ID: DECA - model: 1, version: 3, revision: 0
    Unique ID: FF:FF:FF:FF:00:00:00:00
    Network ID & Device Address: PAN: 0A, Short Address: 05
    Device mode: Data rate: 110 kb/s, PRF: 16 MHz, Preamble: 2048 symbols (code #4), Channel: #5
    Transmitting packet ... #0
    3
    26
    26
    26

As you can see, the array is read (in bytes) but the values are wrong. Anyway the length is read correctly.

Can you please share your output? Thanks.

Rotzbua, 2016-06-03:
Arduino Uno output:

```
### DW1000-arduino-sender-test ###
DW1000 initialized ...
Committed configuration ...
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 0A, Short Address: 05
Device mode: Data rate: 110 kb/s, PRF: 16 MHz, Preamble: 2048 symbols (code #4), Channel: #5
Transmitting packet ... #0
3
72
101
108
ARDUINO delay sent [ms] ... 29
Processed packet ... #0
Sent timestamp ... 316171.62
DW1000 delta send time [ms] ... 316.17
```

Notify that this library is made for Arduino architecture AVR means 8bit, but Due is architecture ARM means 32bit! So there can be differences...

pakizip, 2016-06-03:
Your output seems fine, but also if the result can be different on 32 bit, the 26 26 26 is obviously wrong.
So maybe the code used for writing doesn't work on Due?
@thotro any suggestions?

pakizip, 2016-06-04:
This is my SYS_STATUS read directly from the DWM1000 (and not just the var):
    00 00 80 00 02
So according to the manual, this should be CPLOCK (1) and SLP2INIT (1).

Sometimes i get `00 02 80 00 02` so there is an CLKPLL_LL error, but if I move the wires and restart the system, it'll work eventually.

My jumpers are 20 cm, too much for SPI? Anyway i can read the values from the DWM1000, but it doesn't send anything.

az-z, 2016-08-08:
@Rotzbua : Did you made any changes to the code/wiring/anything to make UNO to work with this code ?

Rotzbua, 2016-08-08:
@az-z no

beautifulsmall, 2016-11-17:
I've been comparing a Due with a 3V3 converted Nano. The MISO data from the DWM1000 appears to shift clock phases after the first read via enableClock(), if I print out the read pmscctrl0[0] 0-3, I get 0,2,0x30,0xF0 but after the reset there is a second call to enableClock() and this is interpreted as 0,4,0x61,0xE0 , a bit shift in all bytes.
My logic analyser still reports 0,2,0x30,0xF0.  The MISO data hold time from the clock rising edge is only about 30ns. Ive finally managed to get the send receive script working by setting all the SPI calls to a specific pin, ie   SPI.begin(10); using transfer(10,data,SPI_CONTINUE); and setting the clockdivider to 35 AND adding an RC delay onto the MISO line , 50R and 2.2nF ~ 100ns added delay. 
Possibly more than one problem here. Ive also a pulldown on interrupt, pullup on MISO, pulldowns on the mode selects (shouldn't need them) on the DWM1000.

aacton, 2016-11-17:
This sounds a lot like some issues I was facing with the ESP8266. I traced it back to the way the clock was being changed by this library. Short version: the manual calls for 2 bytes to be written, but the library was writing 4. I just opened a pull request with branch name "lde_load_fix". Please give it a try and let us know if it resolves you issues.